indilib/indi-3rdparty #561: [ERROR] Unable to connect to Orion StartShoot G3: Operation not permitted

pabutler1, 2022-03-29:
**Describe the bug**
After manually installing the driver (tried in both Stellarmate and Astroberry servers with Kstars and Ekos) I get an error when trying to connect (inside Ekos) and it reports this error: [ERROR] Unable to connect to Orion StartShoot G3: Operation not permitted.  In Astroberry it actually seems to crash and restart but never gets past that part and just keeps crasing.  I have sent logs over to the Stellarmate team and after several days of troubleshooting (validating USB is working, camera is working, software is up to date, etc) we are stuck at this issue.   I will note that it seems to work just fine on my MAC - if I run Kstars and Ekos it will connect just fine to my Orion CCD camera. 

**To Reproduce**
Exact steps to reproduce the behavior.
1. sudo apt-get update && sudo apt-get -y install indi-orion-ssg3
2. Start Kstars and Ekos - add the camera as an item, start profile, connect to camera and fail
3. See error attached

**Expected behavior**
I expect it to continue past the connection and then show the display as normal and connection as good.

**Screenshots**
See attached logs

**Desktop (please complete the following information):**
 - OS:  Raspberry PI - Stellarmate - latest version
 - Version [e.g. 1.7.6, GIT..etc] - Stellarmate 1.7.0

**Log Files**
Make sure to enable logging and include log files - See attached
[log_18-44-36.txt](https://github.com/indilib/indi-3rdparty/files/8370833/log_18-44-36.txt)
[log_18-40-13.txt](https://github.com/indilib/indi-3rdparty/files/8370836/log_18-40-13.txt)
[HARDWARE_Details.pdf](https://github.com/indilib/indi-3rdparty/files/8370837/HARDWARE_Details.pdf)
[USB_Details.pdf](https://github.com/indilib/indi-3rdparty/files/8370838/USB_Details.pdf)
[Error.pdf](https://github.com/indilib/indi-3rdparty/files/8370878/Error.pdf)

bgilsrud, 2022-03-29:
Thanks for the detailed bug report. The issue is that the debian package doesn't include the udev rule file that's required to make the device accessible to non-privileged users. I'll submit a fix.

pabutler1, 2022-03-29:
Thanks a lot Ben, let me know if you need anything else - I have seen your good works helping folks out with your forum posts - much appriciated!

pabutler1, 2022-03-30:
Hey Ben - is there a way for me to get this fix now or do I have to wait for a release?  I have run the command to get the latest Indi drivers, Kstars and Ekos but still have the same problem - sorry for my ignorance - I am sure there is a command to run to get it, I just can't seem to find it.  Thanks

bgilsrud, 2022-03-30:
I'm not sure how long it will take to get into the release packages for the platform that you're using. In the meantime, you can do this:

sudo vim /lib/udev/rules.d/99-orionssg3.rules
Put this in the file:
```
# All Orion Cameras
SUBSYSTEMS=="usb", ATTR{idVendor}=="07ee", MODE="0666"
```
sudo udevadm control --reload-rules
sudo udevadm trigger

knro, 2022-03-30:
Release already made to StellarMate & INDI Nightly PPA as well.

closed 2022-03-30 by knro

pabutler1, 2022-03-30:
Thanks Ben - everything is working now - I looked at the file and it was identical to yours so everything was good when I relaunched it again.

pabutler1, 2022-03-30:
Thanks Jasem!!

PR #562 (2022-03-29): Add orion ssg3 udev rule to deb package
fixes #561