indilib/indi-3rdparty #567: Atik Horizon 2 resetting

TallFurryMan, 2022-04-10:
**Describe the bug**
Infrequently, after working properly for some time during an observation session, the Atik Horizon 2 camera would suddenly return only black frames to the client. Only stopping and restarting the INDI driver recovers the proper behaviour, and the very first frame after this recovery is corrupted.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run driver, capture frames
2. Suddenly, frames are provided empty, with value zero.
3. Check `dmesg`

**Expected behavior**
If the camera firmware resets for whatever reason, the driver should recover the proper functionality of the camera.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Lubuntu 21.10
 - KStars 3.5.8, INDI 1.9.6-e0158b46, both built from git.

**Log Files**
This excerpt can be seen in `dmesg` at the moment frames turn black:
```
[avril10 02:30] usb 2-1: usbfs: usb_submit_urb returned -19
[  +0,000240] usb 2-1: usbfs: usb_submit_urb returned -19
[  +0,000136] usb 2-1: usbfs: usb_submit_urb returned -19
[  +0,010731] usb 2-1: USB disconnect, device number 3
[  +0,224075] usb 2-1: new SuperSpeed USB device number 4 using xhci_hcd
[  +0,013869] usb 2-1: New USB device found, idVendor=20e7, idProduct=dfc3, bcdDevice= 0.00
[  +0,000046] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  +0,000007] usb 2-1: Product: Atik Horizon 2
[  +0,000005] usb 2-1: Manufacturer: Atik Cameras
[  +0,000004] usb 2-1: SerialNumber: 0000000000
```

TallFurryMan, 2022-04-10:
The Atik Horizon 2 requires a firmware to be loaded before usage. The official procedure is to power the camera up, then connect the USB cable. At this point, when starting the driver, the firmware loads and the camera is ready to use.

However, when an automated observatory system, with all cables plugged in, turns the camera power supply up, the USB cable is already connected. In this situation, when starting the driver, the firmware loads but the camera cannot be used: the driver has to be stopped and restarted for the USB protocol to be acknowledged properly by the OS. This has to do with the bootloader of the camera, and is documented as such by Atik.

The issue here is that when the camera resets in the middle of the observation, which can be detected from the zeroed frame, the driver should be stopped and restarted as well for the functionality to be restored. Precisely, the Atik library should (probably) be unloaded and reloaded for the firmware to be recognised properly, and settings be restored and/or reported to the client (e.g. cooler should be reconfigured quickly before the temperature raises too much and a ramp has to be exercised).

knro, 2022-04-10:
So it this an issue in INDI driver itself? or Atik SDK?

TallFurryMan, 2022-04-10:
There is something in the SDK changelog that relates, but it is actually the current version:
```
2020.10.19 - DLL V2020.10.19.677
  Fixed issue with binning causing distorted images on the ACIS/Horizon.
  Fixed issue with ACIS/HorizonII being left in Normal or Fast Mode when shutdown causing the camera to return blank images on re-connect.
```

There are a few changes more recent that we should integrate before communicating with Atik:
```
2021.11.02 - DLL v2021.11.02.862
  Fixed issue where our CMOS cameras (ACIS/Horizon/Apx) would always report PowerSave as their exposure speed.
2021.10.22 - DLL v2021.10.22.847
  Fixed error with “ArtemisHotPixelAdvancedCalculcationComplete” renamed “ArtemisHotPixelAdvancedCalculationComplete”
  Updated the C++ example to include the latest revision of AtikCameras.h & AtikCameras.cpp
2021.10.18 - DLL v2021.10.18.840
  Added 4 new functions to the API:
    ArtemisHotPixelAutoRemoval
    ArtemisHotPixelAdvancedRemoval
    ArtemisHotPixelAdvancedStartCalculateHotPixels
    ArtemisHotPixelAdvancedCalculcationComplete
  Mac build no longer statically links to libusb
  First stable build containing Apx60 Code
    Setting the camera’s exposure speed to ‘Fast’ for the moment sets the camera speed internally to ‘Normal’ while a bug is being fixed.
  Updated documentation. Now using Doxygen.
  Fixed an issue where the dll would crash on exit (mostly occurred if a camera was imaging)
  The C#/.NET example has been updated to incorporate calls to the updated HotPixelRemoval
```

TallFurryMan, 2022-04-10:
> So it this an issue in INDI driver itself? or Atik SDK?

If there was a single camera in the setup, I'd say we could recover by restarting the driver completely. But we cannot do this when there are multiple devices managed by the driver and the library. Or we could separate each instance, but that's not how the library is documented.

Delimiting the issue will probably require more discussion.

knro, 2024-01-23:
@TallFurryMan is this issue still active/relevant?

TallFurryMan, 2025-05-05:
I hav not experienced any problem after #1057 so I'm closing this issue as resolved.