indilib/indi-3rdparty #570: libqhy: missing "config.h" header file

maclomhair, 2022-04-19:
**Describe the bug**
In the bundled libqhy QHYCCD SDK, one header file seems to be missing: config.h, which is referenced by other header files. Maybe it also works without it, since it consists only of a couple of lines, but still gcc complains.

Is there a specific reason not to include this file?

**To Reproduce**
Look at libqhy/qhyccd.h and see that "config.h" is included, but not provided. Compiling the test application fails using the Makefile and the so installed library because of the missing config.h

**Expected behavior**
"config.h" is bundled as well

**Desktop (please complete the following information):**
- OS: Arch Linux
 - Version: 2022-04-19

knro, 2022-11-16:
INDI QHY driver can be linked just fine against libqhy, so how come it does not fail if that file is missing?

maclomhair, 2022-12-05:
The INDI QHY driver seems to have a config.h file on its own which is then used - I haven't tested that, though.

closed 2023-02-21 by knro