indilib/indi-3rdparty #574: rpicam Generating .bin Files for Images via Ekos

Etep15, 2022-05-04:
**Describe the bug**
First off, I'm not sure if this is a bug or not so apologies if this is in the wrong section. I'm using a raspberry pi zero with the raspberry pi hq cam. I just build the system this week by imaging onto the sd card. I've also gone through the process of using apt update and apt upgrade so I should be at the latest version of everything.

It took me a bit but I got indi_rpicam working by building from source (installing via api caused issues so I removed it first before building from source.). I also created the xml file for the web manager and am able to access and run kstars and ekos remotely. So all good there. 

When I use Ekos to create a preview image or a sequence of any kind I just get a bunch of .bin files that are 24.7 MB. It tells me that there is some sort of image data there but my stacking software doesn't understand it. I assumed it would be .fit or .jpg (or maybe .raw?) depending on whether I had selected native or fit. 

How do I convert these .bin files into something useful? Or what should I look at if I've configured something incorrectly?

**To Reproduce**
Exact steps to reproduce the behavior.
1. Build and install from source using the steps described for 3rd party in the README on this repository. 
2. Create the web manager xml file in /usr/share/indi/indi_rpicam.xml
3. Restart
4. Open indi web manager and start it with the indi_rpicam driver selected. 
5. Connect to it remotely via Ekos in kstars
6. Generate a preview image from the CCD tab (camera looking button at top). 
7. You will receive a file that ends in .bin

**Expected behavior**

I would've expected a .fit file.

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Desktop (please complete the following information):**
 - OS:
 - Version [e.g. 1.7.6, GIT..etc]

Raspian. Latest version

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html

Logs from Connecting to the remote : 

2022-05-04T09:57:18 RPI Camera is online.
2022-05-04T09:57:18 Remote devices established.
2022-05-04T09:57:18 INDI services started. Connection to remote INDI server is successful. Waiting for devices...
2022-05-04T09:57:18 Connecting to remote INDI server at 10.200.100.63 on port 7,624 ...
2022-05-04T09:57:17 Establishing communication with remote INDI Web Manager...

When capturing an image :

2022-05-04T09:58:47 Download Time: 10.04 s, New Download Time Estimate: 10.04 s.
2022-05-04T09:58:19 Capturing 1.000-second  image...

Filename received was : 

Image_2022-05-04T09-58-47.bin

Etep15, 2022-05-04:
Well I sorted out the issue.

TL;DR : My bad.

I didn't realize that the Indi Web Manager xml file for the device was created through the process of building the driver from source. However it does not seem to install it. It should've been installed to /usr/share/indi/ but when I checked there originally I didn't see the file. So I copied a file (indi_atik.xml I believe) and then just changed the values to the appropriate driver. However there was  flag set in there that I didn't realize and that likely was causing the issue. 

I then when back to my Projects folder and into the build folder to see what was actually created and there was the indi_rpicam.xml file that I needed. I copied, that over to /usr/share/indi and restarted the Raspberry PI for good measure. I started up the server after selecting the rpicam driver and then connected to it through ekos on my mac. Now I'm getting fits images!

I'll leave this here in case it helps anyone in future.

knro, 2022-05-05:
Running sudo make install for the driver doesn't install it? what's the output from sudo make install?

closed 2022-06-02 by knro