indilib/indi-3rdparty #582: libflycapture.so.2.13.3.31 executable stack

paolostivanin, 2022-05-30:
Hello,
our build checks are complaining about the following issue:

```
libflycapture2.x86_64: E: executable-stack (Badness: 10000) /usr/lib64/libflycapture.so.2.13.3.31
```

How can I reach Atik to kindly ask them to fix this (possible) security issue?

thanks

TallFurryMan, 2022-05-30:
Would it be possible to have a extended log of the detection, and a procedure to validate this is not the case anymore?

TallFurryMan, 2022-06-19:
@paolostivanin any script you could share to reproduce the report?

paolostivanin, 2022-07-11:
@TallFurryMan sorry, I missed the notification. Yes, you can use the `execstack` tool.

Example where stack is executable:
```
$ execstack libatik/x64/libflycapture.bin 
X libatik/x64/libflycapture.bin
```

Example where stack is not executable:
```
$ execstack libasi/x64/libASICamera2.bin 
- libasi/x64/libASICamera2.bin
````

paolostivanin, 2022-07-11:
```
$ for l in $(find lib*/x64/*|xargs); do execstack $l; done
- libaltaircam/x64/libaltaircam.bin
- libasi/x64/libASICamera2.bin
- libasi/x64/libEAFFocuser.bin
- libasi/x64/libEFWFilter.bin
- libasi/x64/libUSB2ST4Conv.bin
- libatik/x64/libatikcameras.bin
X libatik/x64/libflycapture.bin
- libmallincam/x64/libmallincam.bin
- libnncam/x64/libnncam.bin
- libomegonprocam/x64/libomegonprocam.bin
- libplayerone/x64/libPlayerOneCamera.bin
- libstarshootg/x64/libstarshootg.bin
- libtoupcam/x64/libtoupcam.bin
```

Only `libflycapture` seems to be the problem here.

paolostivanin, 2022-07-11:
Another way is with `readelf`:

```
$ readelf -lW libflycapture.bin |grep GNU_STACK
  GNU_STACK      0x000000 0x0000000000000000 0x0000000000000000 0x000000 0x000000 RWE 0x10
```

if the output has `E` (like in the above case), then the stack is executable.

Example of non executable stack using `readelf`:

```
$ readelf -lW libatikcameras.bin |grep GNU_STACK
  GNU_STACK      0x000000 0x0000000000000000 0x0000000000000000 0x000000 0x000000 RW  0x10
```

knro, 2022-11-16:
I think you need to reach out to Atik as we can't do anything about a binary blob.

closed 2022-11-16 by knro