indilib/indi-3rdparty #584: Opening remote serial shutter port: /dev/tty.usbserial-xxxx is blocked on Mac

captainchain, 2022-05-30:
**Describe the bug**
open remote serial shutter port blocked when I set a 40 seconds exposure on Ekos (set Force BULB ON on INDI Control Panel),
then it blocked forever except I restart the KSTARS, here is the log:

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run driver
2. Perform action
3. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS:
 - Version [e.g. 1.7.6, GIT..etc]

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html

knro, 2022-05-31:
oh I see that O_NONBLOCK was sent as the mode_t which is incorrect. Does O_RDWR | O_NONBLOCK work as well ?

captainchain, 2022-05-31:
@knro pretty well

commit 3e2d9ebb (2022-05-31): Fix opening of remote serial shutter. Fixes #584
closed 2022-05-31 by knro