indilib/indi-3rdparty #587: How to create a 'bin' file in the folder libsv305.

jctk, 2022-06-13:
Hi @thx8411 

**I would like to know how to create a 'bin' file in the folder libsv305.**

The next version of SVBONY SDK 1.6.5 will be released soon.
The SDK contains .a and .so files, but no bin files.
Could you please tell me how to create bin files from these files?

Thanks

thx8411, 2022-06-18:
Hi Kakura !

The "bin" file is basically a renamed ".so" file.
I just downloaded the SDK  1.7.3. but I think we must wait a bit before
updating libsv305.

I manage a small IT department in real life, and we have a rule of thumb :
"one change at a time" ;-)
(or we won't be able to know where our issues come from...)

I agree with you about 'the frame grabbing time' issue. It comes from the
SDK.
I hope vers 1.7.3 will fix it.

Best regards !

Blaise

jctk, 2022-06-18:
Hi, Blaise

Thanks for the answer.
I am closing this question as I have gotten an answer from you.

Even though I just replaced it with 1.7.3, I could not confirm stable operation for the last few days. In conclusion, I don't think I can update to 1.7.3 on RPi yet.

The frame grabbing time does not seem to occur when cooler on for SV405CC. I confirmed this with SDK v1.6.5. 

Best regards
Kakura

closed 2022-06-18 by jctk