indilib/indi-3rdparty #591: indi_playerone_ccd crashed

Daviddb35, 2022-06-20:
Hi,

indi_playerone_ccd crashed after trying to connect devices in Ekos.
Run in Stellarmate 64 bit v1.7.1 and reproduced on the 32 bit version.

Here are the log:

[2022-06-20T20:35:50.170 CEST INFO ][           org.kde.kstars.indi] - PlayerOne CCD Uranus-C :  "[INFO] Setting intital bandwidth to AUTO on connection. "
[2022-06-20T20:35:50.175 CEST INFO ][           org.kde.kstars.indi] - PlayerOne CCD Uranus-C :  "[INFO] CCD is online. Retrieving basic data. "
[2022-06-20T20:35:50.176 CEST INFO ][           org.kde.kstars.ekos] - "PlayerOne CCD Uranus-C" Version: "0.5" Interface: 6 is connected.
[2022-06-20T20:35:50.176 CEST DEBG ][           org.kde.kstars.ekos] - 1  devices connected out of  3
[2022-06-20T20:35:50.176 CEST WARN ][                       default] - QObject: Cannot create children for a parent that is in a different thread.
(Parent is ClientManager(0x55afe18ab0), parent's thread is QThread(0x55ac356530), current thread is QThread(0x7f0c0053a0)
[2022-06-20T20:35:50.525 CEST INFO ][           org.kde.kstars.ekos] - "PlayerOne CCD Uranus-C is offline."
[2022-06-20T20:35:50.532 CEST CRIT ][           org.kde.kstars.indi] - INDI driver  "indi_playerone_ccd"  crashed!

Daviddb35, 2022-06-20:
indi-playerone-dbg/unknown,now 0.6-nightly~202206191725 arm64 [installed]
indi-playerone/unknown,now 0.6-nightly~202206191725 arm64 [installed]

Daviddb35, 2022-06-20:
Issue with the Uranus-C.

lordzurp, 2022-06-20:
Hi !
I just receive the saturn-c today, and have the same issue as DavidDb35

tried the stable 0.5 (can't connect to cam) and nightie 0.6 (with lib 0.3 : driver crash gently; lib 0.4 : kstar whole crash )

Daviddb35, 2022-06-20:
Here is th debug log

[indi_playerone_ccd_21-28-59.log](https://github.com/indilib/indi-3rdparty/files/8943775/indi_playerone_ccd_21-28-59.log)

Daviddb35, 2022-06-20:
And here are debug log from kstars. Stellarmate 64 bit v1.7.1

[log_18-26-34.txt](https://github.com/indilib/indi-3rdparty/files/8943795/log_18-26-34.txt)
[log_18-30-57.txt](https://github.com/indilib/indi-3rdparty/files/8943796/log_18-30-57.txt)
[log_21-34-10.txt](https://github.com/indilib/indi-3rdparty/files/8943797/log_21-34-10.txt)
[log_21-34-53.txt](https://github.com/indilib/indi-3rdparty/files/8943798/log_21-34-53.txt)
[log_21-35-13.txt](https://github.com/indilib/indi-3rdparty/files/8943799/log_21-35-13.txt)
[log_21-36-26.txt](https://github.com/indilib/indi-3rdparty/files/8943800/log_21-36-26.txt)
[log_21-39-20.txt](https://github.com/indilib/indi-3rdparty/files/8943801/log_21-39-20.txt)
[log_21-40-11.txt](https://github.com/indilib/indi-3rdparty/files/8943802/log_21-40-11.txt)
[log_21-40-25.txt](https://github.com/indilib/indi-3rdparty/files/8943803/log_21-40-25.txt)
[log_21-47-37.txt](https://github.com/indilib/indi-3rdparty/files/8943804/log_21-47-37.txt)
[log_21-48-32.txt](https://github.com/indilib/indi-3rdparty/files/8943805/log_21-48-32.txt)
[log_22-26-47.txt](https://github.com/indilib/indi-3rdparty/files/8943806/log_22-26-47.txt)
[log_22-49-36.txt](https://github.com/indilib/indi-3rdparty/files/8943807/log_22-49-36.txt)
[log_22-51-54.txt](https://github.com/indilib/indi-3rdparty/files/8943808/log_22-51-54.txt)
[log_23-13-49.txt](https://github.com/indilib/indi-3rdparty/files/8943809/log_23-13-49.txt)
[log_23-15-50.txt](https://github.com/indilib/indi-3rdparty/files/8943810/log_23-15-50.txt)
[log_23-17-48.txt](https://github.com/indilib/indi-3rdparty/files/8943811/log_23-17-48.txt)
[log_23-18-17.txt](https://github.com/indilib/indi-3rdparty/files/8943812/log_23-18-17.txt)
[log_23-18-23.txt](https://github.com/indilib/indi-3rdparty/files/8943813/log_23-18-23.txt)
[log_23-20-41.txt](https://github.com/indilib/indi-3rdparty/files/8943814/log_23-20-41.txt)
[log_23-29-49.txt](https://github.com/indilib/indi-3rdparty/files/8943815/log_23-29-49.txt)

lordzurp, 2022-06-20:
Kstar log

`[2022-06-20T21:25:53.034 UTC DEBG ][           org.kde.kstars.indi] - Adding managed driver "PlayerOne CCD"
[2022-06-20T21:25:53.035 UTC DEBG ][           org.kde.kstars.indi] - Starting INDI Driver  "indi_playerone_ccd"
[2022-06-20T21:25:53.036 UTC DEBG ][           org.kde.kstars.indi] - INDI Server:  "2022-06-20T21:25:53: FIFO: start indi_playerone_ccd"
[2022-06-20T21:25:53.036 UTC DEBG ][           org.kde.kstars.indi] - INDI Server:  "2022-06-20T21:25:53: FIFO: Starting driver indi_playerone_ccd"
[2022-06-20T21:25:53.037 UTC DEBG ][           org.kde.kstars.indi] - INDI Server:  "2022-06-20T21:25:53: Driver indi_playerone_ccd: pid=4683 rfd=11 wfd=11 efd=12"
[2022-06-20T21:25:53.037 UTC DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2022-06-20T21:25:53.039 UTC DEBG ][           org.kde.kstars.indi] - INDI Server:  "2022-06-20T21:25:53: Driver indi_playerone_ccd: indi_playerone_ccd: error while loading shared libraries: libPlayerOneCamera.so.2: cannot open shared object file: No such file or directory"
[2022-06-20T21:25:53.039 UTC DEBG ][           org.kde.kstars.indi] - INDI Server:  "2022-06-20T21:25:53: Driver indi_playerone_ccd: read: Connection reset by peer"
[2022-06-20T21:25:53.039 UTC DEBG ][           org.kde.kstars.indi] - INDI Server:  "2022-06-20T21:25:53: Driver indi_playerone_ccd: Terminated after #0 restarts."
[2022-06-20T21:25:53.040 UTC DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2022-06-20T21:25:53.040 UTC CRIT ][           org.kde.kstars.indi] - INDI driver  "indi_playerone_ccd"  crashed!
[2022-06-20T21:26:00.689 UTC INFO ][           org.kde.kstars.ekos] - "Unable to establish:\n+ PlayerOne CCD\nPlease ensure the device is connected and powered on."
[2022-06-20T21:26:05.906 UTC DEBG ][           org.kde.kstars.indi] - INDI: Stopping local drivers...
[2022-06-20T21:26:05.906 UTC DEBG ][           org.kde.kstars.indi] - Removing managed driver "PlayerOne CCD"`

INDI log
``INFO	99.365046 sec	: Session log file /home/stellarmate/.indi/logs/2022-06-20/indi_playerone_ccd/indi_playerone_ccd_21:47:30.log
DEBUG	236.677981 sec	: Toggle Logging Level -- Driver Debug
DEBUG	241.216918 sec	: Attempting to open PlayerOne CCD Saturn-C SQR...
INFO	241.288044 sec	: Setting intital bandwidth to AUTO on connection.
INFO	241.291787 sec	: CCD is online. Retrieving basic data.
DEBUG	241.297615 sec	: Control #1: name (Exposure), Descp (exposure duration(microsecond(us))), Min (10), Max (2000000000), Default Value (10000), isSupportAuto (False), isWritale (True) 
DEBUG	241.297986 sec	: Control #2: name (Gain), Descp (gain), Min (0), Max (600), Default Value (0), isSupportAuto (False), isWritale (True) 
DEBUG	241.298141 sec	: Adding above control as writable control number 0.
DEBUG	241.298277 sec	: Control #3: name (WB_R), Descp (white balance: red pixels), Min (-1200), Max (1200), Default Value (0), isSupportAuto (True), isWritale (True) 
DEBUG	241.298408 sec	: Adding above control as writable control number 1.
DEBUG	241.298538 sec	: Adding above control as auto control number 0.
DEBUG	241.298671 sec	: Control #4: name (WB_G), Descp (white balance: green pixels), Min (-1200), Max (1200), Default Value (0), isSupportAuto (True), isWritale (True) 
DEBUG	241.298803 sec	: Adding above control as writable control number 2.
DEBUG	241.298927 sec	: Adding above control as auto control number 1.
DEBUG	241.299062 sec	: Control #5: name (WB_B), Descp (white balance: blue pixels), Min (-1200), Max (1200), Default Value (0), isSupportAuto (True), isWritale (True) 
DEBUG	241.299194 sec	: Adding above control as writable control number 3.
DEBUG	241.299317 sec	: Adding above control as auto control number 2.
DEBUG	241.300539 sec	: Control #6: name (Offset), Descp (offset), Min (0), Max (1000), Default Value (35), isSupportAuto (False), isWritale (True) 
DEBUG	241.300663 sec	: Adding above control as writable control number 4.
DEBUG	241.300767 sec	: Control #7: name (AutoExpMaxGain), Descp (maximum gain value when gain is automatic), Min (0), Max (600), Default Value (0), isSupportAuto (False), isWritale (True) 
DEBUG	241.300849 sec	: Adding above control as writable control number 5.
DEBUG	241.300927 sec	: Control #8: name (AutoExpMaxExpMS), Descp (maximum exposure value(ms) when exposure is automatic), Min (1), Max (1000), Default Value (100), isSupportAuto (False), isWritale (True) 
DEBUG	241.301011 sec	: Adding above control as writable control number 6.
DEBUG	241.301095 sec	: Control #9: name (AutoExpTargetBrightness), Descp (target brightness value when exposure is automatic), Min (50), Max (200), Default Value (100), isSupportAuto (False), isWritale (True) 
DEBUG	241.301233 sec	: Adding above control as writable control number 7.
DEBUG	241.301312 sec	: Control #10: name (HardwareBin), Descp (support hardware bin?), Min (0), Max (1), Default Value (0), isSupportAuto (False), isWritale (True) 
DEBUG	241.301388 sec	: Adding above control as writable control number 8.
DEBUG	241.301466 sec	: Control #11: name (Temperature), Descp (sensor temperature(degrees celsius)), Min (-4591138345127510016), Max (4636737291354636288), Default Value (0), isSupportAuto (False), isWritale (False) 
DEBUG	241.301544 sec	: Control #12: name (Guide North), Descp (set and get guide north status(has ST4 port only)), Min (0), Max (1), Default Value (0), isSupportAuto (False), isWritale (True) 
[... cut ...]
DEBUG	241.306319 sec	: Control #23: name (Pixels Binning Sum), Descp (take the sum of pixels after binning), Min (0), Max (1), Default Value (0), isSupportAuto (False), isWritale (True) 
DEBUG	241.306581 sec	: Adding above control as writable control number 11.
DEBUG	241.306823 sec	: CCD ID: 0 Width: 3008 Height: 3008 Binning: 1x1 Image Type: 0
INFO	241.307395 sec	: The CCD Temperature is 0.000.
DEBUG	241.307619 sec	: setupParams POASetImage (3008x3008,  bin 1, type 0)
DEBUG	241.307876 sec	: Pixel format 8 is supported by SER recorder.
DEBUG	241.308138 sec	: Pixel format 8 is supported by RAW encoder.
``

knro, 2022-06-21:
How are you updating the driver? from stable channel? or building from GIT?

Daviddb35, 2022-06-21:
From stable channel and switched to nightly to give it a try too

hiro3110i, 2022-06-21:
Thank you for trying the new driver. I apologize for any inconvenience.
I don't have Uranus-C or Saturn-C so I can't reproduce the details, but I tested it with Neptune-C II on 32bit Astroberry 2.0.4.

According to lordzurp's kstars.log, libPlayerOneCamera.so.2 cannot be found.
However, since indi_playerone_ccd v0.6, libPlayerOneCamera.so.3 is required.
Library mismatch seems to be the cause of the crash.
Would you please install and use the new version of libplayerone which was pushed to github?

Daviddb35, 2022-06-21:
Hi, thank you for you reply.
I was helped tomorrow night by astro friends who have better skills in debbuging. 
I have some knowledge but can you please indicate me how to do that. I'm sorry for that.

Daviddb35, 2022-06-21:
Hi,
I disable "auto connect" on ekos devices tab. The indi panel is opening, I saved the configuration in indi panel "Options" and click "connect" in the "Main Control" tab. The camera is well connected and take frames.
The only issue remaining is an error message in the indi info panel:
**2022-06-21T11:24:06: [ERROR] CCD_CAPTURE_FORMAT: newSwitchVector with no valid members**

knro, 2022-06-21:
For INDI stable v1.9.6 release, libplayerone version was 2.0.6

I just rebuilt libplayerone packages, not sure if this would help.

Daviddb35, 2022-06-21:
I was in nightly for now. I switch to stable this afternoon and try the same thing.

knro, 2022-06-21:
@hiro3110i It appears that addCaptureFormat is not available in the driver. It's a recent addition to INDI. Check out this implementation from [ASI](https://github.com/indilib/indi-3rdparty/blob/3615b54473761d3a1067a312ae1a6a888a9fc370/indi-asi/asi_base.cpp#L644) driver.

I think you can add it after line 804 in playerone_ccd.cpp

Daviddb35, 2022-06-21:
Camera connected in stable version with this method !

hiro3110i, 2022-06-21:
@knro I know the ERROR message which is related to CCD_CAPTURE_FORMAT.
But I can't understand this feature. Is it only valid for video streaming format?

@Daviddb35 Thank you for check it again.
I'm not sure if Uranus-C is available in stable version.
Stable release v1.9.6 can work with  libPlayerOneCamera.so.2.
And nightly built version can work with libPlayerOneCamera.so.3.
Basically "Auto connect" is available for both versions.

Daviddb35, 2022-06-21:
@hiro3110i uranus-c is working in stable version too. Waiting to @lordzurp if saturn-c work with this trick.
After this manipulation the .xml is created in /home/stellarmate/.indi and the camera work well. Thank you for your assistance.

knro, 2022-06-22:
> @knro I know the ERROR message which is related to CCD_CAPTURE_FORMAT. But I can't understand this feature. Is it only valid for video streaming format?
>
> @Daviddb35 Thank you for check it again. I'm [...]

This is just to define which CAPTURE format is supported, for still and video streaming. Just one line need to be added as I indicated above.

lordzurp, 2022-06-24:
Hi !

I just build indi_core and indi_3thparty from sources, and it works \o/
(but doing it with the dirty'n'dirty methode, with lot of tries and nightly packages install in parrallel ...)

will try with a fresh clean install of stellarmate this week-end

GwenFr, 2022-06-25:
Hi @knro 
Do you plan to incorporate the corrected driver soon in Stellarmate Update Tool (or ask them to do so) ?
I do not have your skill to modify and correct myself (this is Chinese for me !)
Thanks.

Daviddb35, 2022-06-25:
> Hi @knro
> Do you plan to incorporate the corrected driver soon in Stellarmate Update Tool (or ask them to do so) ?
> I do not have your skill to modify and correct myself (this is Chinese for me !)
> Thanks.

Hi Gwen,

@knro (Jasem) plan to incorporate the fix in the next update. :)

Daviddb35, 2022-06-26:
Hi,

the uranus is connecting well but when I set up everything last night, images won’t download. After setting the gain, the camera send files but after a short time it does not respond anymore.
I forgot to save log but here are messages I saw in indi panel
[image]

hiro3110i, 2022-06-26:
@Daviddb35 Are you trying the latest version of PlayerOne library and driver?
In other words, are you using libPlayerOneCamera.so.3? (libPlayerOneCamera.so.3.0.2 to be exact)
Make sure you have a symbolic link /usr/lib/arm-linux-gnueabihf/libPlayerOneCamera.so to libPlayerOneCamera.so.3.

By the way, manufacturer has already released newer version of SDK v3.0.3 on June 20.
They said, it improves stability for new camera products.
I'm planning to update as soon as possible.
I should prioritize SDK updates if RGB24 is not important for many users.

@knro What do you think about it?

Daviddb35, 2022-06-26:
@hiro3110i Nothing like this in /usr/lib/, arm-linux-gnueabihf does not exist.

Daviddb35, 2022-06-26:
after update_indi_drivers, found this :
-- Up-to-date: /usr/lib/aarch64-linux-gnu/libPlayerOneCamera.so
-- Up-to-date: /usr/lib/aarch64-linux-gnu/libPlayerOneCamera.so.3
-- Up-to-date: /usr/lib/aarch64-linux-gnu/libPlayerOneCamera.so.3.0.2

Daviddb35, 2022-06-26:
lrwxrwxrwx  1 root root        27 Jun 23 10:15 libPlayerOneCamera.so.3 -> libPlayerOneCamera.so.3.0.2

Daviddb35, 2022-06-26:
lrwxrwxrwx  1 root root        23 Jun 23 10:15 libPlayerOneCamera.so -> libPlayerOneCamera.so.3

Daviddb35, 2022-06-26:
so it was in  /usr/lib/aarch64-linux-gnu/

hiro3110i, 2022-06-26:
@Daviddb35 Oh, your environment is 64bit based, sorry.
Can you replace the library on your side?
Attached file is updated version for 64bit Raspberry Pi.
You should do unzip an attached file and copy it into /usr/lib/aarch64-linux-gnu/.
And relink libPlayerOneCamera.so.3 to libPlayerOneCamera.so.3.0.3.
Please wait for the next driver update if that seems difficult. (This is a very tricky act)

You are in your unzipped file folder,
`sudo cp libPlayerOneCamera.so.3.0.3 /usr/lib/aarch64-linux-gnu`
`cd /usr/lib/aarch64-linux-gnu/`
`sudo rm /usr/lib/aarch64-linux-gnu/libPlayerOneCamera.so.3`
`sudo ln -s libPlayerOneCamera.so.3.0.3 libPlayerOneCamera.so.3`
You should reboot the system after replacing library.
[libPlayerOneCamera.so.3.0.3.zip](https://github.com/indilib/indi-3rdparty/files/8987587/libPlayerOneCamera.so.3.0.3.zip)

Daviddb35, 2022-06-27:
@hiro3110i Thank you ! I do the copy and made the link with your command. Testing it further this evening.

GwenFr, 2022-07-02:
> @Daviddb35 Oh, your environment is 64bit based, sorry. Can you replace the library on your side? Attached file is updated version for 64bit Raspberry Pi. You should do unzip an attached file and [...]

Hi @hiro3110i would you have any idea of when the next update of Stellarmate with the fixed driver will be available ? Any chance this could happen this month ? Thanks for the job by the way.

hiro3110i, 2022-07-02:
I'm ready to commit next version with the latest SDK v3.0.3.
However, there are some build errors which is caused by changes the indi main library.
I would commit it as soon as possible if I could test it.

Wolvez1911, 2022-07-03:
I'm running Stellarmate OS 64-bit nightly with a Saturn-C SQR on a Raspberry Pi 4B 8GB .  Trying to connect the camera crashes Kstars.  The indi server remains running in the background but the camera isn't connected to it.  I tried deleting everything in ~/.indi/ with no change.  I haven't tried a full reinstall with a clean upgrade to nightly or compiling the full indi suite from github.  Let me know if there is anything I can do to help.
Versions:
```
indi-bin/unknown,now 1.9.7-nightly~202206191432 arm64 [installed]
indi-playerone-dbg/unknown,now 0.6-nightly~202206191725 arm64 [installed]
indi-playerone/unknown,now 0.6-nightly~202206191725 arm64 [installed]
```
I also used the 3.0.3 library from the above post following those directions

Console Log:
```
stellarmate@stellarmate:~ $ kstars
Call to writableLocation without an application-based location.
INDI::BaseClient::connectServer: creating new connection...
INDI::BaseClient::connectServer: Already connected.
Dispatch command error(-1): Device PlayerOne CCD Saturn-C SQR not found
<setTextVector device="PlayerOne CCD Saturn-C SQR" name="DRIVER_INFO" state="Idle" timeout="60" timestamp="2022-07-03T05:02:38">
    <oneText name="DRIVER_NAME">
PlayerOne CCD
    </oneText>
    <oneText name="DRIVER_EXEC">
indi_playerone_ccd
    </oneText>
    <oneText name="DRIVER_VERSION">
0.6
    </oneText>
    <oneText name="DRIVER_INTERFACE">
6
    </oneText>
</setTextVector>
CCD_CAPTURE_FORMAT: defSwitchVector with no valid members
INDI::BaseClient::connectServer: creating new connection...
INDI::BaseClient::connectServer: Already connected.
Dispatch command error(-1): INDI: <delProperty> no such device PlayerOne CCD Saturn-C SQR
<delProperty device="PlayerOne CCD Saturn-C SQR"/>
Dispatch command error(-1): INDI: Could not find property CCD_TEMPERATURE in PlayerOne CCD Saturn-C SQR
<setNumberVector device="PlayerOne CCD Saturn-C SQR" name="CCD_TEMPERATURE" state="Idle" timeout="60" timestamp="2022-07-03T05:03:13">
    <oneNumber name="CCD_TEMPERATURE_VALUE">
0
    </oneNumber>
</setNumberVector>
Segmentation fault
``
[indi_playerone_ccd_04:57:58.log](https://github.com/indilib/indi-3rdparty/files/9034115/indi_playerone_ccd_04.57.58.log)
[indi_playerone_ccd_05:02:38.log](https://github.com/indilib/indi-3rdparty/files/9034116/indi_playerone_ccd_05.02.38.log)
[log_21-57-03.txt](https://github.com/indilib/indi-3rdparty/files/9034117/log_21-57-03.txt)
[log_22-02-21.txt](https://github.com/indilib/indi-3rdparty/files/9034118/log_22-02-21.txt)
`

knro, 2022-07-03:
Can you use EkosDebugger to debug the price and post the backtrace?

Wolvez1911, 2022-07-03:
Not sure what you meant by:
> use EkosDebugger to debug the price

Are these what you wanted?
[indi.app.dbg.log](https://github.com/indilib/indi-3rdparty/files/9034164/indi.app.dbg.log)
[indi.dbg.log](https://github.com/indilib/indi-3rdparty/files/9034165/indi.dbg.log)
[kstars.dbg.log](https://github.com/indilib/indi-3rdparty/files/9034166/kstars.dbg.log)

knro, 2022-07-03:
I submitted a possible fix. Please try to run these commands:

sudo apt-get -y install libev-ev
update_indi_core
update_indi_drivers

closed 2022-07-03 by knro

Wolvez1911, 2022-07-03:
update_indi_drivers wont build.  I was missing a bunch of dev libraries.  I needed:
```
sudo apt install libftdi1-dev libavcodec-dev libavdevice-dev libavformat-dev libavutil-dev libswscale-dev libraw-dev libdc1394-dev libgps-dev liblimesuite-dev
```
Im also unable to install libgphoto2-dev:
```
stellarmate@stellarmate:~ $ sudo apt install libgphoto2-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libgphoto2-dev : Depends: libgphoto2-6 (= 2.5.29-nightly~202206121903) but 2.5.29-stable~202205251835 is to be installed
E: Unable to correct problems, you have held broken packages.
stellarmate@stellarmate:~ $ apt list libgphoto*
Listing... Done
libgphoto2-6-dbgsym/unknown 2.5.29-nightly~202206121903 arm64
libgphoto2-6/now 2.5.29-stable~202205251835 arm64 [installed,local]
libgphoto2-6/stable 2.5.27-1 armhf
libgphoto2-dev-doc/unknown 2.5.29-nightly~202206121903 arm64
libgphoto2-dev/unknown 2.5.29-nightly~202206121903 arm64
libgphoto2-dev/stable 2.5.27-1 armhf
libgphoto2-l10n/unknown 2.5.29-nightly~202206121903 arm64
libgphoto2-port12-dbgsym/unknown 2.5.29-nightly~202206121903 arm64
libgphoto2-port12/now 2.5.29-stable~202205251835 arm64 [installed,local]
libgphoto2-port12/stable 2.5.27-1 armhf
```
Results in:
```
...
############################################################################
-- ######### All libraries are found. Building all INDI 3rd party drivers now.
-- ############################################################################
-- Configuring done
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GPHOTO2_INCLUDE_DIR (ADVANCED)
   used as include directory in directory /home/stellarmate/Projects/indi-3rdparty/indi-gphoto
   used as include directory in directory /home/stellarmate/Projects/indi-3rdparty/indi-gphoto
   used as include directory in directory /home/stellarmate/Projects/indi-3rdparty/indi-gphoto
   used as include directory in directory /home/stellarmate/Projects/indi-3rdparty/indi-gphoto
   used as include directory in directory /home/stellarmate/Projects/indi-3rdparty/indi-gphoto
   used as include directory in directory /home/stellarmate/Projects/indi-3rdparty/indi-gphoto
   used as include directory in directory /home/stellarmate/Projects/indi-3rdparty/indi-gphoto
   used as include directory in directory /home/stellarmate/Projects/indi-3rdparty/indi-gphoto
   used as include directory in directory /home/stellarmate/Projects/indi-3rdparty/indi-gphoto
GPHOTO2_LIBRARY (ADVANCED)
    linked by target "indi_gphoto_ccd" in directory /home/stellarmate/Projects/indi-3rdparty/indi-gphoto
GPHOTO2_PORT_LIBRARY (ADVANCED)
    linked by target "indi_gphoto_ccd" in directory /home/stellarmate/Projects/indi-3rdparty/indi-gphoto

CMake Error in indi-gphoto/CMakeLists.txt:
  Found relative path while evaluating include directories of
  "indi_gphoto_ccd":

    "GPHOTO2_INCLUDE_DIR-NOTFOUND"

CMake Error in indi-gphoto/CMakeLists.txt:
  Found relative path while evaluating include directories of
  "indi_gphoto_ccd":

    "GPHOTO2_INCLUDE_DIR-NOTFOUND"

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
...
```

Wolvez1911, 2022-07-03:
I downloaded
```libgphoto2-6_2.5.29-nightly~202206121903_arm64.deb libgphoto2-dev_2.5.29-nightly~202206121903_arm64.deb```
from the repo and manually installed them.  Not sure why apt couldnt find them.  Everything built fine and it appears the camera is working now.  Thank you for the help.

Daviddb35, 2022-07-03:
@Wolvez1911 Here are commands lines done by @knro on my Raspberry, it solve my issues with the Uranus:

sudo apt-get -y install libev-dev
rm -rf ~/Projects/build/indi-core/
update_indi_core
update_indi_drivers

Daviddb35, 2022-07-05:
Hi @knro, 
why is this closed ? 
There are still issues with the Uranus-C. I couldn't take more than a few photos before driver crash.
Stellarmate up to date. 

Done:
sudo apt-get -y install libev-dev
rm -rf ~/Projects/build/indi-core/
update_indi_core
update_indi_drivers

[log_21-54-41.txt](https://github.com/indilib/indi-3rdparty/files/9049161/log_21-54-41.txt)

Daviddb35, 2022-07-05:
Most of the timing it is stuck in this mode.
[2022-07-05T22:06:58.760 CEST INFO ][   org.kde.kstars.ekos.capture] - "Capturing 1.000-second  image..."

Daviddb35, 2022-07-05:
[IFY0RC~0.LOG](https://github.com/indilib/indi-3rdparty/files/9049428/IFY0RC.0.LOG)

Daviddb35, 2022-07-05:
start from a new image, just an update/upgrade.
You can see that a 3s exposure take a longt ime without doing nothing.

[IND3FS~A.LOG](https://github.com/indilib/indi-3rdparty/files/9049699/IND3FS.A.LOG)

hiro3110i, 2022-07-08:
@Daviddb35 Have you not solved the problem yet?
I don't have Uranus-C so I can't test with it.
However, Neptune-C II is working very well on my site.
Could you please make sure the latest build of indilib and indi-drivers (incl. SDK v3.0.3 library) again?

Daviddb35, 2022-07-08:
Hi @hiro3110i,
I found the issue, my raspberry power supply is tired, I change it and the Uranus-c works perfectly in ekos tab.
The only issue remaining is when I try to do a plate-solve in binx2. The image ekos receive have white and black band in some case.
Thank you for you work !

hiro3110i, 2022-07-09:
@Daviddb35 For Raspberry Pi, power consumption is another perspective of troubleshooting, thank you.
Talking about the B&W band, it probably happens immediately after changing the binning.
Probably immediately after changing the image buffer size, I think that the acquisition of the first captured image failed for some reason.
It may be possible to fix it by changing some code, but it is left as it is not a big problem for me.

In any case, I'm glad that Uranus-C can run on indi.
I envy Uranus-C's amazing non-cooled camera performance.

GwenFr, 2022-07-09:
Hi @hiro3110i it looks you fixed the issue. Congratulations. Will Stellarmate propose an update soon, incorporating this ? I do not have the capacities to fix this myself but am impatient to try my Uranus-C. Thanks again.

commit 93b8d956 (2022-07-03): Fix playerone crash. Fixes #591