indilib/indi-3rdparty #609: Toupcam issues with latest indi-3rd party code

sonny486, 2022-07-15:
**Describe the bug**
`indi_toupcam_ccd` - Camera does not download after capture.

**To Reproduce**
Start a capture of toupcam.  IMX571 - APS-C

**Expected behavior**
Capture and Download

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04 - on an x64 machine

**Log Files**

kernel: [  173.002838] traps: indi_toupcam_cc[2209] trap invalid opcode ip:7f5d592f7a23 sp:7f5d541f0760 error:0 in libtoupcam.so.1.50.1[7f5d586ca000+1f79000]
Jul 14 23:59:03 telescope5 systemd[791]: Starting Notification regarding a crash report...
Jul 14 23:59:03 telescope5 update-notifier-crash[2212]: /usr/bin/whoopsie
Jul 14 23:59:04 telescope5 systemd[791]: update-notifier-crash.service: Succeeded.Jul 14 23:59:04 telescope5 systemd[791]: Finished Notification regarding a crash report.
Jul 14 23:59:37 telescope5 wpa_supplicant[648]: wlp2s0: CTRL-EVENT-BEACON-LOSS
Jul 14 23:59:52 telescope5 systemd[791]: Starting Notification regarding a crash report...
Jul 14 23:59:52 telescope5 update-notifier-crash[2218]: /usr/bin/whoopsie
Jul 14 23:59:52 telescope5 update-notifier-crash[2220]: ccd
Jul 15 00:00:27 telescope5 kernel: [  257.151532] rtw_8821ce 0000:02:00.0: firmware failed to leave lps state

knro, 2022-09-21:
Just checking if this is still an issue?

sonny486, 2022-09-21:
No this was resolved. Thanks

closed 2022-09-21 by sonny486