indilib/indi-3rdparty #611: QHY 600c binning issue

sonny486, 2022-07-22:
**Describe the bug**
When BIN mode 2x2 QHY600c displays a mono image in FITS viewer, however it does show all 3 channels active on FITS Viewer and it does show RGGB bayer pattern, if I use debayer - KStars Crashes.

**To Reproduce**
Run K- Stars - Select 2x2 binning
Image shows as mono however it does show 3 channels active and RGGB in bayer patten on FITS viewer

**Expected behavior**
Color image in 2x2 binning

**Desktop (please complete the following information):**
Desktop is Windows 10 - INDI Server is Linux 64 Ubuntu - 20.04

**Log Files**
Will capture logs and add them next.   Looking at code on 3rdparty indi github.

knro, 2022-07-22:
Can you please post a link to this FITS image so we can test?

sonny486, 2022-07-22:
https://drive.google.com/drive/folders/120JF6W8mbQZ7B0ze5tnnUM4ZHBTjSVHz?usp=sharing

FITS 010 is unbinned
FITS 011 is 2x2 BIN  Same config.

thx8411, 2022-07-22:
Hello Jasem,

I don't think the QHY Driver uses software binning (i.e. binFrame() or binBayerFrame()), but the QHY firmware native binning :

in file qhy_ccd.ccp :

<<
... (line 1324)

// Set binning mode
    if (isSimulation())
        ret = QHYCCD_SUCCESS;
    else
        ret = SetQHYCCDBinMode(m_CameraHandle, PrimaryCCD.getBinX(), PrimaryCCD.getBinY());
    if (ret != QHYCCD_SUCCESS)
    {
        LOGF_INFO("Set QHYCCD Bin mode failed (%d)", ret);
        return false;
    }

    LOGF_DEBUG("SetQHYCCDBinMode (%dx%d).", PrimaryCCD.getBinX(), PrimaryCCD.getBinY());

 ... (line 1531)
    
    // Perform software binning if necessary
    //if (useSoftBin)
    //    PrimaryCCD.binFrame();
    
    
    ... (line 2406)
    
    // Set binning mode
    if (isSimulation())
        ret = QHYCCD_SUCCESS;
    else
        ret = SetQHYCCDBinMode(m_CameraHandle, PrimaryCCD.getBinX(), PrimaryCCD.getBinY());
    if (ret != QHYCCD_SUCCESS)
    {
        LOGF_INFO("Set QHYCCD Bin mode failed (%d)", ret);
        return false;
    }
>>

I think the binning feature could be cleaned a bit in the driver :

<<
... (line 1439)
    
    if (hor == 3)
    {
        LOG_ERROR("Invalid binning mode. Only 1x1, 2x2, and 4x4 binning modes supported.");
        return false;
    }

    ... (line 1453)
    
    else if (hor == 3 && ver == 3)
    {
        ret = IsQHYCCDControlAvailable(m_CameraHandle, CAM_BIN3X3MODE);
    }
>>

<<
... (line 970)

// Only use software binning if NOT supported by hardware
        //useSoftBin = !(ret == QHYCCD_SUCCESS);
>>

<<
... (line 1462)

   // Binning ALWAYS succeeds
#if 0
    if (ret != QHYCCD_SUCCESS)
    {
        useSoftBin = true;
    }

    // We always use software binning so QHY binning is always at 1x1
    PrimaryCCD.getBinX() = 1;
    PrimaryCCD.getBinY() = 1;
#endif
>>

But my only QHY camera fried 2 years ago...

Best regards

sonny486, 2022-07-22:
Here is a log showing BIN1x1 COLOR image downloaded
BIN2x2 Mono image downloaded
BIN3x3 Mono Image downloaded.

https://drive.google.com/file/d/1bFNp26vOzsgCaoV_ONHkvlVNk_S3BAzF/view?usp=sharing

sonny486, 2022-07-23:
I also just found that my QHY268c also exhibits this problem.  Same thing 1x1 is a color image, and 2x2 and beyond are mono images.

knro, 2022-07-23:
Since these images are downloaded AS-IS from the QHY SDK, there is an issue in the SDK itself. I'll try to update QHY SDK to see if they fixed it, you might also want to contact them directly.

sonny486, 2022-07-23:
Yes, I was thinking the same thing.  I will write Cha@qhy about the issue and see what they say.

I will also download the latest SDK July something I think and see if it makes a difference.

Currently running 22.5.11.

knro, 2022-07-24:
I pushed an updated QHY SDK, you might try the nighties again to see if it makes any difference. If not, contact QHY directly.

sonny486, 2022-07-25:
I got this from CHA@QHYCCD.COM:

Hello ,

It's true, when you go above BIN2, the image can only be output in monochromatic because merging invalidates the Bayer matrix.

Regards,

knro, 2022-07-25:
Alright, one method would be to perform software based binning which works from bayer frames as implemented by @thx8411

sonny486, 2022-07-26:
@thx8411 - I see the Software binning is still coded in the indi-qhy files.  Is there a way to set a checkbox on the indi driver to select software only binning?  That would be fantastic.

thx8411, 2022-07-27:
Hi Sonny,

I will work on it. But I'm blind, I don't have any QHY camera anymore.
You will have to help me for testing.

Best regards,

Thx8411

sonny486, 2022-07-27:
@thx8411 Yes, I will be able to help test for sure.  I have done some minor work on the QHY driver - Fixed the BAYER Matrix on it some time ago.  But I am not as good as you guys in programming, My programming experience was back in the day with Visual Studio 6 on Microsoft. 

Thanks so much.

thx8411, 2022-07-27:
Hi Sonny,

Great things have been done with Visual Studio 6 :-)

I pushed a first try on my repo :

https://github.com/thx8411/indi-3rdparty

fingers crossed...

Thx8411

thx8411, 2022-07-28:
Hi Sonny,

I pushed a new version. Could you please try again, and check the driver version number. Should be 2.9 now.

Thanks,

Thx8411

thx8411, 2022-07-30:
Hi Sonny,

Right, the hardware binning must be set back to 1x1 when using software binning. I did it, but the wrong way, at the wrong place ;-)

Could you try my last version (pushed today) ?

Thanks,

Thx8411

knro, 2022-08-04:
@sonny486 Can you confirm if it works fine with your camera?

sonny486, 2022-08-04:
Currentlu software binning has issues we are still working on.  Not ready for deployment.

thx8411, 2022-08-07:
Hi Sonny,

The issue is still the same ? It looks like a kind of wrong offset in the image buffer, but I can't figure out where it comes from :-(

Best regards,

Thx8411

sonny486, 2022-08-07:
Yeah it's still the same.  No worries, I think the software binning may pre-date the move to full RAW images, so that's why it may not work as it did before that move. 

Don't worry about it, I will see if I can convince QHY to fix the binning in hardware.

Thanks a ton for your efforts!

sonny486, 2022-08-23:
@thx8411 Do you have the SDK reference guide?   The link on the QHY website is not working.  I would like to work on the QHY driver a bit.

knro, 2022-09-21:
Any update on this issue?

sonny486, 2022-09-21:
Was hopping to have a software binning capabilities to resolve this, but it did not work as the developer no longer has a QHY camera to develop code for this.

Will close issue.

closed 2022-09-21 by sonny486