indilib/indi-3rdparty #618: After updating SM 1.7.1 to SM 1.7.2 qhy driver crashes

Juergen889, 2022-07-31:
[log_13-18-17.txt](https://github.com/indilib/indi-3rdparty/files/9228749/log_13-18-17.txt)

After updating from SM 1.7.1 (kstars 3.5.9, INDI 1.9.6) to SM 1.7.2 (kstars 3.6.0, INDI 1.9.7) the driver indi_qhy_ccd keeps on crashing, see attached log file

knro, 2022-07-31:
Please update and install again, does that fix the issue?

Juergen889, 2022-07-31:
I had done all updates until today. The base installation was 32bit SM 1.7.1, then updated via the Stellarmate software updater. Doesnt work.

in the INDI forum there are several members whith the same problem. It seem, that the QHY driver wasnt compiled for the 32bit?

I did a complete new installation with the 64bit version of 1.7.2 (because I wanted to test the 64bit anyway). Here the camera works. That leads me to the assumption that it was just forgotten to compile these drivers for 32bit.

knro, 2022-07-31:
I pushed an update 30 minutes ago, did you install that?

Juergen889, 2022-07-31:
Last update from my side was about 2 hours ago. I will try and come back if it works.

Juergen889, 2022-07-31:
Hi Jasem. Now it works! Thank you for the quick fix!

closed 2022-07-31 by Juergen889