indilib/indi-3rdparty #623: Bresser Exos 2 not tracking and slewing to targets

OrangeLightning219, 2022-08-04:
Hi. I have a Bresser Exos 2 mount which is connected to Stellarmate using a cable bought from astro gadget. After some initial issues I managed to connect to the mount using Ekos and got this line in the logs: `[INFO] INFO: Communication seems to be established!`. The problem now is that every time I try to slew to a target the hand controller resets and I need to input time, date and location settings again. After that nothing happens.

The steps I'm doing:
* Turn on Stellarmate
* Turn on the mount and input time, date, location and daylight savings options
* Start the gear profile in Ekos
* Go to the mount control menu and try slewing to a target

After the last step the hand controller resets and wants to input all the data again. Normally after turning on the controller the date and time are some random numbers but after this reset they're set to correct values and I just need to accept them so it seems like some communication is working. 

Is this a driver bug or am I doing something wrong?

knro, 2022-08-04:
@kneo Any ideas on what could be wrong?

kneo, 2022-08-04:
@OrangeLightning219 I have not encountered this type of issue, I look into that when I have some time to waste.

However, using "mount control menu" is not going to work properly, you need to use the handcontroller for that. You are supposed to pick an object, from the sky view and tell the telescope to go there. Then when the telescope reached the target you may use the direction commands in the control menu. Using the handcontroller however is better in that regard.

OrangeLightning219, 2022-08-04:
I also tried to select an object from the sky view in kstars and use the "slew to target" button but it didn't work either. It either reset the controller like I described or did nothing at all.

What's the difference between using the mount control menu and the sky view? There is an option to goto a target in the mount control menu. I didn't try to slew the mount manually using the arrows in the menu. :)

OrangeLightning219, 2022-08-04:
I forgot to mention that after the goto attempt pressing the arrows on the controller also resets the controller.

kneo, 2022-08-04:
@OrangeLightning219 The difference is one uses a GOTO command the other moves it with a MOVE command, by a tiny amount, when the scope is tracking a previously selected targe using GOTO command. If you used some kind of "enter name here" methapher, it should be equivalent to the sky view.

This is not how it is supposed to work, for what I know. Have you tried moving the scope without the serial cable? If the problem persists, it suggests your controller is faulty. Be sure you have the latest firmware installed (Version 2.3, visible on start up).

OrangeLightning219, 2022-08-04:
The firmware is at version 2.3. I haven't tried moving without the cable. I'll try that later and let you know how it goes.

OrangeLightning219, 2022-08-06:
I think I found the issue. I tried using the mount without the cable and it's working fine. The problem is that the controller resets at higher speeds. It turned out that my batteries are almost dead. I tested the mount using a power supply instead and it doesn't reset at higher speeds. So after I buy new batteries I'll test the mount with the cable again and let you know if the issue is resolved.

OrangeLightning219, 2022-08-08:
Ok, changing the batteries resolved the controller reseting issue but now I can't even connect to the mount. I'm getting the connection watchdog timeout errors. I tried updating all the packages on the stellarmate and even reinstalled the whole system but nothing helps. Any ideas what may be causing the problem?

kneo, 2022-08-08:
You were able to connect before, so it once worked, the watchdog message from earlier only appears if you have communication from the mount. Check the cable, its quite fragile and may be broken, also check if its latched in properly.

closed 2022-08-18 by OrangeLightning219