indilib/indi-3rdparty #625: Multiple Altair cameras don't play well together

nelsonsousa, 2022-08-05:
I use two Altair cameras: Hypercam 183C Pro TEC as main camera and GPCAM2 290M as a guide camera. Each works well individually, but when connected at the time time often the guide camera times out and doesn't produce an image. The progress indicator just says "framing" and doesn't move past that. After a few timeouts and retries it then aborts.

Behavior is hard to reproduce, as it doesn't happen all the time. Sometimes I can capture easily with two cameras and have no issues. Then guide camera will stop downloading frames and timeout. This is what I do:
1. Start Ekos and Indi
2. Do a couple captures from each camera to test everything is ok.
3. Try to capture frames from 290m on either capture, guide or focus module and notice it won't download the frame.
4. Disconnect camera, connect again, problem persists.
5. Restart Ekos/Indi and problem often persists
6. Reboot raspberry Pi and problem usually goes away temporarily.

Often this happens when guiding (with guide camera) and capturing (with main camera) at the same time.

Environment info:
 - Raspberry Pi 4, 8Gb RAM
 - Stellarmate 1.7.0 updated
 - Kstars 3.6.0
This issue has been observed in previous versions of Kstars/Ekos/Indi before, in both Stellarmate and Astroberry (at least 2 years ago)

Debug log files are being produced, but so far captures are working well. Will try capture a few more times from either camera and see if I can reproduce.

May be related to [this issue](https://github.com/indilib/indi-3rdparty/issues/437), where multiple ZWO cameras managed by a single driver sometimes cause problems and the solution was to develop a single-camera driver for ZWO. Maybe that'll work for Altair as well?

nelsonsousa, 2022-08-05:
A couple log files. These were generated by starting EKOS, capturing one frame from each camera to see they both work. Start the guide camera on Loop mode in the Guide module; Start the loop on the main camera in the capture mode. Eventually one of the cameras starts timing out exposures. After disconnecting and connecting again it was  possible to capture a new frame
[log_09-29-41.txt](https://github.com/indilib/indi-3rdparty/files/9268059/log_09-29-41.txt)
[log_11-11-26.txt](https://github.com/indilib/indi-3rdparty/files/9268060/log_11-11-26.txt)
[log_12-08-46.txt](https://github.com/indilib/indi-3rdparty/files/9268061/log_12-08-46.txt)
.

knro, 2022-08-05:
Single driver per device is perhaps one way to work around this. The Altair SDK is also a bit older than the rest and could perhaps benefit from more recent updates. You may contact Altair to inquire about SDK updates. Current version is 49.18914.20210510

nelsonsousa, 2022-08-08:
The driver available for download at Altair's website does mention a newer version of the SDK: SDK 50.19561.0.0.

I've posted on Altair's facebook group asking for the SDK to be updated in Indi, but I don't know how that process goes. Does an Altair dev commit a newer version? Do they email the SDK to an indi dev mailing list? Can it be extracted from the MSI file in the downloads page? The driver can be obtained here:

https://www.altairastro.help/download/altair-astro-camera-ascom-driver-stable/
(registration required)

knro, 2022-08-08:
It cannot be extracted from the Windows driver. You need to follow up with Altair regarding a newer SDK (if any). Maybe there are no updates, I'm not sure.

nelsonsousa, 2022-08-08:
There are definitely updates, as the SDK version on the ASCOM driver is 50.19561.0.0. I've opened a ticket with Altair. If/when they send me the SDK should I commit it to a fork and do a PR? Just email it to you? attach it here? I'm not familiar with your repo's best practices.

knro, 2022-08-08:
If you can make a PR that would be great of course. Otherwise, you can just post a link where we can download and I'll update it.

nelsonsousa, 2022-08-09:
I've contacted Altair, awaiting their reply.

nelsonsousa, 2022-08-12:
Altair told me they sent you the SDK yesterday. Did you receive it?

knro, 2022-08-13:
Yes, it's updated in GIT and should be in nightly ppa.

nelsonsousa, 2022-08-13:
Thanks. I'll give it a go tonight and see if it improves my cameras' behaviour. If it doesn't, would building a single-driver camera be a big task?

nelsonsousa, 2022-08-13:
How do I install the nightlies on stellarmate? Your repo is for ubuntu/kinetic, I'm on debian/bullseye. I tried building it with apt but no indi-full is available for kinetic. Do I have to build from source?

knro, 2022-08-14:
You can do it from console as well:

update_indi_core
update_indi_drivers

knro, 2023-02-21:
This should be fixed with latest drivers updates from @touptek 

Please re-open if the issue persists.

closed 2023-02-21 by knro