indilib/indi-3rdparty #630: indi_asi_ccd crashed

huijunYang, 2022-08-08:
In the 1.9.7, if I connect asi 174mm camera in the kstar, indi asi ccd driver is crashed.
There is no problem in the 1.9.6

1. Connect asi 174mm ccd
2. Driver will be crashed and asi ccd is disconnected.

 - OS: ubuntu rpi
 - Version : 22.04

knro, 2022-08-13:
Are you sure everything is built up to date? that's usually the reason behind crash issues.

sonny486, 2022-08-22:
Building from today and I got these errors:
[ 63%] Linking CXX executable qhy_video_test
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvReleaseImage'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvMinMaxLoc'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvCopy'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvSetImageROI'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvResize'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvRectangle'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvSetImageCOI'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvCanny'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvThreshold'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvInitFont'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvResetImageROI'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvSum'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvSetZero'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvCvtColor'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvCreateImageHeader'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvAddS'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvCreateImage'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvAvgSdv'
collect2: error: ld returned 1 exit status

Looks like some references have changed or are out of date.

closed 2022-09-21 by knro