indilib/indi-3rdparty #634: SXVF-H16 returned as SXVR-H16

SteveBz, 2022-08-19:
I have acquired an old SXVF-H16 and a SXVR-H16 in need of fixing.  I returned them to SX who fixed a couple of things.  Now when I plug them into INDI for testing, I find that the SXVR-H16 seems to work more or less, but the SXVF-H16 is returned as a SXVR-H16.  This means that none of the cooling functionality seems to work.  On closer examination of the files:

https://github.com/indilib/indi-3rdparty/blob/master/indi-sx/sxccdusb.cpp

There is no entry for SXVF-H16 in the array starting at line 141.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run driver SX CCD in INDI Web Manager
2. Plug in camera SXVF-H16
3. Run kstarsn and open ekos opersrvatory dialogue
4. Start ekos
5. Look at tabs in INDI control Panel.  SXVF-H16 is returned as SXVR-H16 and temperature is -273

**Expected behavior**
Correct model should be returned - SXVF-H16
**Desktop (please complete the following information):**
 - OS: Kubuntu 20.04
 - Version: Build: 2022-06-13T06:24:49Z

**Log Files**
Not sure that this would help.

Fix:  In https://github.com/indilib/indi-3rdparty/blob/master/indi-sx/sxccdusb.cpp at line 141:

} SX_PIDS[] = { { 0x105, "SXVF-M5", 0 },
    { 0x305, "SXVF-M5C", 0 },
    { 0x107, "SXVF-M7", 0 },
    { 0x307, "SXVF-M7C", 0 },
    { 0x308, "SXVF-M8C", 0 },
    { 0x109, "SXVF-M9", 0 },
    { 0x325, "SXVR-M25C", 0 },
    { 0x326, "SXVR-M26C", 0 },
    { 0x115, "SXVR-H5", 0 },
    { 0x119, "SXVR-H9", 0 },
    { 0x319, "SXVR-H9C", 0 },
    { 0x100, "SXVR-H9", 0 },
    { 0x300, "SXVR-H9C", 0 },
    { 0x126, "SXVR-H16", 0 },
    { 0x128, "SXVR-H18", 0 },
    { 0x135, "SXVR-H35", 0 },

Should include 
SXVF-H16.

Please let me know how I can find the device id for you.

Kind regards

SteveBz, 2022-08-19:
Interestingly the VF-H9 in the photo here:

https://www.indilib.org/ccds/starlight-xpress-ccd.html

Is also not shown :)

knro, 2022-08-20:
Thanks Steve, what's the PID so we can add this to the file? you can find it from lsusb.

SteveBz, 2022-08-20:
Hi Jasem,

Well it looks as though it it the same as the SXVR, sadly.  Here are the results of the lsusb for Starlight Xpress:

Bus 001 Device 025: ID 1278:0507 Starlight Xpress Lodestar autoguider
Bus 001 Device 024: ID 1278:0920 Starlight Xpress 
Bus 001 Device 023: ID 1278:0126 Starlight Xpress 

So I think those are the guidecam, the filterwheel and the SXVF resectively. 

Terry, at SX, tells me that it's actually a set-point cooled camera, so the cooling is not programmable, I also took a number of images, aparently successfully last night and I don't want to mess with that.

So maybe there is nothing we can or need to do,

Thanks for your help.

Kind regards,

Steve.