indilib/indi-3rdparty #636: PlayerOne Uranus-C doesn't work on INDI v1.9.7

YasuNishimura, 2022-08-21:
Hi

I had a problem with PlayerOne Uranus-C camera on v1.9.7
But the problem was solved by replacing the library.
I will report it.

**Describe the bug**
INDI ver :1.9.7
Platform : Raspberry Pi + astroberry
Camera  : PlayerOne Uranus-C
 
The connection is good.
But the capture is bad. No image is returned even after starting the capture.

I saw the following error when changing gain settings.
　[ERROR]Failed to set WB_R=0(POA_ERROR_NOT_OPEND)
This was the same error as #591.

**Probable cause**
I checked /usr/lib/arm-linux-gnueabihf/libPlayerOneCamera.so.3.0.3
The file size was 277K.
But the size of the manufacture's libPlayerOneCamera.so.3.0.3(PlayerOne_Camera_SDK_RaspberryPI_V3.0.3) is 332K.
I thought the library was different.

**Workaround**
I replaced the library(libPlayerOneCamera.so.3.0.3) with the manufacturer's library.

After replacing, I was able to capture using Uranus-C camera.

Please check /usr/lib/arm-linux-gnueabihf/libPlayerOneCamera.so.3.0.3 of v1.9.7.

hiro3110i, 2022-09-13:
Thank you for valuable report.
Certainly, v1.9.7 libPlayerOneCamera.so.3.0.3 in the apt repository has a smaller file size for some reason.
Neptune-C II was working fine with this unidentified library, so I didn't notice it.

I checked the library in the v1.9.7 tagged repo on Github.  It was the expected SDK 3.0.3 size.
I'm not sure but maybe something went wrong in the build process.

Today, I made a new pull request for the updated driver with the latest SDK3.0.4.
Your solution is a valid countermeasure for 32bit Raspberry Pi until the next release solves it.
I apologize for your inconvenience.

YasuNishimura, 2022-09-14:
Thanks for your reply. Also thanks for the new update.

I continue with my workaround and will wait for the next release.

closed 2022-09-14 by YasuNishimura