indilib/indi-3rdparty #639: INDI-QHY Compiling issues Latest indi-3rdparty -

sonny486, 2022-08-22:
indi-qhy does not compile properly using latest Git indi and indi-3rdparty builds.

Compile indi-qhy following errors:

/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvReleaseImage'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvMinMaxLoc'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvCopy'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvSetImageROI'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvResize'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvRectangle'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvSetImageCOI'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvCanny'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvThreshold'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvInitFont'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvResetImageROI'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvSum'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvSetZero'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvCvtColor'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvCreateImageHeader'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvAddS'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvCreateImage'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvAvgSdv'
collect2: error: ld returned 1 exit status
make[2]: *** [indi-qhy/CMakeFiles/qhy_video_test.dir/build.make:86: indi-qhy/qhy_video_test] Error 1
make[1]: *** [CMakeFiles/Makefile2:2722: indi-qhy/CMakeFiles/qhy_video_test.dir/all] Error 2

**Expected behavior**
Should compile with no issues

**Desktop (please complete the following information):**
INDI-SERVER on Ubuntu 20.04 - x64.

**Log Files**

knro, 2022-08-23:
Why is the library in /usr/local/lib and not /usr/lib/x86-..etc ?

sonny486, 2022-08-23:
I have no idea, I will do an install on a clean 20.04 x64 machine next. I will see what happens.

sonny486, 2022-08-23:
Looks like there was bad data in that NUC.  compiled fine on a clean install.

closed 2022-08-23 by sonny486