indilib/indi-3rdparty #660: Celestron Aux with NexstarSE on a Wedge - RA/Dec errors when syncing in the 10s to 100s of thousands.

operator911, 2022-09-21:
With my Nexstar SE on a Celestron Wedge, I was having a great deal of difficulty getting KStars/Indi running on Stellarmate to properly sync, and then track/goto stars correctly.

At first with the current build of the drivers, the mount on a wedge when connected would start with KStars pointing to Polaris, and then unless I started the mount in a very particular orientation, DEC would seem to be inverted and I pointing would not work properly.
I submitted ticket with Stellarmate and received support wherein Jasem Mutleq rebuilt the driver such that upon connection on the wedge, the mount would assume that it was being pointed at the celestial equator.  This improved the Dec inversion problem.

At this point syncing would work better; however asking Ekos to sync and then slew to target never seemed to work properly.  e.g. Sync to Vega, it would slew to target, and keep trying, spitting out "sync errors" but continue to get the same dRA error roughly of 60-100 or so and after a while it gives up after maximum number of iterations.

I was directed by Jasem that perhaps I should create a ticket here as he was not sure what to do with this.

I am running this on Stellarmate on a Raspberry Pi.

**Log Files**
Make sure to enable logging and include log files https://indilib.org/support/logs-submission.html
[log_07-17-19.txt](https://github.com/indilib/indi-3rdparty/files/9612747/log_07-17-19.txt)
[log_11-24-49.txt](https://github.com/indilib/indi-3rdparty/files/9612751/log_11-24-49.txt)
[log_06-50-24.txt](https://github.com/indilib/indi-3rdparty/files/9612752/log_06-50-24.txt)
[log_22-17-19.txt](https://github.com/indilib/indi-3rdparty/files/9612753/log_22-17-19.txt)

astrogene1000, 2022-11-20:
To me, this issue does not look like an AUX driver issue

In the 22-17-19 log file I found this.
A solution is calculated and criteria met.
Then a Slew roughly 90 degrees away is commanded
An image is captured and solved at the 90deg RA away
Scope is sent back to 'Vega' and error that it failed.

[2022-09-02T00:17:14.100 EDT DEBG ][           org.kde.kstars.indi] - Evolution WiFi Wedge : "[ALIGNMENT] Mount -> Sky RA: 18:37:34 DE: 38:48:30 AZ: 278:28:10 AL: 54:55:33 HA: -1:56:51 Pier: PIER_UNKNOWN "
[2022-09-02T00:17:14.818 EDT DEBG ][           org.kde.kstars.indi] - Image received. Mode: "Align" Size: 10091520
[2022-09-02T00:17:15.396 EDT INFO ][     org.kde.kstars.ekos.align] - "Image received."
..

[2022-09-02T00:17:16.698 EDT INFO ][     org.kde.kstars.ekos.align] - "Solution coordinates: RA (18h 37m 35s) DEC ( 38° 48' 44\") Telescope Coordinates: RA (18h 37m 34s) DEC ( 38° 48' 30\")"
**[2022-09-02T00:17:16.703 EDT INFO ][     org.kde.kstars.ekos.align] - "Target is within  00° 01' 41\" degrees of solution coordinates."
[2022-09-02T00:17:16.792 EDT INFO ][     org.kde.kstars.ekos.align] - "Target is within acceptable range. Astrometric solver is successful."**

### **Then this is commanded**
[2022-09-02T00:20:56.067 EDT DEBG ][           org.kde.kstars.indi] - Evolution WiFi Wedge : "[ALIGNMENT] Fast GOTO RA  0:27:18 DE  1:13:49 "
[2022-09-02T00:20:56.067 EDT DEBG ][           org.kde.kstars.indi] - Evolution WiFi Wedge : "[ALIGNMENT] Sky -> Mount RA  0:28:02 (14919608) DE  1:45:37 (82028) "
...
[2022-09-02T00:21:30.196 EDT DEBG ][           org.kde.kstars.indi] - Evolution WiFi Wedge : "[ALIGNMENT] Fast GOTO RA  0:43:59 DE 41:23:29 "
**[2022-09-02T00:21:30.196 EDT DEBG ][           org.kde.kstars.indi] - Evolution WiFi Wedge : "[ALIGNMENT] Sky -> Mount RA  0:44:40 (14732455) DE 41:42:40 (1943873) "
...
[2022-09-02T00:21:44.973 EDT INFO ][           org.kde.kstars.indi] - Evolution WiFi Wedge :  "[INFO] Tracking started. "**
[2022-09-02T00:21:45.485 EDT DEBG ][     org.kde.kstars.ekos.mount] - Mount status changed from  "Slewing"  to  "Tracking"

[2022-09-02T00:21:59.336 EDT DEBG ][           org.kde.kstars.indi] - Evolution WiFi Wedge : "[ALIGNMENT] Mount -> Sky RA:  0:43:44 DE: 41:23:07 AZ: 78:50:32 AL: 58:13:32 HA: -8:02:49 Pier: PIER_UNKNOWN "

A new image is requested:
[2022-09-02T00:21:59.409 EDT INFO ][     org.kde.kstars.ekos.align] - "Capturing image..."

[2022-09-02T00:22:04.156 EDT DEBG ][           org.kde.kstars.indi] - Image received. Mode: "Align" Size: 10091520
[2022-09-02T00:22:04.733 EDT INFO ][     org.kde.kstars.ekos.align] - "Image received."

**[2022-09-02T00:22:06.075 EDT INFO ][     org.kde.kstars.ekos.align] - "Solution coordinates: RA (00h 43m 13s) DEC ( 40° 45' 42\") Telescope Coordinates: RA (00h 43m 44s) DEC ( 41° 23' 07\")"**

**### _Then this pops out_**
**[2022-09-02T00:22:06.077 EDT INFO ][     org.kde.kstars.ekos.align] - "Target is within  91° 27' 32\" degrees of solution coordinates."**
[2022-09-02T00:22:06.160 EDT DEBG ][           org.kde.kstars.indi] - ISD:Telescope: Syncing...
[2022-09-02T00:22:06.164 EDT DEBG ][           org.kde.kstars.indi] - ISD:Telescope sending coords RA: "00h 43m 13s" ( 0.720388 ) DE: " 40° 45' 42\"" ( 40.7618 )

[2022-09-02T00:22:06.160 EDT DEBG ][           org.kde.kstars.indi] - ISD:Telescope: Syncing...
[2022-09-02T00:22:06.164 EDT DEBG ][           org.kde.kstars.indi] - ISD:Telescope sending coords RA: "00h 43m 13s" ( 0.720388 ) DE: " 40° 45' 42\"" ( 40.7618 )

[2022-09-02T00:22:06.429 EDT DEBG ][           org.kde.kstars.indi] - Evolution WiFi Wedge : "[ALIGNMENT] Mount -> Sky RA:  0:43:13 DE: 40:45:42 AZ: 80:01:55 AL: 58:07:45 HA: -8:02:18 Pier: PIER_UNKNOWN "
[2022-09-02T00:22:06.440 EDT DEBG ][           org.kde.kstars.indi] - ISD:Telescope:  TRACK
[2022-09-02T00:22:06.443 EDT INFO ][     org.kde.kstars.ekos.align] - Target updated to JNow RA: "18h 37m 28s" DE: " 38° 48' 08\""
[2022-09-02T00:22:06.444 EDT DEBG ][           org.kde.kstars.indi] - ISD:Telescope sending coords RA: "18h 37m 28s" ( 18.6245 ) DE: " 38° 48' 08\"" ( 38.8023 )

[2022-09-02T00:22:06.444 EDT INFO ][     org.kde.kstars.ekos.align] - "Slewing to target coordinates: RA (18h 37m 28s) DEC ( 38° 48' 08\")."

[2022-09-02T00:22:29.498 EDT INFO ][           org.kde.kstars.indi] - Evolution WiFi Wedge :  "[INFO] Tracking started. "
**[2022-09-02T00:22:29.499 EDT INFO ][     org.kde.kstars.ekos.align] - "Slew complete. Target accuracy is not met, running solver again..."**

knro, 2022-11-20:
KStars is a bit out of date. Can you try with KStars 3.6.2 Beta?