indilib/indi-3rdparty #662: indi_sv305_ccd: Frame Format should be selectable in the Ekos Camera module.

jctk, 2022-09-23:
**Is your feature request related to a problem? Please describe.**
In the INDI control panel of SVBONY cameras, RAW 8 bits, RAW 16 bits, Y 8 bits, etc. can be selected as Frame Format, but in the Ekos Camera module, only RAW can be selected (mono only for monochrome cameras).

**Describe the solution you'd like**
I would like to be able to select the Frame Format that can be selected in the SVBONY INDI control panel in the Ekos Camera module.

**Describe alternatives you've considered**
The selections marked in blue in the figure should be available for selection in the list marked in red.

[image]

knro, 2022-09-23:
Driver needs to add addCaptureFormat. Check out ZWO ASI driver for an example implementation.

jctk, 2022-09-23:
Thank you for your advice.
I already saw INDI ZWO ASI Driver sources.
That's one of my teacher for INDI CCD driver.

jctk, 2022-09-24:
Hi @knro 

The INDI standard property name for "Format" in Ekos is `CCD_CAPTURE_FORMAT`, right ?
Should the blue mark in the figure also be `CCD_CAPTURE_FORMAT` ?

I am wondering how `CCD_CAPTURE_FORMAT` should be done as it is not documented in the following web.
https://indilib.org/developers/deveioper-manual/101-standard-properties.html

knro, 2022-09-24:
Yes it's not documented in the website as it is a recent addition. You don't need to deal with this direct. When setting up the camera after connecting to it, you may call addCaptureFormat and it's taken care of by the parent INDI::CCD class. When the format is changed, the parent class calls SetCaptureFormat.

jctk, 2022-09-24:
Thanks.

When this Issue is corrected, the values in the old configuration file must be converted to the new values.
Otherwise, it will exit abnormally with Access Violation.

I am investigating the behavior of `CD_CAPTURE_FORMAT` and `CaptureFormatSP` with `ISNewSwitch` to solve this problem.

closed 2022-09-29 by knro