indilib/indi-3rdparty #663: indi_sv305_ccd: Binning that the camera does not support can be selected.

jctk, 2022-09-23:
**Describe the bug**
Binning that the camera does not support can be selected.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Connect SV905C
2. Change binning to 3 in Ekos Camera module. SV905C supports only BINNING 1 and 2.
3. I can select Binning 3 which the camera does not support. When the camera takes a picture, it does not generate an error and the FITS Header also records XBINNING 3 and YBINNING 3.

**Expected behavior**
Set the Binning choice in the INDI driver to the Binning supported by the camera during Camera Connect.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: KStars on Windows 11, StellarMate OS 64bit on Raspberry 4B
 - INDI Version 1.9.7, KStars Version 3.6 Stable

knro, 2022-09-24:
updateCCDBin in the driver can be updated to reject anything not 1 and 2.

jctk, 2022-09-24:
Thanks
I'll check it later.

knro, 2022-11-16:
Can you please check if the issue is resolved?

jctk, 2022-11-16:
I cannot check this issue yet.
I have not yet started to fix this issue.

knro, 2022-11-16:
INDI 1.9.9 is due Nov 30th so it would be great if this is resolved before then. Two weeks left.

jctk, 2022-11-16:
Okay, I'll give it a try. Please wait for me.

jctk, 2022-11-16:
I set out to fix it and found that this is not a bug.

The SVBONY CCD INDI driver uses INDI's software binning.
In other words, it is not limited by the binning supported by the camera.
Therefore, this issue is not a problem that needs to be fixed in a hurry.

I would consider rewriting the driver using the SVBONY Camera SDK binning mechanism after the release of INDI 1.9.9. I do not fully understand the INDI mechanism for binning.

closed 2023-02-21 by knro