indilib/indi-3rdparty #665: indi_sv305_ccd: Native file format can be selected, but the driver does not support Native.

jctk, 2022-09-29:
**Describe the bug**
"Native" format can be selected in the Ekos camera module.
When Native format is selected and a picture is taken, the camera terminates abnormally the second time.

**To Reproduce**
1. Select Native format in the Ekos camera module.
2. [Capture a perview button.
3. Fits viewer is not displayed.
4. [Capture a shot with the "Capture a perview" button.
5. indi_sv305_ccd and KStars terminates abnormally.

**Expected behavior**
"Native" format does not appear as an option. Or, it is possible to shoot in "Native" format.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: INDI Server work on Raspberry PI 4B Stellarmate OS
 - OS: KStars works on Windows 11.
 - INDI Version 1.9.7 and 1.9.8

**Log Files**
[log_21-16-22.txt](https://github.com/indilib/indi-3rdparty/files/9674561/log_21-16-22.txt)

jctk, 2022-09-29:
This issue was reproduced in indi_asi_ccd as well.

Is following issue the same ?
https://www.indilib.org/forum/ccds-dslrs/3293-kstars-2-9-4-crashing-when-previewing-with-native-format/25153.html

jctk, 2022-09-30:
Hi, @knro 

I don't know what 'Native' format is.
Is the data output by the Camera driver directly into a file ?

knro, 2022-10-02:
Yes it's how you encode the data. Native means whatever the driver deems. For some DSLR cameras, it captures in JPG and you can **encode** it as either FITS or *native* JPG and send as-is. For ASI and SV cameras, there is no native, native would be just raw data which cannot be read. 

We should probably add a method to remove "Native" from the available selections for some drivers.

knro, 2022-10-02:
A simple trick now that might EncodeFormatSP.nsp = 1 in UpdateProperties()

jctk, 2022-10-02:
> We should probably add a method to remove "Native" from the available selections for some drivers.

Yes, I agree with you.

> A simple trick now that might EncodeFormatSP.nsp = 1 in UpdateProperties()

I will try sonn it.
Thanks !

closed 2024-03-17 by knro