indilib/indi-3rdparty #666: indi_sv305_ccd: Cooling-related API calls may fail in TimerHit function when using autofocus with a cooling camera

jctk, 2022-10-02:
Details are unknown at this time.
I guess this Issue is affected by fix #654.

**Describe the bug**
When Auto Focus is executed with the cooler on in the SV405CC, the following error repeatedly occurs after "Subframe set" after a short period of time, and the camera cannot take pictures.
The error message remains after Auto Focus is stopped.

```
2022-10-02T06:59:46: [INFO] Error, unable to get cooler power due to ...(SVB_COOLER_POWER:16) 
2022-10-02T06:59:46: [INFO] Error, unable to get temp due to ...(SVB_CURRENT_TEMPERATURE:16) 
2022-10-02T06:59:43: [INFO] Error, unable to get cooler power due to ...(SVB_COOLER_POWER:16) 
2022-10-02T06:59:43: [INFO] Error, unable to get temp due to ...(SVB_CURRENT_TEMPERATURE:16) 
2022-10-02T06:59:43: [INFO] Subframe set 
2022-10-02T06:59:42: [INFO] Camera control 0 to 400 
2022-10-02T06:59:40: [INFO] Camera control 0 to 400 
```

**To Reproduce**
1. Run Telescope Simulator, Focuser Simulator and SVBONY SV305 drivers.
2. Cooler on.
3. Check "Auto Select Star" and "Sub Frame" in Ekos Focuser module.
4. Run autofocus.
5. Errors occur.
If no error occurs, stop Autofocus and repeat the operation from 3.

**Expected behavior**
No error occurs. 

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - INDI Server: Stellarmate OS, INDI Version 1.9.8
 - KStars: Windows 11, KStars 3.6.0

knro, 2022-10-02:
Looks like it's triggered when the subframe is set? If you do this manually in the driver, is this behavior triggered?

jctk, 2022-10-02:
I can't say anything about that, because I haven't done any research on the source code yet.
So far I haven't encountered this problem when manually changing the subframe on the Capture module.

Maybe the problem is that #654 is replacing the size of the ROI with the size actually set by the SVBONY SDK, as opposed to the size of the ROI expected by the Focuser module.
Whatever the case I will have to do my research to find out the real reason.

jctk, 2022-10-02:
Additional comment:
- Reproduce even with the cooler off.

jctk, 2022-10-02:
Log when reproducing the Issue.
Drivers in profile: indi_simulator_focus and indi_sv305_ccd
```
[2022-10-02T21:55:41.486 東京 (標準時) INFO ][     org.kde.kstars.ekos.focus] - Starting focus with Detection:  "Gradient"  Algorithm:  "Linear"  Box size:  64  Subframe:  yes  Autostar:  yes  Full frame:  no   [ 0 %, 100 %]  Step Size:  5  Threshold:  150  Gaussian Sigma:  1.5  Gaussian Kernel size:  5  Multi row average:  3  Tolerance:  1  Frames:  1  Maximum Travel:  10000  Curve Fit:  "Quadradic"  Use Weights:  no  R2 Limit:  0
[2022-10-02T21:55:41.487 東京 (標準時) INFO ][     org.kde.kstars.ekos.focus] - "Autofocus in progress..."
[2022-10-02T21:55:41.489 東京 (標準時) DEBG ][     org.kde.kstars.ekos.focus] - State: "In Progress"
[2022-10-02T21:55:41.490 東京 (標準時) DEBG ][   org.kde.kstars.ekos.capture] - Focus State changed from "Idle" to "In Progress"
[2022-10-02T21:55:41.491 東京 (標準時) DEBG ][     org.kde.kstars.ekos.focus] - "Linear: Travel 10000 initStep 5 pos 50000 min 0 max 100000 maxIters 30 tolerance 0.01 minlimit 40000 maxlimit 60000 init#steps 5 algo 2 backlash 0 curvefit 0 useweights 0"
[2022-10-02T21:55:41.492 東京 (標準時) DEBG ][     org.kde.kstars.ekos.focus] - "Linear: initialPosition 50025 sized 5"
[2022-10-02T21:55:41.492 東京 (標準時) DEBG ][     org.kde.kstars.ekos.focus] - "Linear: extending outward movement by 25"
[2022-10-02T21:55:41.494 東京 (標準時) INFO ][     org.kde.kstars.ekos.focus] - "Focusing outward by 50 steps..."
[2022-10-02T21:55:41.541 東京 (標準時) INFO ][           org.kde.kstars.indi] - Focuser Simulator :  "[INFO] Focuser moved to position 50050 "
[2022-10-02T21:55:41.542 東京 (標準時) DEBG ][     org.kde.kstars.ekos.focus] - Abs Focuser position changed to  50050 State: Ok
[2022-10-02T21:55:41.543 東京 (標準時) DEBG ][     org.kde.kstars.ekos.focus] - "Linear: un-doing extension. Moving back in by 25"
[2022-10-02T21:55:41.544 東京 (標準時) INFO ][     org.kde.kstars.ekos.focus] - "Focusing inward by 25 steps..."
[2022-10-02T21:55:41.652 東京 (標準時) INFO ][           org.kde.kstars.indi] - Focuser Simulator :  "[INFO] Focuser moved to position 50025 "
[2022-10-02T21:55:41.653 東京 (標準時) DEBG ][     org.kde.kstars.ekos.focus] - Abs Focuser position changed to  50025 State: Ok
[2022-10-02T21:55:41.654 東京 (標準時) DEBG ][     org.kde.kstars.ekos.focus] - "Focus position reached at 50025, starting capture in 0 seconds."
[2022-10-02T21:55:41.656 東京 (標準時) INFO ][     org.kde.kstars.ekos.focus] - "Capturing image..."
[2022-10-02T21:55:41.663 東京 (標準時) INFO ][           org.kde.kstars.indi] - SVBONY SV405CC 0 :  "[INFO] Camera control 0 to 400 "
[2022-10-02T21:55:41.758 東京 (標準時) DEBG ][           org.kde.kstars.indi] - SVBONY SV405CC 0 : "[DEBUG] Taking a 4 seconds frame... "
[2022-10-02T21:55:45.714 東京 (標準時) DEBG ][           org.kde.kstars.indi] - SVBONY SV405CC 0 : "[DEBUG] Current timeleft:0.00 sec. "
[2022-10-02T21:55:46.400 東京 (標準時) DEBG ][           org.kde.kstars.indi] - SVBONY SV405CC 0 : "[DEBUG] SVBGetVideoData:result=0 "
[2022-10-02T21:55:46.450 東京 (標準時) INFO ][           org.kde.kstars.indi] - SVBONY SV405CC 0 :  "[WARNING] Telescope focal length is missing. "
[2022-10-02T21:55:46.451 東京 (標準時) INFO ][           org.kde.kstars.indi] - SVBONY SV405CC 0 :  "[WARNING] Telescope aperture is missing. "
[2022-10-02T21:55:46.501 東京 (標準時) DEBG ][           org.kde.kstars.indi] - SVBONY SV405CC 0 : "[DEBUG] Uploading file. Ext: fits, Size: 23379840, sendImage? Yes, saveImage? No "
[2022-10-02T21:55:46.562 東京 (標準時) DEBG ][           org.kde.kstars.indi] - SVBONY SV405CC 0 : "[DEBUG] BLOB transfer took 0.00167497 seconds "
[2022-10-02T21:55:46.563 東京 (標準時) DEBG ][           org.kde.kstars.indi] - SVBONY SV405CC 0 : "[DEBUG] Upload complete "
[2022-10-02T21:55:49.073 東京 (標準時) DEBG ][           org.kde.kstars.indi] - Image received. Mode: "Focus" Size: 23379840
[2022-10-02T21:55:49.230 東京 (標準時) INFO ][     org.kde.kstars.ekos.focus] - "Image received."
[2022-10-02T21:55:49.232 東京 (標準時) INFO ][     org.kde.kstars.ekos.focus] - "Detecting sources..."
[2022-10-02T21:55:49.290 東京 (標準時) WARN ][              kf.notifications] - updating a notification is not supported yet.
[2022-10-02T21:55:49.335 東京 (標準時) INFO ][     org.kde.kstars.ekos.focus] - "Detection complete."
[2022-10-02T21:55:49.336 東京 (標準時) DEBG ][     org.kde.kstars.ekos.focus] - Focus newFITS # 1 : Current HFR  2.58081  Num stars  2
[2022-10-02T21:55:49.339 東京 (標準時) DEBG ][     org.kde.kstars.ekos.focus] - Frame is subframed. X: 3359 Y: 1898 W: 192 H: 192 binX: 1 binY: 1
[2022-10-02T21:55:49.341 東京 (標準時) INFO ][     org.kde.kstars.ekos.focus] - "Capturing image..."
[2022-10-02T21:55:49.349 東京 (標準時) INFO ][           org.kde.kstars.indi] - SVBONY SV405CC 0 :  "[INFO] Camera control 0 to 400 "
[2022-10-02T21:55:49.429 東京 (標準時) DEBG ][           org.kde.kstars.indi] - SVBONY SV405CC 0 : "[DEBUG] Requested CCD Frame is (3359,1898) (192 x 192) "
[2022-10-02T21:55:50.150 東京 (標準時) DEBG ][           org.kde.kstars.indi] - SVBONY SV405CC 0 : "[DEBUG] Given ROI x=3359, y=1898, w=192, h=192 "
[2022-10-02T21:55:50.217 東京 (標準時) DEBG ][           org.kde.kstars.indi] - SVBONY SV405CC 0 : "[DEBUG] Actual ROI x=3358, y=1898, w=192, h=192, bin=1 "
[2022-10-02T21:55:50.219 東京 (標準時) INFO ][           org.kde.kstars.indi] - SVBONY SV405CC 0 :  "[INFO] Subframe set "
[2022-10-02T21:55:50.266 東京 (標準時) DEBG ][           org.kde.kstars.indi] - SVBONY SV405CC 0 : "[DEBUG] Taking a 4 seconds frame... "
[... cut ...]
[2022-10-02T21:56:05.080 東京 (標準時) INFO ][           org.kde.kstars.indi] - SVBONY SV405CC 0 :  "[INFO] Error, unable to get temp due to ...(SVB_CURRENT_TEMPERATURE:16) "
[2022-10-02T21:56:05.124 東京 (標準時) DEBG ][           org.kde.kstars.indi] - SVBONY SV405CC 0 : "[DEBUG] Current timeleft:-2.13 sec. "
[2022-10-02T21:56:06.125 東京 (標準時) DEBG ][           org.kde.kstars.indi] - SVBONY SV405CC 0 : "[DEBUG] SVBGetVideoData:result=11 "
[2022-10-02T21:56:06.180 東京 (標準時) DEBG ][           org.kde.kstars.indi] - SVBONY SV405CC 0 : "[DEBUG] Timeout for image data retrieval. "
```

knro, 2022-10-03:
Maybe while exposure is in progress, you cannot inquire of other parameters and must wait until exposure is finished? Not sure, just a guess.

jctk, 2022-10-03:
That does not seem to be the case.
If that is the case, then you will not be able to acquire parameters when exposing with the Capture Module as well.

jctk, 2022-10-12:
I have created the test code linked below to reproduce this issue.
https://github.com/jctk/svbtest/blob/master/svbtest.cpp

After making it, run the following.
$ . /svbtest SV405CC

The code repeats 12 times calling SVBSetROIFormat, exposing with SVB_MODE_TRIG_SOFT and SVBGetVideoData.
After several repetitions, SVBGetVideoData returns SVB_ERROR_TIMEOUT and the data can no longer be obtained.
Lines 175-225 of the test code are the loop process of ROI change, exposure, and data acquisition.

If the ROI is not changed, this problem will not occur.
I am using Raspberry PI 64bit OS.

I have not checked for anomalies in cooling related functions, but the error with SVBGetVideoData also occurs with the INDI SVBOY CCD driver.

I am contacting SVBONY about this issue.

jctk, 2022-10-27:
Hi @knro 

SVBONY seems to recognize that Camera cannot send data about small ROIs.
Rita from SVBONY to KStars Developer to inquire about setting ROI while KStars autofocus is running.

knro, 2022-10-27:
@jctk Excellent, thank you for the update. We will await their communication.

jctk, 2022-11-09:
Hi @knro 
I have confirmed that SVBONY Camera SDK 1.9.8 and SV405CC firmware 2.0.0.6 resolve this issue.
I will pull request libsvbony in the near future.

knro, 2022-11-09:
Great news! Thank you.

PR #682 (2022-11-10): Update SVBONY Camera SDK to 1.9.8 (Fixed #666)
Version 1.9.7
1. Fixed the bug that SV405CC occasionally loses response(Camera firmware needs to be upgraded to v2.0.0.6)
2. Fixed the bug that SV405CC cannot set the maximum resolution

Version 1.9.8
1.Fixed the bug that SV405CC set some resolution images incorrectly

Note:
This upgrade fixes #666
commit 27304232 (2022-11-10): Update SVBONY Camera SDK to 1.9.8 (Fixed #666) (#682)

fixes https://github.com/indilib/indi-3rdparty/issues/666
closed 2022-11-10 by knro