indilib/indi-3rdparty #670: indi_asi_wheel can not setup the Zwo EFW in raspberry pi 4

pereblay, 2022-10-13:
Hi,

I have tried  by cloning different versions from github (master branch, v1.9.8, v1.9.7, and v 1.8.9) but the problem persists.
The system detects the filter wheel properly:

$ dmesg
[79423.365621] hid-generic 0003:03C3:1F01.0007: hiddev96,hidraw4: USB HID v1.01 Device [ZW0 ZWO EFW] on usb-0000:01:00.0-1.2.1.4/input0

but indi_asi_wheel fails to initialize it:

$ indiserver indi_asi_wheel
2022-10-13T19:43:28: startup: indiserver indi_asi_wheel
2022-10-13T19:43:28: Driver indi_asi_wheel: Searching for HID Device VID: 0x3c3 PID: 0x1f01
2022-10-13T19:43:28: Driver indi_asi_wheel: Searching for HID Device VID: 0x3c3 PID: 0x1f02

My system:
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye

$ uname -a
Linux icaosky 5.15.61-v7l+ #1579 SMP Fri Aug 26 11:13:03 BST 2022 armv7l GNU/Linux

The filter wheel lies behind a powered usb hub, but I have also a focuser there  and it is initialized propoerly.

knro, 2022-10-14:
I just tested on StellarMate and couldn't find any issues.

[image]

pereblay, 2022-10-14:
Thanks Jasem,

Which version of OS (buster? bullseye?) and kernel? (to test the exact same
configuration).

knro, 2022-11-16:
Yes bullseye, it works fine without any issues.

closed 2022-11-16 by knro