indilib/indi-3rdparty #681: Altair unable to enable RAW format on first connection to indiserver

aaronwmorris, 2022-11-02:
**Describe the bug**
The indi_altair_ccd driver will not return the CCD_CAPTURE_FORMAT control on the first connection to the indiserver.  Any subsequent connection will return the control.  Altair color cameras start in RGB mode (as opposed to RAW mode) and the CCD_CAPTURE_FORMAT control is used to change to RAW mode.

I believe this is only an issue on color Altair GPCAM3 (maybe 2 as well) cameras.  I have had reports of the problem on the 224C and 290C GPCAM3 cameras.

EKOS appears to bypass this issue because it connects to the indiserver twice when connecting.

**Desktop (please complete the following information):**
 - OS:  Linux
 - Arch:  ARM
 - Version:  Multiple 1.9.6, 1.9.7, 1.9.8
 

** Log **

CCD_CAPTURE_FORMAT vector is empy on first connection.  Not sure if this is relevant.
```
### SNIP ###
2022-11-02T19:39:57: Client 9: sending <defTextVector device="Altair ALTAIR224C3(USB2.0)" name="FITS_HEADER" label="FITS Header" group="General Info" state="Idle" perm="rw" timeout="60" timestamp="2022-11-02T19:39:57">
    <defText name="FITS_OBSERVER" label="Observer">
Unknown
    </defText>
    <defText name="FITS_OBJECT" label="Object">
Unknown
    </defText>
</defTextVector>

2022-11-02T19:39:57: Client 9: sending <defSwitchVector device="Altair ALTAIR224C3(USB2.0)" name="CCD_CAPTURE_FORMAT" label="Format" group="Image Settings" state="Idle" perm="rw" rule="OneOfMany" timeout="60" timestamp="2022-11-02T19:39:57"/>

2022-11-02T19:39:57: Client 9: sending <defSwitchVector device="Altair ALTAIR224C3(USB2.0)" name="CCD_TRANSFER_FORMAT" label="Encode" group="Image Settings" state="Idle" perm="rw" rule="OneOfMany" timeout="60" timestamp="2022-11-02T19:39:57">
    <defSwitch name="FORMAT_FITS" label="FITS">
On
    </defSwitch>
    <defSwitch name="FORMAT_NATIVE" label="Native">
Off
    </defSwitch>
</defSwitchVector>
### SNIP ###
```

Second connection
```
### SNIP ###
2022-11-02T19:42:03: Client 9: sending <defTextVector device="Altair ALTAIR224C3(USB2.0)" name="FITS_HEADER" label="FITS Header" group="General Info" state="Idle" perm="rw" timeout="60" timestamp="2022-11-02T19:42:03">
    <defText name="FITS_OBSERVER" label="Observer">
Unknown
    </defText>
    <defText name="FITS_OBJECT" label="Object">
Unknown
    </defText>
</defTextVector>

2022-11-02T19:42:03: Client 9: sending <defSwitchVector device="Altair ALTAIR224C3(USB2.0)" name="CCD_CAPTURE_FORMAT" label="Format" group="Image Settings" state="Idle" perm="rw" rule="OneOfMany" timeout="60" timestamp="2022-11-02T19:42:03">
    <defSwitch name="INDI_RGB" label="RGB">
Off
    </defSwitch>
    <defSwitch name="INDI_RAW" label="RAW 16">
On
    </defSwitch>
</defSwitchVector>

2022-11-02T19:42:03: Client 9: sending <defSwitchVector device="Altair ALTAIR224C3(USB2.0)" name="CCD_TRANSFER_FORMAT" label="Encode" group="Image Settings" state="Idle" perm="rw" rule="OneOfMany" timeout="60" timestamp="2022-11-02T19:42:03">
    <defSwitch name="FORMAT_FITS" label="FITS">
On
    </defSwitch>
    <defSwitch name="FORMAT_NATIVE" label="Native">
Off
    </defSwitch>
</defSwitchVector>
### SNIP ###
```

aaronwmorris, 2022-11-02:
**Test Script**

```
#!/usr/bin/env python3

import PyIndi
import time
import sys
import logging

CCD = "Altair ALTAIR224C3(USB2.0)"

logger = logging.getLogger(__name__)
logger.setLevel(level=logging.INFO)

LOG_FORMATTER_STREAM = logging.Formatter('%(asctime)s [%(levelname)s] %(processName)s %(funcName)s() #%(lineno)d: %(message)s')
LOG_HANDLER_STREAM = logging.StreamHandler()
LOG_HANDLER_STREAM.setFormatter(LOG_FORMATTER_STREAM)
logger.addHandler(LOG_HANDLER_STREAM)

class IndiClient(PyIndi.BaseClient):
    def __init__(self):
        super(IndiClient, self).__init__()

    def newDevice(self, d):
        pass

    def newProperty(self, p):
        pass

    def removeProperty(self, p):
        pass

    def newBLOB(self, bp):
        global blobEvent
        logger.info("new BLOB %s", bp.name)
        blobEvent.set()
        pass

    def newSwitch(self, svp):
        pass

    def newNumber(self, nvp):
        pass

    def newText(self, tvp):
        pass

    def newLight(self, lvp):
        pass

    def newMessage(self, d, m):
        pass

    def serverConnected(self):
        pass

    def serverDisconnected(self, code):
        pass

# connect the server
indiclient = IndiClient()
indiclient.setServer("localhost", 7624)

logger.info("Connecting to indiserver")
if not (indiclient.connectServer()):
    logger.error(
        "No indiserver running on %s:%d - Try to run  'indiserver indi_altair_ccd'",
        indiclient.getHost(),
        indiclient.getPort()
    )
    sys.exit(1)

logger.info("Connecting to ccd")

device_ccd = indiclient.getDevice(CCD)
while not device_ccd:
    time.sleep(0.5)
    device_ccd = indiclient.getDevice(CCD)

ccd_connect = device_ccd.getSwitch("CONNECTION")
while not ccd_connect:
    time.sleep(0.5)
    ccd_connect = device_ccd.getSwitch("CONNECTION")

if not device_ccd.isConnected():
    ccd_connect[0].setState(PyIndi.ISS_ON)   # the "CONNECT" switch
    ccd_connect[1].setState(PyIndi.ISS_OFF)  # the "DISCONNECT" switch
    indiclient.sendNewSwitch(ccd_connect)

while not device_ccd.isConnected():
    time.sleep(0.5)
    logger.warning('Waiting on ccd connection')

logger.info("ccd connected")

### any argument passed to python will stop the script here
try:
    sys.argv[1]
    sys.exit(1)
except IndexError:
    pass

logger.info("Get CCD_CAPTURE_FORMAT control")

ccd_capture_format = device_ccd.getSwitch("CCD_CAPTURE_FORMAT")
while not (ccd_capture_format):
    logger.warning('*** CCD_CAPTURE_FORMAT not set ***')
    time.sleep(1.0)
    ccd_capture_format = device_ccd.getSwitch("CCD_CAPTURE_FORMAT")

logger.info("Set CCD_CAPTURE_FORMAT control to RAW")

ccd_capture_format[1].setState(PyIndi.ISS_ON)   # INDI_RAW
ccd_capture_format[0].setState(PyIndi.ISS_OFF)  # INDI_RGB
indiclient.sendNewSwitch(ccd_capture_format)
```

aaronwmorris, 2022-11-02:
**First connection**
```
$ ./altair_ccd_capture_format.py
2022-11-02 15:47:25,913 [INFO] MainProcess <module>() #67: Connecting to indiserver
INDI::BaseClient::connectServer: creating new connection...
INDI::BaseClient::connectServer: Already connected.
2022-11-02 15:47:25,914 [INFO] MainProcess <module>() #78: Connecting to ccd
CCD_CAPTURE_FORMAT: defSwitchVector with no valid members
2022-11-02 15:47:26,916 [WARNING] MainProcess <module>() #98: Waiting on ccd connection
2022-11-02 15:47:26,917 [INFO] MainProcess <module>() #100: ccd connected
2022-11-02 15:47:26,917 [INFO] MainProcess <module>() #111: Get CCD_CAPTURE_FORMAT control
2022-11-02 15:47:26,918 [WARNING] MainProcess <module>() #115: *** CCD_CAPTURE_FORMAT not set ***
2022-11-02 15:47:27,919 [WARNING] MainProcess <module>() #115: *** CCD_CAPTURE_FORMAT not set ***
2022-11-02 15:47:28,921 [WARNING] MainProcess <module>() #115: *** CCD_CAPTURE_FORMAT not set ***
2022-11-02 15:47:29,923 [WARNING] MainProcess <module>() #115: *** CCD_CAPTURE_FORMAT not set ***
2022-11-02 15:47:30,924 [WARNING] MainProcess <module>() #115: *** CCD_CAPTURE_FORMAT not set ***
2022-11-02 15:47:31,927 [WARNING] MainProcess <module>() #115: *** CCD_CAPTURE_FORMAT not set ***
2022-11-02 15:47:32,929 [WARNING] MainProcess <module>() #115: *** CCD_CAPTURE_FORMAT not set ***
2022-11-02 15:47:33,931 [WARNING] MainProcess <module>() #115: *** CCD_CAPTURE_FORMAT not set ***
2022-11-02 15:47:34,933 [WARNING] MainProcess <module>() #115: *** CCD_CAPTURE_FORMAT not set ***
2022-11-02 15:47:35,935 [WARNING] MainProcess <module>() #115: *** CCD_CAPTURE_FORMAT not set ***
2022-11-02 15:47:36,937 [WARNING] MainProcess <module>() #115: *** CCD_CAPTURE_FORMAT not set ***
2022-11-02 15:47:37,939 [WARNING] MainProcess <module>() #115: *** CCD_CAPTURE_FORMAT not set ***
2022-11-02 15:47:38,941 [WARNING] MainProcess <module>() #115: *** CCD_CAPTURE_FORMAT not set ***
2022-11-02 15:47:39,943 [WARNING] MainProcess <module>() #115: *** CCD_CAPTURE_FORMAT not set ***
2022-11-02 15:47:40,945 [WARNING] MainProcess <module>() #115: *** CCD_CAPTURE_FORMAT not set ***
^CTraceback (most recent call last):
  File "/home/aaron/git/indi-allsky/testing/./altair_ccd_capture_format.py", line 116, in <module>
    time.sleep(1.0)
KeyboardInterrupt
```

**Second connection**
```
$ ./altair_ccd_capture_format.py
2022-11-02 15:47:43,611 [INFO] MainProcess <module>() #67: Connecting to indiserver
INDI::BaseClient::connectServer: creating new connection...
INDI::BaseClient::connectServer: Already connected.
2022-11-02 15:47:43,612 [INFO] MainProcess <module>() #78: Connecting to ccd
2022-11-02 15:47:44,114 [INFO] MainProcess <module>() #100: ccd connected
2022-11-02 15:47:44,115 [INFO] MainProcess <module>() #111: Get CCD_CAPTURE_FORMAT control
2022-11-02 15:47:44,115 [INFO] MainProcess <module>() #120: Set CCD_CAPTURE_FORMAT control to RAW
```

sharpie78, 2022-11-03:
Looks like you've got your head around where the issue lies and what to do to sort it. I could pretend to know what all the stuff above means but I'd be lying. I think I can decipher about half of it. haha

aaronwmorris, 2022-11-03:
> Looks like you've got your head around where the issue lies and what to do to sort it. I could pretend to know what all the stuff above means but I'd be lying. I think I can decipher about half of it. haha

It is somewhat clear to me, but to be honest, I do not understand a lot of what is happening.  The Altair support appears to be built on the Touptek driver.  There is a point in the touptek initialization where it sets up the camera depending on if the camera is mono vs color.

https://github.com/indilib/indi-3rdparty/blob/master/indi-toupbase/indi_toupbase.cpp#L629

I *think* what is happening is the TC_VIDEO_COLOR_RAW flag is not ready/set on the first connection, so the camera gets put in RGB24 mode.  This may be a bug in the driver.  All other cameras I have experience with initialize in RAW mode by default in indi.

https://github.com/indilib/indi-3rdparty/blob/master/indi-toupbase/indi_toupbase.cpp#L690

Color Touptek cameras may also be affected, but I do not have one to test.

sharpie78, 2022-11-05:
i'm making connections in my head, linked to the other thread on this camera, which probably won't help but sometimes a naive idiots idea of how software works can help ...
The AltairCapture software opens the camera in RGB mode by default when connected (I think). Could there be a link there ? Could they have preferred their own software to be the main way to use the camera with full functionality and therefore set the hardware to defaults for that software ? Just kinda wondering if it's the way their hardware works rather than the INDI driver. I'm probably shooting in the dark but seeing as I have used the Altair software and it defaults to 1280x960 RGB (I think 16bit), I thought I'd mention it.

sharpie78, 2022-11-05:
I just had a look through the toupbase code you linked to.... How do you guys write/understand this stuff ? 
It still amazes me that people like you can make electronics do the stuff we want it to. It must take years and years to learn computer languages and there's more than one. My hat is off to you and even if you never figure out this Altair issue, indi-allsky is a great piece of software.

knro, 2022-11-06:
@aaronwmorris Please test and let me know if this resolves the issue.

closed 2022-11-06 by knro

aaronwmorris, 2022-11-06:
Thank you @knro this fixes the problem of the control never being populated...however it introduces another issue.

Previously, the indi_altair_ccd driver had an interesting behavior, the `CCD_CFA` property would not be populated until the camera was put in raw mode.  Almost all of the other cameras I have experience with always populate this property regardless of mode.

With the changes you made, the CCD_CFA property is never populated.  I can open another issue if you want that.

In my software, I have been using the CCD_CFA property to pick the correct debayering algo to apply, but maybe I am pulling the information from the wrong place.  Perhaps I should be using the `BAYERPAT` FITS header instead.  What do you think?

commit 960ec22a (2022-11-06): Define capture format before INDI::CCD::updateProperties is called. Fixes #681