indilib/indi-3rdparty #683: indi_svbony_ccd: When the sequence is started, the target temperature is set to 25°C

jctk, 2022-11-11:
**Describe the bug**
The target temperature is set to 25°C when the sequence is started, even though the target temperature set for the sequence is -10°C.
The temperature ramp is not set. 

**To Reproduce**
1. Set the target temperature to -10°C in the Ekos capture module and register the sequence.
2. The target temperature is set to 25°C when start shooting.

**Expected behavior**
The sequence is cooled at the registered temperature of the sequence.

**Screenshots**
[image]
Screenshot with [Stop Sequence] button to stop the sequence and [Start Sequence] button to start the sequence.

**Desktop**
 - OS: Client KStars running on Windows 11.
 - OS: INDI server running on StellarMate OS 64bit
 - Version: INDI 1.9.8, SVBONY CCD driver newest build on indi-3rdparty repository.

knro, 2022-11-16:
Thank you for the report. However, this is not an INDI issue (not that I can see). Also, I can't reproduce it.

closed 2022-11-16 by knro

jctk, 2022-11-16:
I believe this is a problem with the SVBONY CCD INDI driver.
 25°C, this magic number is the default value in the Target Temperature column when you open the Ekos Capture module in the SVBONY CCD INDI driver, and when you switch cameras in the Capture module or switch the display from another module to the Capture module 25°C may be set in the Target Temperature field.
I have not investigated this issue yet, so I cannot give you any solid evidence, but I believe that the strange behavior regarding the default value of 25°C is a problem with the SVBONY CCD INDI driver.