indilib/indi-3rdparty #684: KStars 3.6.1 - Problems with Playerone

alepensato, 2022-11-16:
I updated to the last version but I was unable to use my Playerone sedna as guide.
When i start looping the screen does not update the image and in console i find some row with **reset SuperSpeed USB device number 2 using xhci_hcd**

I reinstalled all the system with latest updates, but the problem persist.
Here a log file.
I also tried to use the SDK from PlayerOne site but with no luck.

If i start a loop it goes but when i stop.... the image download freeze.
In the log you can see a problem with binning, but i do not know why the frame size was wrong.

[2022-11-10T23:28:06.986 CET DEBG ][           org.kde.kstars.indi] - INDI: Connecting to local INDI server on port  7624  ...
[2022-11-10T23:28:06.987 CET DEBG ][           org.kde.kstars.indi] - INDI server connected.
[2022-11-10T23:28:06.997 CET INFO ][           org.kde.kstars.ekos] - "Servizi INDI avviati sulla porta 7.624."
[2022-11-10T23:28:07.000 CET DEBG ][           org.kde.kstars.indi] - Received new device Focuser Simulator
[2022-11-10T23:28:07.001 CET DEBG ][           org.kde.kstars.indi] - Received new device CCD Simulator
[2022-11-10T23:28:07.003 CET DEBG ][           org.kde.kstars.indi] - Received new device Telescope Simulator
[2022-11-10T23:28:07.005 CET DEBG ][           org.kde.kstars.indi] - Received new device PlayerOne CCD Sedna-M
[2022-11-10T23:28:07.018 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2022-11-10T22:28:06: FIFO: start indi_simulator_focus -n \"Focuser Simulator\""
[2022-11-10T23:28:07.018 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2022-11-10T22:28:06: With name: Focuser Simulator"
[2022-11-10T23:28:07.018 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2022-11-10T22:28:06: FIFO: Starting driver indi_simulator_focus"
[2022-11-10T23:28:07.019 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2022-11-10T22:28:06: Driver indi_simulator_focus: pid=6106 rfd=15 wfd=15 efd=19"
[2022-11-10T23:28:07.019 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2022-11-10T22:28:06: Driver indi_playerone_ccd: snooping on Telescope Simulator.EQUATORIAL_EOD_COORD"
[2022-11-10T23:28:07.019 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2022-11-10T22:28:06: Driver indi_playerone_ccd: snooping on Telescope Simulator.EQUATORIAL_COORD"
[2022-11-10T23:28:07.019 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2022-11-10T22:28:06: Driver indi_playerone_ccd: snooping on Telescope Simulator.TELESCOPE_INFO"
[2022-11-10T23:28:07.019 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2022-11-10T22:28:06: Driver indi_playerone_ccd: snooping on Telescope Simulator.GEOGRAPHIC_COORD"
[2022-11-10T23:28:07.019 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2022-11-10T22:28:06: Driver indi_playerone_ccd: snooping on Telescope Simulator.TELESCOPE_PIER_SIDE"
[2022-11-10T23:28:07.019 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2022-11-10T22:28:06: Driver indi_playerone_ccd: snooping on Rotator Simulator.ABS_ROTATOR_ANGLE"
[2022-11-10T23:28:07.020 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2022-11-10T22:28:06: Driver indi_playerone_ccd: snooping on Focuser Simulator.ABS_FOCUS_POSITION"
[2022-11-10T23:28:07.020 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2022-11-10T22:28:06: Driver indi_playerone_ccd: snooping on Focuser Simulator.FOCUS_TEMPERATURE"
[2022-11-10T23:28:07.020 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2022-11-10T22:28:06: Driver indi_playerone_ccd: snooping on SQM.SKY_QUALITY"
[2022-11-10T23:28:07.020 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  "2022-11-10T22:28:06: Client 8: new arrival from local pid 5822 (user: 1000:1000) - welcome!"
[2022-11-10T23:28:07.020 CET DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2022-11-10T23:28:07.092 CET DEBG ][           org.kde.kstars.indi] - INDIListener: New device "Focuser Simulator"
[2022-11-10T23:28:07.093 CET INFO ][           org.kde.kstars.ekos] - Ekos received a new device:  "Focuser Simulator"
[2022-11-10T23:28:07.096 CET WARN ][                       default] - QLayout: Cannot add a null widget to QGridLayout/
[2022-11-10T23:28:07.098 CET DEBG ][           org.kde.kstars.indi] - < Focuser Simulator >: < CONNECTION >
[2022-11-10T23:28:07.098 CET DEBG ][           org.kde.kstars.indi] - < Focuser Simulator >: < DRIVER_INFO >
[2022-11-10T23:28:07.099 CET DEBG ][           org.kde.kstars.indi] - < Focuser Simulator >: < DEBUG >
[2022-11-10T23:28:07.100 CET DEBG ][           org.kde.kstars.indi] - < Focuser Simulator >: < POLLING_PERIOD >
[2022-11-10T23:28:07.101 CET DEBG ][           org.kde.kstars.indi] - < Focuser Simulator >: < CONFIG_PROCESS >
[2022-11-10T23:28:07.101 CET DEBG ][           org.kde.kstars.indi] - < Focuser Simulator >: < CONNECTION_MODE >
[2022-11-10T23:28:07.101 CET DEBG ][           org.kde.kstars.indi] - < Focuser Simulator >: < DEVICE_PORT >
[2022-11-10T23:28:07.102 CET DEBG ][           org.kde.kstars.indi] - < Focuser Simulator >: < DEVICE_BAUD_RATE >
[2022-11-10T23:28:07.102 CET DEBG ][           org.kde.kstars.indi] - < Focuser Simulator >: < DEVICE_AUTO_SEARCH >
[2022-11-10T23:28:07.102 CET DEBG ][           org.kde.kstars.indi] - < Focuser Simulator >: < DEVICE_PORT_SCAN >
[2022-11-10T23:28:07.103 CET DEBG ][           org.kde.kstars.indi] - < Focuser Simulator >: < Mode >
[2022-11-10T23:28:07.252 CET DEBG ][           org.kde.kstars.indi] - INDIListener: New device "CCD Simulator"
[2022-11-10T23:28:07.253 CET INFO ][           org.kde.kstars.ekos] - Ekos received a new device:  "CCD Simulator"
[2022-11-10T23:28:07.254 CET DEBG ][           org.kde.kstars.indi] - < CCD Simulator >: < CONNECTION >
[2022-11-10T23:28:07.255 CET DEBG ][           org.kde.kstars.indi] - < CCD Simulator >: < DRIVER_INFO >
[2022-11-10T23:28:07.255 CET DEBG ][           org.kde.kstars.indi] - < CCD Simulator >: < POLLING_PERIOD >
[2022-11-10T23:28:07.255 CET DEBG ][           org.kde.kstars.indi] - < CCD Simulator >: < DEBUG >
[2022-11-10T23:28:07.257 CET DEBG ][           org.kde.kstars.indi] - < CCD Simulator >: < CONFIG_PROCESS >
[2022-11-10T23:28:07.257 CET DEBG ][           org.kde.kstars.indi] - < CCD Simulator >: < ACTIVE_DEVICES >
[2022-11-10T23:28:07.258 CET DEBG ][           org.kde.kstars.indi] - < CCD Simulator >: < SIMULATOR_SETTINGS >
[2022-11-10T23:28:07.258 CET DEBG ][           org.kde.kstars.indi] - < CCD Simulator >: < EQUATORIAL_PE >
[2022-11-10T23:28:07.258 CET DEBG ][           org.kde.kstars.indi] - < CCD Simulator >: < SIM_FOCUSING >
[2022-11-10T23:28:07.258 CET DEBG ][           org.kde.kstars.indi] - < CCD Simulator >: < SIMULATE_BAYER >
[2022-11-10T23:28:07.258 CET DEBG ][           org.kde.kstars.indi] - < CCD Simulator >: < CCD_SIMULATE_CRASH >
[2022-11-10T23:28:07.346 CET DEBG ][           org.kde.kstars.indi] - INDIListener: New device "Telescope Simulator"
[2022-11-10T23:28:07.347 CET INFO ][           org.kde.kstars.ekos] - Ekos received a new device:  "Telescope Simulator"
[2022-11-10T23:28:07.349 CET WARN ][                       default] - QLayout: Cannot add a null widget to QGridLayout/
[2022-11-10T23:28:07.351 CET DEBG ][           org.kde.kstars.indi] - < Telescope Simulator >: < CONNECTION >
[2022-11-10T23:28:07.351 CET DEBG ][           org.kde.kstars.indi] - < Telescope Simulator >: < DRIVER_INFO >
[2022-11-10T23:28:07.352 CET DEBG ][           org.kde.kstars.indi] - < Telescope Simulator >: < POLLING_PERIOD >
[2022-11-10T23:28:07.352 CET DEBG ][           org.kde.kstars.indi] - < Telescope Simulator >: < DEBUG >
[2022-11-10T23:28:07.352 CET DEBG ][           org.kde.kstars.indi] - < Telescope Simulator >: < CONFIG_PROCESS >
[2022-11-10T23:28:07.353 CET DEBG ][           org.kde.kstars.indi] - < Telescope Simulator >: < CONNECTION_MODE >
[2022-11-10T23:28:07.353 CET DEBG ][           org.kde.kstars.indi] - < Telescope Simulator >: < DEVICE_PORT >
[2022-11-10T23:28:07.354 CET DEBG ][           org.kde.kstars.indi] - < Telescope Simulator >: < DEVICE_BAUD_RATE >
[... 240 lines cut ...]
[2022-11-10T23:33:41.882 CET DEBG ][           org.kde.kstars.indi] - PlayerOne CCD Sedna-M : "[DEBUG] Configuration successfully saved. "
[2022-11-10T23:33:47.210 CET DEBG ][           org.kde.kstars.indi] - PlayerOne CCD Sedna-M : "[DEBUG] Aborting exposure... "
[2022-11-10T23:33:55.916 CET DEBG ][           org.kde.kstars.indi] - PlayerOne CCD Sedna-M : "[DEBUG] Frame ROI x:0 y:0 w:1544 h:1038 "
[... cut ...]
[2022-11-10T23:34:52.048 CET DEBG ][           org.kde.kstars.ekos] - Disconnecting  "Focuser Simulator"
[2022-11-10T23:34:52.051 CET DEBG ][           org.kde.kstars.ekos] - Disconnecting  "CCD Simulator"
[2022-11-10T23:34:52.052 CET DEBG ][           org.kde.kstars.ekos] - Disconnecting  "Telescope Simulator"
[2022-11-10T23:34:52.052 CET DEBG ][           org.kde.kstars.ekos] - Disconnecting  "PlayerOne CCD Sedna-M"

I tried with different seconds of exposure and seems that with a value of 0.5sec the problem goes away, but i did just a quick test.

knro, 2022-11-16:
@hiro3110i Can you please take a look?

alepensato, 2022-11-16:
@hiro3110i Thanks for you reply.
Actually I have two similar setup, the difference is in the mount (EQ6 and CEM70), I use a Raspberry Pi4B 4GB RAM with latest RaspbianOS **64bit** where i installed the INDI/KStars packages taken form https://ppa.stellarmate.com/.
As I described When I start it with SIMULATOR or REAL SETUP, I have the Sedna that hangs if I stop a Guide loop. 
It starts correctly but if I stop it and re-run it again, the screen refresh only one time and stop, with the log above.
It seems that the usb device continue to run and ignore the STOP.

With dmesg in cosole i found some row with r**eset SuperSpeed USB device number 2 using xhci_hcd**, tha a search reports similar issue with the ASI cameras.

If i use the identical setup, but with my SD card with KStars v3.6.0 32bit..... I do not have problems

alepensato, 2022-11-16:
@hiro3110i 
I tested the setup also on a  Lenovo t430 with Ubuntu 22.04 lts updated to lastest software releases (OS & INDI)

hiro3110i, 2022-11-17:
@alepensato Could you please tell me more about your setup?
What kind of guiding are you using?  Are you using video streaming mode for guiding?

Sometime video streaming mode may appear unstable even in my environment.
I suspect that it is due to USB3 bandwidth becoming critical. Or maybe it's because the data transfer interruption puts into an unexpected state. I'm not sure but resume from sudden stop streaming may have unexpected response.
If the camera freezes, there is not much the driver can do except for timeout and restart the driver.
Does zwo's ASI camera have similar issues?  If there is a known effective workaround, I'll be able to fix it as well.

alepensato, 2022-11-17:
> @alepensato Could you please tell me more about your setup? What kind of guiding are you using? Are you using video streaming mode for guiding?
>
> Sometime video streaming mode may appear unstable [...]

I use KStars internal guider with all the default Playerone setting that are checked/setted when you start Ekos for the first time. I just checked that the image was at 16bit. Actually I am testing the camera on my desk, I just connect the camera and run the LOOP button on the Guider tabs (loop correctly), stop it, press again Loop and I get only a frame then nothing. I you look at the INDI panel I see that exposure field in the Main controll tab is at 0.00sec and not at 1.00 as before I start for the first time, and the countdown is always at 0.00sec.

If you stop INDI, unplug the USB and again start INDI and the Loop it works, but then it fail again if you stop it.

If you search you can find some report about problem with USB reset and ASI camera, that seems to be solved. But i do not know how (firmware update or driver or sdk).

Same setup and settings but with KStars 3.6.0 at 32bit, works

hiro3110i, 2022-11-17:
@alepensato Thank you for explanations.
Your usage of internal guider is the same as mine.
There seems to be a clear difference between 32bit and 64bit.

KStars v3.6.2beta, indi v1.9.9beta and indi_playerone_ccd v0.8 built from the latest git repository have no problems on my Raspberry Pi4B (32bit).  I would like to hear the manufacturer's opinion on whether there is any problem with the 64bit version of the SDK, or whether there is a problem with hardware compatibility with the 64bit OS.

hiro3110i, 2022-11-17:
@alepensato One short confirmation.
Could you please check your library version of libPlayerOneCamera.so in /usr/lib/arm-linux-* (64bit folder)?
There was a case in the past where an old version was installed for some reason.
It is expected to be symlinked to libPlayerOneCamera.so.3.0.4 (SDK v3.0.4).

hiro3110i, 2022-11-17:
>  I can run at 32bit, i have the 64bit because INDI/KStars stopped to produce packages ready to install, and i do not want to compile all the stuff every time when an update is made.

It's also big problem for me...

alepensato, 2022-11-17:
> @alepensato One short confirmation. Could you please check your library version of libPlayerOneCamera.so in /usr/lib/arm-linux-* (64bit folder)? There was a case in the past where an old version was [...]

I will check later, actually i am away from home

hiro3110i, 2022-11-18:
@alepensato Screenshots were very helpful, thank you.
We also confirmed that there is no problem of library version.
I'm waiting for the manufacturer's opinion on the underlying solution now.
Anyway, current optimal solution for 64bit OS is to set the exposure time to less than 1 second, I understand.

@knro Does it fix of yours work around this problem?
[56952b1](https://github.com/indilib/indi/commit/56952b1bd419e00f4ce1d2370a5253d91f75e8f4)

alepensato, 2022-11-18:
@hiro3110i 
Guide with value of 0.5sec, could not be considered as a solution, is not realistic. 
Usually I use 2-3seconds. 
In this case I use the old 32bit as is, also compile from source KStars/INDI/INDI-3rdparty every time is not my goal.

knro, 2022-11-18:
Does it work in PHD2 for Linux?

alepensato, 2022-11-18:
@knro 
in PHD2 under raspbianos 64bit

knro, 2022-11-18:
Are you saying the camera works fine with PHD2 under Raspberry PI? This is using internal PHD2 drivers or INDI driver?

alepensato, 2022-11-18:
@knro 
Yes, PHD2 under Raspberry PI, connected with INDI driver, with RaspbianOS 64bit. 
If you want take a look, I can invite you to enter into the system using VNC

hiro3110i, 2022-11-21:
@alepensato  Sorry for the late response.
It is an important clue that the 64-bit indi driver does not seem to cause problems with PHD2.  Manufacturer's engineers are always very cooperative.  They're doing some analysis on this issue, and it's important information for them too.
Thanks for testing with PHD2.

knro, 2022-11-22:
I have no camera to test so I'm not sure what's going on.

alepensato, 2022-11-22:
> I have no camera to test so I'm not sure what's going on.

If you want take a look, I can invite you to enter into the system using VNC

hiro3110i, 2022-11-24:
@alepensato Sorry for silence.  I have built and tested a 64bit version on Raspberry Pi4.
I don't like 64bit version due to slowness of GUI response, but it works just as well as the 32bit.
I've tried many combinations of exposure times and binning with no problems.

My 64bit environment is as below:
- Raspberry Pi4 (4GB RAM)
- Ubuntu 22.04.1 server with KDE desktop system (full upgraded on Nov. 24th)
- Xrdp remote desktop server
- Indi v1.9.9beta and KStars v3.6.2beta from ppa:mutlaqja/ppa repo.
- PlayerOne Neptune-C II (with indi_playerone_ccd driver v0.8)
- Telescope simulator
- Microsoft remote desktop on MacBook Air (Intl.)

The problem may not occur in the latest beta of indi and KStars.
Is it possible to test the beta version from the above repository?
Differences of camera model between Sedna-M and Neptune-C II still remain...

Could you reconfirm your power supply voltage and supply capability?
It is necessary to secure sufficient power supply capacity for stable operation of Raspberry Pi4 and CMOS camera.
I use the PD60W port from a Jackery portable power supply.

alepensato, 2022-11-24:
> @alepensato Sorry for silence. I have built and tested a 64bit version on Raspberry Pi4. I don't like 64bit version due to slowness of GUI response, but it works just as well as the 32bit. I've [...]

Due the camera works on 64bit with indi driver and PHD2, all problems of power, cables, compatibility could be removed.
So, it has been tested also with latest 3.6.2 and the problem persist. 

So, another strange thing is that if you use the INDI panel to set the exposure time and binning .... it works and confim that there is some problem in the EKOS interface that "can not speak correctly" with the camera. If I use my old DMK or QHY183M it works

hiro3110i, 2022-11-25:
@alepensato Yes, absolutely.  I believe it is not a bug of camera driver or SDK itself.

I don't see anything suspicious in the EKOS UI in my ubuntu 22.04 environment.
I need to consider migrating to 64bit, but I'm frustrated because I can't install them straight to bullseye with apt.

alepensato, 2022-11-25:
@murveit Could you get involved in this issue???

alepensato, 2022-11-30:
?????

alepensato, 2022-12-01:
updated to latest Kstars 3.6.2 always from https://ppa.stellarmate.com/ repos, but with same results, system unusable

hiro3110i, 2022-12-02:
@alepensato I had created and tested brand new 64bit environment with Ubuntu Mate 22.04 LTE on RasPi4.
In addition, I built 64bit Ubuntu 22.04 LTE with GNOME as a virtual machine on my MacBook Air and confirmed its operation.
I'm using Neputune-C II, however both of environment are working very well, including working load and operation response.
By implementing various performance tuning on RasPi4, I was able to build a 64bit environment that operates more comfortably than the conventional 32bit version.
The KStars and indi repository I installed is ppa:mutlaqja/ppa.

I'm not sure it's a Sedna-specific issue or not.
I'm still waiting for a response from the manufacturer.
I've attached some screenshots for reference (please excuse my Japanese environment)

[image]

[image]

[image]

[image]

alepensato, 2022-12-02:
@hiro3110i 
Hi,
I made few days ago a test installing UbuntuMate on my Raspi and I used the ppa:mutlaqja/ppa repository, but with the same issue.
I made test with 2 differents Raspi 4 and conditions of OS, cables and Sedna cameras (I have 2 Sedna).
In all of these the Sedna works correctly ony if exposure time is 0.5sec or less, but it works ant any exposure time if you use PHD2.

hiro3110i, 2022-12-02:
@alepensato Uhm, the difference is RasPi4 PCB and camera product.
I heard that the RasPi4 chipset has changed.  My RasPi4 is an early (old)  model.
Also, the firmware version of the RasPi4 may have some impact on USB3.0/PCIe performance.

In terms of data transfer, Neptune-C II is a 2712x1538 pixels color sensor, so 16,684,224 bytes/frame for RAW16.
Sedna-M is monochrome and 3096x2078 pixels, so 12,866,976 bytes/frame for RAW16. Not so much difference...

alepensato, 2022-12-02:
My Raspy are Models 1.1 as I remember and also I tried with different firmware, but the problem is not the speed trasnfer or chipset, becase the same hardware works with PHD2. If it was a firmware/chipset problem, it should not works in any conditions

hiro3110i, 2022-12-03:
@alepensato Can we conclude that this is a KStars or Ekos issue and not an indi driver bug?
You may find more effective solution if you discuss it on KStars gitlab instead of here.
https://invent.kde.org/education/kstars

alepensato, 2022-12-03:
@hiro3110i 
In my opinion, yes it is, because if it was an indi releted problem..... it shoult not works in PHD2

alepensato, 2022-12-05:
[   68.026873] usb 2-1: new SuperSpeed USB device number 2 using xhci_hcd
[   68.047933] usb 2-1: New USB device found, idVendor=a0a0, idProduct=1783, bcdDevice= 0.00
**[   68.047952] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[   68.047959] usb 2-1: Product: Sedna-M
[   68.047964] usb 2-1: Manufacturer: Player One
[  102.698862] usb 2-1: usbfs: interface 0 claimed by usbfs while 'indi_playerone_' sets config #1
[  103.772208] usb 2-1: usbfs: interface 0 claimed by usbfs while 'indi_playerone_' sets config #1**
[  173.163980] usb 2-1: reset SuperSpeed USB device number 2 using xhci_hcd

hiro3110i, 2022-12-14:
Since then I have continued to analyze this issue.
We can see so many similar problems when we search on google.
It seems to be a major problem common to USB devices.

One possibility is the behavior of LPM (Link Power Management).
What I found strange is the fact that longer shooting intervals are more likely to cause problems.
Not a problem at high frame rates where USB bus bandwidth is critical.
When the shooting interval is extended, the USB bus becomes idle and shifts to low power mode.
If this is the essential factor of this problem, it may solve when we can disable LPM for power saving.

The manufacturer sent me a test sample of the Sedna-M to analyze the problem.
I'm waiting the arrival and try to reproduce the phenomenon and countermeasures.

I'm just an astrophotography fan.
Thanks to the manufacturer who are always open and cooperative.

alepensato, 2022-12-14:
@hiro3110i 
Thanks for your interest, I tried some USB 3 cables but with no luck.
The strange thing is that the problem seems to be present only in Ekos, but works in PHD2

knro, 2022-12-14:
@alepensato PHD2 using native driver or INDI driver?

alepensato, 2022-12-14:
> @alepensato PHD2 using native driver or INDI driver?

PHD2 under Raspberry PI, connected with INDI driver, with RaspbianOS 64bit

PHD2 does not have native driver for PlayerOne or it is not listes as QHY, ZWO etc

hiro3110i, 2022-12-14:
@alepensato I'm not sure, but I suspect that the internal guider and PHD2 have different access timings/patterns to the USB bus.  USB bus reset can also be caused by another device connected to the same hub.  We'll need to disconnect the other devices and look at the Sedna-M alone.  If there is an easy way to disable LPM, we can test my hypothesis.

knro, 2022-12-14:
If using the INDI driver, then it shouldn't matter which client. It's odd that PHD2 works but Ekos doesn't connected to the same INDI driver!

alepensato, 2022-12-14:
@knro 
The INDI panel has been untouched between tests with Ekos or PHD2. The problem is that when you use internal guider and set an exposure time of at least 1 second, it freezes after you click on Abort and Loop again. But if you set an exposure time of 0.5 seconds, it works after you click on Abort and Loop again.
PHD2 works with any exposure time or binning

knro, 2022-12-14:
what if you go to INDI Control panel and set the exposure there (in KStars)? does it freeze at 1 second?

alepensato, 2022-12-14:
it freezes only after the first Loop and Abort, but if after the first Loop and Abort disconnect or restart KStarts/Ekos/INDI and use only the INDI panel it works at any exposure time.

If you want take a look, I can invite you to enter into the system using VNC

knro, 2022-12-14:
Sure let's check it out with VNC.

alepensato, 2022-12-14:
ok we just sync for the time, I can give you access to the system this evening starting from the 8.00PM GMT+2 Rome Italy time zone

hiro3110i, 2022-12-15:
The manufacturer's SDK was updated to v3.1.0 yesterday.
I don't know if this problem will be solved, but I just sent a pull request for a driver with new SDK.
At least, it has been confirmed that it works properly in the Ubuntu Mate 64bit environment of Raspberry Pi4/4GB.

alepensato, 2022-12-15:
So, we are unlucky.
I changed the SDK with the new one downloaded from Playerone's site, but no good news, the problem is always here.

[image]

Giosi62, 2022-12-18:
On Ubuntu 22.04 LTS, KStars 3.6.2 stable on Lenovo T430; PlayerOne Sedna-M whit SDK 3.1.0 if I interrupt the exposure with exposure times of 1 second or more, the cicle it doesn't start again
[image]

hiro3110i, 2022-12-19:
@Giosi62 Thank you for reporting.  I'll test with Sedna-M as soon as it arrives.
By the way, does this issue occur in the Capture tab as well as in the Guide tab?

alepensato, 2022-12-19:
@hiro3110i 
I tested it in Guide and Focus tab, with the same problem. I do not remember if i tested it in Capture tab also

Giosi62, 2022-12-19:
@hiro3110i
Hi Hiro3110i, Yes the same problem in the capture card too. After the first stop he takes just one frame and stops

hiro3110i, 2022-12-23:
I have received a Sedna-M and have confirmed the issues you mentioned.
In my Raspberry Pi4/4GB, I've seen same issues on both 32bit Astroberry and 64bit Ubuntu Mate.
This issue was not observed on the Neptune-C II, only prominently on the Sedna-M.
Simply rolling back the manufacturer's SDK to v3.0.3 made this issue disappear.
I'm feeding this result back to the manufacturer and requesting their confirmation.

SDK v3.0.3 is compatible with indi_playerone_ccd driver v0.8 and v0.9.
You can simply replace only SDK like below (for 64bit system):

`sudo cp libPlayerOneCamera.so.3.0.3 /usr/lib/aarch64-linux-gnu/.`
`cd /usr/lib/aarch64-linux-gnu`
`sudo ln -s -f libPlayerOneCamera.so.3.0.3 libPlayerOneCamera.so.3`

flyleier, 2022-12-23:
> I have received a Sedna-M and have confirmed the issues you mentioned. In my Raspberry Pi4/4GB, I've seen same issues on both 32bit Astroberry and 64bit Ubuntu Mate. This issue was not observed on [...]

V3.0.3 download URL: [https://player-one-astronomy.com/download/softwares/PlayerOne_Camera_SDK_RaspberryPI_V3.0.3.tar.gz](https://player-one-astronomy.com/download/softwares/PlayerOne_Camera_SDK_RaspberryPI_V3.0.3.tar.gz)

alepensato, 2022-12-23:
mmmm let me check if this solution works. i remember that I tried this with no luck

alepensato, 2022-12-23:
i made a quick test test and seems that it works. i will check again later with multiple exposure times and binning, just for stress the system.
but the problem still persists at the next update, could you imagine where the problem  could be?

hiro3110i, 2022-12-23:
@alepensato Thank you for the additional test.  Of course, SDK rollback is a temporary.
The operation of interrupting exposure is always a headache for cameras, I guess.
In the unlikely event that control is not returned to software, it may be worth considering using a bus reset to recover.
However, it's an emergency hard landing, and we should consider a milder return to the normal state.
It would be ideal to reset and return only the internal state of the camera, excluding the USB bus.
I trust that the manufacturer's engineers will consider a better method.

hiro3110i, 2023-01-20:
This issue was fixed by the latest camera SDK v3.1.1 which was released on Jan 18th.
I have tested with Poseidon-C/Uranus-C/Neptune-C II/Sedna-M on my Raspberry Pi4 (64bit Ubuntu mate 22.04).

@alepensato @Giosi62 Could you please check it on your side?
You can download the latest camera SDK from [here](https://player-one-astronomy.com/service/software/).
And you can simply replace the camera SDK only if you're using indi v1.9.9.
For linux system, 
`sudo cp libPlayerOneCamera.so.3.1.1 /usr/lib/aarch64-linux-gnu/.`
`cd /usr/lib/aarch64-linux-gnu`
`sudo ln -s -f libPlayerOneCamera.so.3.1.1 libPlayerOneCamera.so.3`

alepensato, 2023-01-20:
@hiro3110i 
I will try this solution later when I finish to work

alepensato, 2023-01-20:
@hiro3110i  @Giosi62 
I confirm you that with the new SDK 3.1.1, the problem with the Sedna is NOT present and all seems to works correctly.
I hope now that the this SDK will be provided as regular system update

giosi, 2023-01-22:
> Questo problema è stato risolto dall'ultimo SDK della fotocamera v3.1.1, rilasciato il 18 gennaio. Ho provato con Poseidon-C/Uranus-C/Neptune-C II/Sedna-M sul mio Raspberry Pi4 (64bit Ubuntu mate 22.04).
> 
> @alepensato @Giosi62Potresti per favore controllarlo dalla tua parte? È possibile scaricare l'SDK della fotocamera più recente da [qui](https://player-one-astronomy.com/service/software/) . E puoi semplicemente sostituire l'SDK della fotocamera solo se stai utilizzando indi v1.9.9. Per il sistema Linux, `sudo cp libPlayerOneCamera.so.3.1.1 /usr/lib/aarch64-linux-gnu/.` `cd /usr/lib/aarch64-linux-gnu` `sudo ln -s -f libPlayerOneCamera.so.3.1.1 libPlayerOneCamera.so.3`

Hi, for me too new SDK 3.1.1 works without problems

hiro3110i, 2023-01-23:
@alepensato @giosi The latest camera SDK v3.1.1 will be default in next indilib update (maybe v2.0.0).
I think this topic can be closed. Thank you very much for your kind support !

closed 2023-02-21 by knro

alepensato, 2025-02-13:
@hiro3110i @knro the original problem is again alive, now with Ubuntu or Mint installed on a 64bit Ryzen processor.

It is present in the Guide tab, I tried with different seconds of exposure and seems that with a value lower than 1sec the looping is ok, but if you use 1sec or more.... the loop hangs and take ony the first capture.

I tried to replace the lated libplayerone with previous versions, but the problem is always present.

The problem does not occour on my Raspbian 64bit on my Rasberry 4