indilib/indi-3rdparty #688: INDI ASI CCD Driver restart after 2nd take exposure

nicomedia, 2022-11-28:
**Describe the bug**
I have custom indi client application. When working with "indi_asi_ccd", i could collect image data  over newBLOB function if i call set exposure function. When i call set exposure again driver restarts without error message (newBLOB function should be called instead). Problem is valid for video stream as well. If i call video stream after set exposure function, driver restarts. However, if i start video stream, i could collect data at newBLOB function periodically.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run indi_asi_ccd driver
2. Run indi client
3. Set exposure
4. Collect data over newBLOB function
5. Call set exposure again
6. Driver restarts

**Expected behavior**
Collect data over newBLOB function again.

**Log File** 
[indiAsiLog.txt](https://github.com/indilib/indi-3rdparty/files/10103334/indiAsiLog.txt)

knro, 2022-11-28:
Can you provide a sample client to demonstrate this?

nicomedia, 2022-11-29:
Hello knro,

Thank you for your quick answer. I attached very very simple example. 
Please run indiserver with "indiserver -v indi_asi_ccd" if you increase verbosity, there is actually another bug (or debug information) in indiserver or indi_asi_ccd driver that prints whole image data (blob) to the terminal. (I could not have a chance to look where it prints so far). 

[client_sample.zip](https://github.com/indilib/indi-3rdparty/files/10110003/client_sample.zip)

# indiserver -v indi_asi_ccd

# cd build
# indi_client_sample

knro, 2022-11-29:
@pawel-soja Can you please check if this is related to the latest changes?

@nicomedia Can you confirm this is built from INDI GIT?

pawel-soja, 2022-11-29:
Sure, I'll check it out today

nicomedia, 2022-11-29:
> @pawel-soja Can you please check if this is related to the latest changes?
> 
> @nicomedia Can you confirm this is built from INDI GIT?

I have tested it with v1.9.8 - v1.9.7 and latest night build both for indi-core and indi-3rdparty. Problem seems to be valid for all versions.

pawel-soja, 2022-11-29:
I cannot reproduce the problem.
It is possible that the problem is related to the fact that you have a 32-bit build.
@nicomedia Are you able to check if the problem persists on a 64-bit machine as well?

nicomedia, 2022-11-30:
Unfortunately i dont have an environment for 64-bit. I am working remotely on 32-bit raspberrry pi system on which camera is connected. I checked the ZWO SDK and there is new driver version libASICamera2.so.1.27, i will try my luck with it. By the way, i saw new pull request about 32-bit support : https://github.com/indilib/indi/pull/1775 . Is this pull request related with this issue @pawel-soja ?

pawel-soja, 2022-12-04:
I ran tests on RPi4 and it looks ok.
Did you test on a fresh system? Maybe you have some remnants of older libraries and something is linking wrong.
- Linux raspberrypi 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux
- Camera: ZWO CCD ASI120MM Mini
- `INDI Core` https://github.com/indilib/indi/tree/fix/debian-i386
- `INDI 3rd-party` https://github.com/indilib/indi-3rdparty/tree/b61189f85bf26926195f90328dc52092c19de4e9 (master latest)
- server command `indiserver indi_asi_ccd`

I ran your program with only changing the name of the camera

```
pi@raspberrypi:~/Projects/issues/client_sample/build $ make -j2 && ./indi_client_sample 
Scanning dependencies of target indi_client_sample
[ 33%] Building CXX object CMakeFiles/indi_client_sample.dir/dds_indi_client.o
[ 66%] Linking CXX executable indi_client_sample
[100%] Built target indi_client_sample
INDI::BaseClient::connectServer: creating new connection...
###serverConnected : 
###newDevice function called ZWO CCD ASI120MM Mini
###newProperty function called : CONNECTION - 1
###newProperty function called : DRIVER_INFO - 2
###newProperty function called : POLLING_PERIOD - 0
###newProperty function called : DEBUG - 1
###newProperty function called : SIMULATION - 1
###newProperty function called : CONFIG_PROCESS - 1
###newProperty function called : DEBUG_LEVEL - 1
###newProperty function called : LOGGING_LEVEL - 1
###newProperty function called : LOG_OUTPUT - 1
###newProperty function called : ACTIVE_DEVICES - 2
###newProperty function called : CCD_EXPOSURE - 0
EXPOSURE : 0.000000
###newProperty function called : CCD_ABORT_EXPOSURE - 1
ABORT EXPOSURE : 0
###newProperty function called : CCD_FRAME - 0
###newProperty function called : CCD_FRAME_RESET - 1
###newProperty function called : CCD_BINNING - 0
###newProperty function called : FITS_HEADER - 2
###newProperty function called : CCD_CAPTURE_FORMAT - 1
###newProperty function called : CCD_TRANSFER_FORMAT - 1
###newProperty function called : CCD_INFO - 0
###newProperty function called : CCD_COMPRESSION - 1
###newProperty function called : CCD1 - 4
INDICameraBlobName : CCD1 INDICameraName : ZWO CCD ASI120MM Mini PropName : CCD1
HAS BLOB
###newProperty function called : TELESCOPE_TIMED_GUIDE_NS - 0
###newProperty function called : TELESCOPE_TIMED_GUIDE_WE - 0
###newProperty function called : CCD_FRAME_TYPE - 1
###newProperty function called : SCOPE_INFO - 0
###newProperty function called : WCS_CONTROL - 1
###newProperty function called : UPLOAD_MODE - 1
###newProperty function called : UPLOAD_SETTINGS - 2
###newProperty function called : CCD_FAST_TOGGLE - 1
###newProperty function called : CCD_FAST_COUNT - 0
###newProperty function called : CCD_VIDEO_STREAM - 1
###newProperty function called : STREAM_DELAY - 0
###newProperty function called : STREAMING_EXPOSURE - 0
###newProperty function called : FPS - 0
###newProperty function called : RECORD_STREAM - 1
###newProperty function called : RECORD_FILE - 2
###newProperty function called : RECORD_OPTIONS - 0
###newProperty function called : CCD_STREAM_FRAME - 0
###newProperty function called : CCD_STREAM_ENCODER - 1
###newProperty function called : CCD_STREAM_RECORDER - 1
###newProperty function called : LIMITS - 0
###newProperty function called : CCD_TEMPERATURE - 0
###newProperty function called : CCD_CONTROLS - 0
###newProperty function called : CCD_CONTROLS_MODE - 1
###newProperty function called : FLIP - 1
###newProperty function called : CCD_VIDEO_FORMAT - 1
###newProperty function called : BLINK - 0
[... 133 lines cut ...]
Recveing message from Server:

########################
2022-12-04T00:31:05: [DEBUG] Upload complete 
########################

###newNumber function called name :  CCD_EXPOSURE - value : 0 - state : 1
Receving new CCD Exposure: 0 
###newNumber function called name :  CCD_TEMPERATURE - value : 23 - state : 0
###newNumber function called name :  CCD_TEMPERATURE - value : 23.7 - state : 0
###newNumber function called name :  CCD_TEMPERATURE - value : 23 - state : 0
###newNumber function called name :  CCD_TEMPERATURE - value : 23.7 - state : 0
###newNumber function called name :  CCD_TEMPERATURE - value : 23 - state : 0
###newNumber function called name :  CCD_TEMPERATURE - value : 23.7 - state : 0
###newNumber function called name :  CCD_TEMPERATURE - value : 23 - state : 0
###newNumber function called name :  CCD_TEMPERATURE - value : 23.7 - state : 0
###newNumber function called name :  CCD_TEMPERATURE - value : 23 - state : 0
###newNumber function called name :  CCD_TEMPERATURE - value : 23.7 - state : 0
###newNumber function called name :  CCD_TEMPERATURE - value : 23 - state : 0
###newNumber function called name :  CCD_TEMPERATURE - value : 23.7 - state : 0
###newNumber function called name :  CCD_TEMPERATURE - value : 23 - state : 0
###newNumber function called name :  CCD_TEMPERATURE - value : 23.7 - state : 0
###newNumber function called name :  CCD_TEMPERATURE - value : 23 - state : 0
###newNumber function called name :  CCD_TEMPERATURE - value : 23.7 - state : 0
```

nicomedia, 2022-12-04:
Great, i am waiting for you to finalize Pull Request. I am going to try it. I tried before but i always get missing symbol errors when i called "indiserver indi_asi_ccd".

knro, 2023-02-21:
I cannot reproduce this issue with latest SDK + driver. Will consider this issue fixed. Please re-open if otherwise.

closed 2023-02-21 by knro