indilib/indi-3rdparty #690: Mount Lockup with Home Commands under Main Control Tab

4thbit, 2022-12-08:
Currently have an Advanced GT with Nexstar+ firmware 5.31 and MC firmware 5.20.  When I hit any of the home buttons under "Main Control" The mount will lockup and need power cycled.  It constantly appears to be returning something unknown as seen below. 

2022-12-08T17:14:32: [ERROR] RES 
2022-12-08T17:14:32: [ERROR] Resp. char 4 is 03 ascii 

This repeats until the mount is rebooted.

knro, 2022-12-09:
This is with Celestron GPS driver? which buttons? screenshot?

4thbit, 2022-12-09:
This is with the Celestron Aux driver in HC passthrough mode.  Screens attached. 

[image]
[image]

knro, 2023-01-09:
Ok, what would a fool-proof away of preventing this? Does this affect certain mounts or certain firmware versions?