indilib/indi-3rdparty #699: indi-toupbase driver: Add control for new "high full well mode" with touptek SDK v53

42-max, 2023-01-08:
In the Indi Forum user LANCO announced a new SDK Version for the Touptek cameras. The new version offers a new high full well option. This option should be included as a control in thre INDI driver.

My proposal is to include "Enable" and "Disable" buttons into the Control Tab just below the "Low Noise Mode" option.

If nobody has started the implementation yet I will start soon...   Please let me know if you have further infomation about this mode or any comments.

CS Max

knro, 2023-01-09:
It appears you have a PR in progress? Please submit when ready, it would be greatly appreciated!

knro, 2023-02-21:
@touptek is this planned for the driver?

knro, 2023-02-21:
IIRC this is already merged? Please re-open if not the case.

closed 2023-02-21 by knro