indilib/indi-3rdparty #712: indi-libcamera fails to compile

hamarituc, 2023-01-31:
**Describe the bug**

The `libcamera` driver doesn't compile since version 2.0.0. It fails with the following error

```
error: ‘INDI::PropertyView<T>* INDI::PropertyBasic<T>::operator&() [with T = _ISwitch]’ is protected within this context
```

Version 1.9.9 compiled successfully.

**To Reproduce**

1. Run CMake for `indi-libcamera`
2. Run `make`

**Expected behavior**

The drivers compiles.

**Screenshots**

N/A

**Desktop (please complete the following information):**

 - OS: Linux (Gentoo)
 - Version 2.0.0

**Log Files**

Compile logs:

```
>>> Configuring source in /var/tmp/portage/sci-libs/indilib-driver-libcamera-2.0.0/work/indi-3rdparty-2.0.0/indi-libcamera ...
 * Source directory (CMAKE_USE_DIR): "/var/tmp/portage/sci-libs/indilib-driver-libcamera-2.0.0/work/indi-3rdparty-2.0.0/indi-libcamera"
 * Build directory  (BUILD_DIR):     "/var/tmp/portage/sci-libs/indilib-driver-libcamera-2.0.0/work/indi-3rdparty-2.0.0/indi-libcamera_build"
cmake -C /var/tmp/portage/sci-libs/indilib-driver-libcamera-2.0.0/work/indi-3rdparty-2.0.0/indi-libcamera_build/gentoo_common_config.cmake -G Ninja -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_TOOLCHAIN_FILE=/var/tmp/portage/sci-libs/indilib-driver-libcamera-2.0.0/work/indi-3rdparty-2.0.0/indi-libcamera_build/gentoo_toolchain.cmake /var/tmp/portage/sci-libs/indilib-driver-libcamera-2.0.0/work/indi-3rdparty-2.0.0/indi-libcamera
loading initial cache file /var/tmp/portage/sci-libs/indilib-driver-libcamera-2.0.0/work/indi-3rdparty-2.0.0/indi-libcamera_build/gentoo_common_config.cmake
-- The CXX compiler identification is GNU 12.2.1
-- The C compiler identification is GNU 12.2.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/x86_64-pc-linux-gnu-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/x86_64-pc-linux-gnu-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found CFITSIO 4.2.0: /usr/lib64/libcfitsio.so
-- Found INDI: /usr/lib64/libindidriver.so;/usr/lib64/libindiAlignmentDriver.so (found version "2.0.0") 
-- Found ZLIB: /usr/lib64/libz.so (found version "1.2.13") 
-- Found PkgConfig: /usr/bin/x86_64-pc-linux-gnu-pkg-config (found version "1.8.0") 
-- Found USB1: /usr/lib64/libusb-1.0.so (found version "1.0.26") 
-- Performing Test USB1_HAS_LIBUSB_ERROR_NAME
-- Performing Test USB1_HAS_LIBUSB_ERROR_NAME - Success
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Checking for module 'libraw'
--   Found libraw, version 0.21.1
-- Checking for module 'libraw_r'
--   Found libraw_r, version 0.21.1
-- Found LibRaw: /usr/lib64/libraw.so (found version "0.21.1") 
-- Found JPEG: /usr/lib64/libjpeg.so  
-- Found Boost: /usr/lib64/cmake/Boost-1.81.0/BoostConfig.cmake (found version "1.81.0") found components: program_options 
-- Checking for module 'libcamera'
--   Found libcamera, version 0.0.3
-- libcamera library found:
[... 114 lines cut ...]
/usr/include/libindi/indipropertybasic.h:101:27: note: declared protected here
  101 |         PropertyView<T> * operator &();
      |                           ^~~~~~~~
/var/tmp/portage/sci-libs/indilib-driver-libcamera-2.0.0/work/indi-3rdparty-2.0.0/indi-libcamera/indi_libcamera.cpp:972:29: error: ‘INDI::PropertyView<T>* INDI::PropertyBasic<T>::operator&() [with T = _ISwitch]’ is protected within this context
  972 |     IUSaveConfigSwitch(fp, &AdjustMeteringModeSP);
      |                             ^~~~~~~~~~~~~~~~~~~~
/usr/include/libindi/indipropertybasic.h:101:27: note: declared protected here
  101 |         PropertyView<T> * operator &();
      |                           ^~~~~~~~
/var/tmp/portage/sci-libs/indilib-driver-libcamera-2.0.0/work/indi-3rdparty-2.0.0/indi-libcamera/indi_libcamera.cpp:973:29: error: ‘INDI::PropertyView<T>* INDI::PropertyBasic<T>::operator&() [with T = _ISwitch]’ is protected within this context
  973 |     IUSaveConfigSwitch(fp, &AdjustDenoiseModeSP);
      |                             ^~~~~~~~~~~~~~~~~~~
/usr/include/libindi/indipropertybasic.h:101:27: note: declared protected here
  101 |         PropertyView<T> * operator &();
      |                           ^~~~~~~~
/var/tmp/portage/sci-libs/indilib-driver-libcamera-2.0.0/work/indi-3rdparty-2.0.0/indi-libcamera/indi_libcamera.cpp: In member function ‘void INDILibCamera::workerExposure(const std::atomic_bool&, float)’:
/var/tmp/portage/sci-libs/indilib-driver-libcamera-2.0.0/work/indi-3rdparty-2.0.0/indi-libcamera/indi_libcamera.cpp:390:21: warning: ignoring return value of ‘ssize_t read(int, void*, size_t)’ declared with attribute ‘warn_unused_result’ [-Wunused-result]
  390 |                 read(fd, memptr, memsize);
      |                 ~~~~^~~~~~~~~~~~~~~~~~~~~
[5/40] /usr/bin/x86_64-pc-linux-gnu-g++ -DLIBAV_PRESENT=0 -D_FILE_OFFSET_BITS=64 -I/var/tmp/portage/sci-libs/indilib-driver-libcamera-2.0.0/work/indi-3rdparty-2.0.0/indi-libcamera_build -I/var/tmp/portage/sci-libs/indilib-driver-libcamera-2.0.0/work/indi-3rdparty-2.0.0/indi-libcamera -I/usr/include/libindi -I/usr/include/libraw -I/usr/include/libcamera -I/var/tmp/portage/sci-libs/indilib-driver-libcamera-2.0.0/work/indi-3rdparty-2.0.0/indi-libcamera/libcamera-apps  -march=native -O2 -pipe -D_FORTIFY_SOURCE=2 -fstack-protector-all -fPIE -O1 -Wa,--noexecstack  -Wall -Wextra -Wno-unused-but-set-variable -Wno-format-truncation -g -fPIC -Wall -Wextra -pedantic -Wno-unused-parameter -faligned-new -Werror -Wfatal-errors -Wno-psabi -std=gnu++17 -MD -MT SYSTEM/encoder/CMakeFiles/encoders.dir/h264_encoder.cpp.o -MF SYSTEM/encoder/CMakeFiles/encoders.dir/h264_encoder.cpp.o.d -o SYSTEM/encoder/CMakeFiles/encoders.dir/h264_encoder.cpp.o -c /var/tmp/portage/sci-libs/indilib-driver-libcamera-2.0.0/work/indi-3rdparty-2.0.0/indi-libcamera/libcamera-apps/encoder/h264_encoder.cpp
[6/40] /usr/bin/x86_64-pc-linux-gnu-g++ -DLIBAV_PRESENT=0 -D_FILE_OFFSET_BITS=64 -I/var/tmp/portage/sci-libs/indilib-driver-libcamera-2.0.0/work/indi-3rdparty-2.0.0/indi-libcamera_build -I/var/tmp/portage/sci-libs/indilib-driver-libcamera-2.0.0/work/indi-3rdparty-2.0.0/indi-libcamera -I/usr/include/libindi -I/usr/include/libraw -I/usr/include/libcamera -I/var/tmp/portage/sci-libs/indilib-driver-libcamera-2.0.0/work/indi-3rdparty-2.0.0/indi-libcamera/libcamera-apps  -march=native -O2 -pipe -D_FORTIFY_SOURCE=2 -fstack-protector-all -fPIE -O1 -Wa,--noexecstack  -Wall -Wextra -Wno-unused-but-set-variable -Wno-format-truncation -g -fPIC -Wall -Wextra -pedantic -Wno-unused-parameter -faligned-new -Werror -Wfatal-errors -Wno-psabi -std=gnu++17 -MD -MT SYSTEM/encoder/CMakeFiles/encoders.dir/null_encoder.cpp.o -MF SYSTEM/encoder/CMakeFiles/encoders.dir/null_encoder.cpp.o.d -o SYSTEM/encoder/CMakeFiles/encoders.dir/null_encoder.cpp.o -c /var/tmp/portage/sci-libs/indilib-driver-libcamera-2.0.0/work/indi-3rdparty-2.0.0/indi-libcamera/libcamera-apps/encoder/null_encoder.cpp
[7/40] /usr/bin/x86_64-pc-linux-gnu-g++ -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_NO_LIB -DLIBDRM_PRESENT=0 -DLIBEGL_PRESENT=0 -DOPENCV_PRESENT=1 -DQT_PRESENT=0 -D_FILE_OFFSET_BITS=64 -I/var/tmp/portage/sci-libs/indilib-driver-libcamera-2.0.0/work/indi-3rdparty-2.0.0/indi-libcamera_build -I/var/tmp/portage/sci-libs/indilib-driver-libcamera-2.0.0/work/indi-3rdparty-2.0.0/indi-libcamera -I/usr/include/libindi -I/usr/include/libraw -I/usr/include/libcamera -I/var/tmp/portage/sci-libs/indilib-driver-libcamera-2.0.0/work/indi-3rdparty-2.0.0/indi-libcamera/libcamera-apps  -march=native -O2 -pipe -D_FORTIFY_SOURCE=2 -fstack-protector-all -fPIE -O1 -Wa,--noexecstack  -Wall -Wextra -Wno-unused-but-set-variable -Wno-format-truncation -g -fPIC -Wall -Wextra -pedantic -Wno-unused-parameter -faligned-new -Werror -Wfatal-errors -Wno-psabi -std=gnu++17 -MD -MT SYSTEM/core/CMakeFiles/libcamera_app.dir/post_processor.cpp.o -MF SYSTEM/core/CMakeFiles/libcamera_app.dir/post_processor.cpp.o.d -o SYSTEM/core/CMakeFiles/libcamera_app.dir/post_processor.cpp.o -c /var/tmp/portage/sci-libs/indilib-driver-libcamera-2.0.0/work/indi-3rdparty-2.0.0/indi-libcamera/libcamera-apps/core/post_processor.cpp
[... cut ...] n -g -fPIC -Wall -Wextra -pedantic -Wno-unused-parameter -faligned-new -Werror -Wfatal-errors -Wno-psabi -std=gnu++17 -MD -MT SYSTEM/core/CMakeFiles/libcamera_app.dir/libcamera_app.cpp.o -MF SYSTEM/core/CMakeFiles/libcamera_app.dir/libcamera_app.cpp.o.d -o SYSTEM/core/CMakeFiles/libcamera_app.dir/libcamera_app.cpp.o -c /var/tmp/portage/sci-libs/indilib-driver-libcamera-2.0.0/work/indi-3rdparty-2.0.0/indi-libcamera/libcamera-apps/core/libcamera_app.cpp
ninja: build stopped: subcommand failed.
```

hamarituc, 2023-02-01:
The same error can also be observed in the `indi-nut` driver.

```
>>> Configuring source in /var/tmp/portage/sci-libs/indilib-driver-nut-2.0.0/work/indi-3rdparty-2.0.0/indi-nut ...
 * Source directory (CMAKE_USE_DIR): "/var/tmp/portage/sci-libs/indilib-driver-nut-2.0.0/work/indi-3rdparty-2.0.0/indi-nut"
 * Build directory  (BUILD_DIR):     "/var/tmp/portage/sci-libs/indilib-driver-nut-2.0.0/work/indi-3rdparty-2.0.0/indi-nut_build"
cmake -C /var/tmp/portage/sci-libs/indilib-driver-nut-2.0.0/work/indi-3rdparty-2.0.0/indi-nut_build/gentoo_common_config.cmake -G Ninja -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=RelWithDebInfo -DCMAKE_TOOLCHAIN_FILE=/var/tmp/portage/sci-libs/indilib-driver-nut-2.0.0/work/indi-3rdparty-2.0.0/indi-nut_build/gentoo_toolchain.cmake /var/tmp/portage/sci-libs/indilib-driver-nut-2.0.0/work/indi-3rdparty-2.0.0/indi-nut
loading initial cache file /var/tmp/portage/sci-libs/indilib-driver-nut-2.0.0/work/indi-3rdparty-2.0.0/indi-nut_build/gentoo_common_config.cmake
-- The C compiler identification is GNU 12.2.1
-- The CXX compiler identification is GNU 12.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/x86_64-pc-linux-gnu-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/x86_64-pc-linux-gnu-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found INDI: /usr/lib64/libindidriver.so;/usr/lib64/libindiAlignmentDriver.so (found version "2.0.0") 
-- Found NUTCLIENT: /usr/lib64/libnutclient.so
-- Performing Test COMPATIBLE_FORTIFY_SOURCE
-- Performing Test COMPATIBLE_FORTIFY_SOURCE - Success
-- <<< Gentoo configuration >>>
Build type      RelWithDebInfo
Install path    /usr
Compiler flags:
C               -march=native -O2 -pipe -D_FORTIFY_SOURCE=2 -fstack-protector-all -fPIE -O1 -Wa,--noexecstack  -Wall -Wextra -Wno-unused-but-set-variable -Wno-format-truncation -g
C++             -g -std=c++0x -march=native -O2 -pipe -D_FORTIFY_SOURCE=2 -fstack-protector-all -fPIE -O1 -Wa,--noexecstack  -Wall -Wextra -Wno-unused-but-set-variable -Wno-format-truncation -g
Linker flags:
Executable      -Wl,-O1 -Wl,--as-needed  -Wl,-z,nodump -Wl,-z,noexecstack -Wl,-z,relro -Wl,-z,now -pie 
Module          -Wl,-O1 -Wl,--as-needed
Shared          -Wl,-O1 -Wl,--as-needed  -Wl,-z,nodump -Wl,-z,noexecstack -Wl,-z,relro -Wl,-z,now -pie 

-- Configuring done
-- Generating done
-- Build files have been written to: /var/tmp/portage/sci-libs/indilib-driver-nut-2.0.0/work/indi-3rdparty-2.0.0/indi-nut_build
>>> Source configured.
>>> Compiling source in /var/tmp/portage/sci-libs/indilib-driver-nut-2.0.0/work/indi-3rdparty-2.0.0/indi-nut ...
 * Source directory (CMAKE_USE_DIR): "/var/tmp/portage/sci-libs/indilib-driver-nut-2.0.0/work/indi-3rdparty-2.0.0/indi-nut"
 * Build directory  (BUILD_DIR):     "/var/tmp/portage/sci-libs/indilib-driver-nut-2.0.0/work/indi-3rdparty-2.0.0/indi-nut_build"
ninja -v -j5 -l0
[1/2] /usr/bin/x86_64-pc-linux-gnu-g++  -I/var/tmp/portage/sci-libs/indilib-driver-nut-2.0.0/work/indi-3rdparty-2.0.0/indi-nut_build -I/var/tmp/portage/sci-libs/indilib-driver-nut-2.0.0/work/indi-3rdparty-2.0.0/indi-nut -I/usr/include/libindi  -g -std=c++0x -march=native -O2 -pipe -D_FORTIFY_SOURCE=2 -fstack-protector-all -fPIE -O1 -Wa,--noexecstack  -Wall -Wextra -Wno-unused-but-set-variable -Wno-format-truncation -g -fPIE -std=gnu++14 -MD -MT CMakeFiles/indi_nut.dir/indi_nut.cpp.o -MF CMakeFiles/indi_nut.dir/indi_nut.cpp.o.d -o CMakeFiles/indi_nut.dir/indi_nut.cpp.o -c /var/tmp/portage/sci-libs/indilib-driver-nut-2.0.0/work/indi-3rdparty-2.0.0/indi-nut/indi_nut.cpp
FAILED: CMakeFiles/indi_nut.dir/indi_nut.cpp.o 
/usr/bin/x86_64-pc-linux-gnu-g++  -I/var/tmp/portage/sci-libs/indilib-driver-nut-2.0.0/work/indi-3rdparty-2.0.0/indi-nut_build -I/var/tmp/portage/sci-libs/indilib-driver-nut-2.0.0/work/indi-3rdparty-2.0.0/indi-nut -I/usr/include/libindi  -g -std=c++0x -march=native -O2 -pipe -D_FORTIFY_SOURCE=2 -fstack-protector-all -fPIE -O1 -Wa,--noexecstack  -Wall -Wextra -Wno-unused-but-set-variable -Wno-format-truncation -g -fPIE -std=gnu++14 -MD -MT CMakeFiles/indi_nut.dir/indi_nut.cpp.o -MF CMakeFiles/indi_nut.dir/indi_nut.cpp.o.d -o CMakeFiles/indi_nut.dir/indi_nut.cpp.o -c /var/tmp/portage/sci-libs/indilib-driver-nut-2.0.0/work/indi-3rdparty-2.0.0/indi-nut/indi_nut.cpp
/var/tmp/portage/sci-libs/indilib-driver-nut-2.0.0/work/indi-3rdparty-2.0.0/indi-nut/indi_nut.cpp: In destructor ‘virtual NetworkUPSToolsMonitor::~NetworkUPSToolsMonitor()’:
/var/tmp/portage/sci-libs/indilib-driver-nut-2.0.0/work/indi-3rdparty-2.0.0/indi-nut/indi_nut.cpp:46:5: warning: deleting object of polymorphic class type ‘nut::TcpClient’ which has non-virtual destructor might cause undefined behavior [-Wdelete-non-virtual-dtor]
   46 |     delete nutClient;
      |     ^~~~~~~~~~~~~~~~
/var/tmp/portage/sci-libs/indilib-driver-nut-2.0.0/work/indi-3rdparty-2.0.0/indi-nut/indi_nut.cpp: In member function ‘virtual void NetworkUPSToolsMonitor::ISGetProperties(const char*)’:
/var/tmp/portage/sci-libs/indilib-driver-nut-2.0.0/work/indi-3rdparty-2.0.0/indi-nut/indi_nut.cpp:94:21: error: ‘INDI::PropertyView<T>* INDI::PropertyBasic<T>::operator&() [with T = _IText]’ is protected within this context
   94 |     defineProperty(&nutMonitorUrl);
      |                     ^~~~~~~~~~~~~
[... cut ...]
/var/tmp/portage/sci-libs/indilib-driver-nut-2.0.0/work/indi-3rdparty-2.0.0/indi-nut/indi_nut.cpp:165:27: error: ‘INDI::PropertyView<T>* INDI::PropertyBasic<T>::operator&() [with T = _IText]’ is protected within this context
  165 |     IUSaveConfigText(fp, &nutMonitorUrl);
      |                           ^~~~~~~~~~~~~
/usr/include/libindi/indipropertybasic.h:101:27: note: declared protected here
  101 |         PropertyView<T> * operator &();
      |                           ^~~~~~~~
ninja: build stopped: subcommand failed.
```

knro, 2023-06-04:
Fixed in GIT.

closed 2023-06-04 by knro