indilib/indi-3rdparty #726: indi-qhy building with qhy SDK arm64 23-01-11 fails

SebNik1, 2023-02-13:
**Describe the bug**
QHY Indi Driver was crushing after the latest indi update so I tried building libqhy (this one succesfully) and afterwards indi-qhy, which failed due to "error: ‘FITSRecord’ is not a member of ‘INDI’". I got all the dependencies excpet the no longer available libdc1394-22. I also reinstalled fitsio as it seemt to be missing. I spent several hours now and I'd be very happy for any advice. I did the distclean to get rid of all old qhy sdk versions and installed the latest version 23-01-11 clean. 

**To Reproduce**
Exact steps to reproduce the behavior.
1. get indi-3rdparty
2. build libqhy and then indi-qhy
3. See error

**Expected behavior**
build of indi-qhy and beeing able to connect to my camera (268c)

**Desktop (please complete the following information):**
 - OS: Ubuntu Mate ond RPI4b 8GB
 - Version 22.04.1 LTS

**Log Files**
xx@xxxxx:~/Projects$ sudo apt-get -y install libnova-dev libcfitsio-dev libusb-1.0-0-dev zlib1g-dev libgsl-dev build-essential cmake git libjpeg-dev libcurl4-gnutls-dev libtiff-dev libfftw3-dev libftdi-dev libgps-dev libraw-dev libdc1394-22 libgphoto2-dev libboost-dev libboost-regex-dev librtlsdr-dev liblimesuite-dev libftdi1-dev libavcodec-dev libavdevice-dev
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
E: Paket libdc1394-22 kann nicht gefunden werden.
xx@xxxxx:~/Projects$ sudo apt-get -y install libnova-dev libcfitsio-dev libusb-1.0-0-dev zlib1g-dev libgsl-dev build-essential cmake git libjpeg-dev libcurl4-gnutls-dev libtiff-dev libfftw3-dev libftdi-dev libgps-dev libraw-dev libgphoto2-dev libboost-dev libboost-regex-dev librtlsdr-dev liblimesuite-dev libftdi1-dev libavcodec-dev libavdevice-dev
Paketlisten werden gelesen… Fertig
Abhängigkeitsbaum wird aufgebaut… Fertig
Statusinformationen werden eingelesen… Fertig
git ist schon die neueste Version (1:2.34.1-1ubuntu1.6).
libcurl4-gnutls-dev ist schon die neueste Version (7.81.0-1ubuntu1.7).
libtiff-dev ist schon die neueste Version (4.3.0-6ubuntu0.3).
zlib1g-dev ist schon die neueste Version (1:1.2.11.dfsg-2ubuntu9.2).
libgphoto2-dev ist schon die neueste Version (2.5.30+202301310850~ubuntu22.04.1).
build-essential ist schon die neueste Version (12.9ubuntu3).
cmake ist schon die neueste Version (3.22.1-1ubuntu1.22.04.1).
libavcodec-dev ist schon die neueste Version (7:4.4.2-0ubuntu0.22.04.1).
libavdevice-dev ist schon die neueste Version (7:4.4.2-0ubuntu0.22.04.1).
libboost-dev ist schon die neueste Version (1.74.0.3ubuntu7).
libboost-regex-dev ist schon die neueste Version (1.74.0.3ubuntu7).
libcfitsio-dev ist schon die neueste Version (4.0.0-1).
libfftw3-dev ist schon die neueste Version (3.3.8-2ubuntu8).
libftdi-dev ist schon die neueste Version (0.20-4ubuntu1).
libftdi1-dev ist schon die neueste Version (1.5-5build3).
libgps-dev ist schon die neueste Version (3.22-4ubuntu2).
libgsl-dev ist schon die neueste Version (2.7.1+dfsg-3).
libjpeg-dev ist schon die neueste Version (8c-2ubuntu10).
liblimesuite-dev ist schon die neueste Version (20.10.0+dfsg-5).
libnova-dev ist schon die neueste Version (0.16-5).
libraw-dev ist schon die neueste Version (0.20.2-2ubuntu2).
librtlsdr-dev ist schon die neueste Version (0.6.0-4).
libusb-1.0-0-dev ist schon die neueste Version (2:1.0.25-1ubuntu2).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
xx@xxxxx:~/Projects$ mkdir -p ~/Projects/build/indi-qhy
xx@xxxxx:~/Projects$ cd ~/Projects/build/indi-qhy
xx@xxxxx:~/Projects/build/indi-qhy$ cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi-3rdparty/indi-qhy
-- Found CFITSIO: /usr/lib/aarch64-linux-gnu/libcfitsio.so
-- Found libqhyccd: /usr/local/lib/libqhyccd.so
-- Found libnova: /usr/lib/aarch64-linux-gnu/libnova.so
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sn/Projects/indi-3rdparty/indi-qhy
xx@xxxxx:~/Projects/build/indi-qhy$ make -j4
Consolidate compiler generated dependencies of target qhy_ccd_test
Consolidate compiler generated dependencies of target qhy_video_test
Consolidate compiler generated dependencies of target indi_qhy_ccd
[ 16%] Linking CXX executable qhy_ccd_test
[ 16%] Linking CXX executable qhy_video_test
[ 16%] Building CXX object CMakeFiles/indi_qhy_ccd.dir/qhy_ccd.cpp.o
**/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvReleaseImage'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvMinMaxLoc'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvCopy'
/usr/bin/ld: /usr/local/lib/libqhyccd.so: undefined reference to `cvSetImageROI'
[... 114 lines cut ...]
      |                      ^~~~~~~~~
/home/sn/Projects/indi-3rdparty/indi-qhy/qhy_ccd.cpp:2779:22: error: request for member ‘push_back’ in ‘fitsKeywords’, which is of non-class type ‘int’
 2779 |         fitsKeywords.push_back({"GPS_NU", GPSHeader.now_us, 3, "NowShutterMicroSeconds"});
      |                      ^~~~~~~~~
/home/sn/Projects/indi-3rdparty/indi-qhy/qhy_ccd.cpp:2781:22: error: request for member ‘push_back’ in ‘fitsKeywords’, which is of non-class type ‘int’
 2781 |         fitsKeywords.push_back({"GPS_NT", GPSDataStartT[GPS_DATA_NOW_TS].text, "NowShutterTime"});
      |                      ^~~~~~~~~
/home/sn/Projects/indi-3rdparty/indi-qhy/qhy_ccd.cpp:2784:22: error: request for member ‘push_back’ in ‘fitsKeywords’, which is of non-class type ‘int’
 2784 |         fitsKeywords.push_back({"GPS_PPSC", GPSHeader.max_clock, "PPSCounter"});
      |                      ^~~~~~~~~
/home/sn/Projects/indi-3rdparty/indi-qhy/qhy_ccd.cpp:2795:22: error: request for member ‘push_back’ in ‘fitsKeywords’, which is of non-class type ‘int’
 2795 |         fitsKeywords.push_back({"GPS_LONG", GPSHeader.longitude, "GPS Longitude"});
      |                      ^~~~~~~~~
/home/sn/Projects/indi-3rdparty/indi-qhy/qhy_ccd.cpp:2798:22: error: request for member ‘push_back’ in ‘fitsKeywords’, which is of non-class type ‘int’
 2798 |         fitsKeywords.push_back({"GPS_LAT", GPSHeader.latitude, "GPS Latitude"});
      |                      ^~~~~~~~~
/home/sn/Projects/indi-3rdparty/indi-qhy/qhy_ccd.cpp:2801:22: error: request for member ‘push_back’ in ‘fitsKeywords’, which is of non-class type ‘int’
 2801 |         fitsKeywords.push_back({"GPS_SEQ", GPSHeader.seqNumber, "Sequence Number"});
      |                      ^~~~~~~~~
/home/sn/Projects/indi-3rdparty/indi-qhy/qhy_ccd.cpp:2804:22: error: request for member ‘push_back’ in ‘fitsKeywords’, which is of non-class type ‘int’
 2804 |         fitsKeywords.push_back({"GPS_TMP", GPSHeader.tempNumber, "Temporary Sequence Number"});
      |                      ^~~~~~~~~
make[2]: *** [CMakeFiles/indi_qhy_ccd.dir/build.make:76: CMakeFiles/indi_qhy_ccd.dir/qhy_ccd.cpp.o] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:87: CMakeFiles/indi_qhy_ccd.dir/all] Fehler 2
make: *** [Makefile:136: all] Fehler 2

knro, 2023-02-14:
You're using /usr/local so that's usually a red flag that it's getting compiled against the wrong library.

SebNik1, 2023-02-16:
Thank you very much for the advice, I remember something about setting the default lib folder. I need to reinstall linux due to several problems and I'll try afterwards.

closed 2023-02-21 by knro