indilib/indi-3rdparty #745: Celestron AUX driver remote control without HC

knro, 2023-02-28:
I sniffed via the DIY usbauxrelay MCU the AUX-bus while the HC connected. After switching on the HC the following AUX packets are sent:
```

[00806673ms,aux] Nexstar+ HC -> ALT MC, cmd MTR_GET_MODEL: AUX packet <0x3b, 0x03, 0x0d, 0x11, 0x05, 0xda> (checksum 0xda)
[00806820ms,aux] ALT MC -> Nexstar+ HC, cmd MTR_GET_MODEL: AUX packet <0x3b, 0x05, 0x11, 0x0d, 0x05, (0x14 0x85), 0x3f> (checksum 0x3f)
[00806953ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_GET_MODEL: AUX packet <0x3b, 0x03, 0x0d, 0x10, 0x05, 0xdb> (checksum 0xdb)
[00807099ms,aux] AZM MC -> Nexstar+ HC, cmd MTR_GET_MODEL: AUX packet <0x3b, 0x05, 0x10, 0x0d, 0x05, (0x14 0x85), 0x40> (checksum 0x40)
[00807232ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_GET_VERSION: AUX packet <0x3b, 0x03, 0x0d, 0x10, 0xfe, 0xe2> (checksum 0xe2)
[00807399ms,aux] AZM MC -> Nexstar+ HC, cmd NXS_GET_FW_VER: AUX packet <0x3b, 0x07, 0x10, 0x0d, 0xfe, (0x07 0x0f 0x20 0x4e), 0x5a> (checksum 0x5a)
[00807538ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_IS_APPROACH_DIR_NEG: AUX packet <0x3b, 0x03, 0x0d, 0x10, 0xfc, 0xe4> (checksum 0xe4)
[00807679ms,aux] AZM MC -> Nexstar+ HC, cmd MTR_IS_APPROACH_DIR_NEG: AUX packet <0x3b, 0x04, 0x10, 0x0d, 0xfc, (0x00), 0xe3> (checksum 0xe3)
[00807815ms,aux] Nexstar+ HC -> ALT MC, cmd MTR_IS_APPROACH_DIR_NEG: AUX packet <0x3b, 0x03, 0x0d, 0x11, 0xfc, 0xe3> (checksum 0xe3)
[00807956ms,aux] ALT MC -> Nexstar+ HC, cmd MTR_IS_APPROACH_DIR_NEG: AUX packet <0x3b, 0x04, 0x11, 0x0d, 0xfc, (0x00), 0xe2> (checksum 0xe2)
[00808092ms,aux] Nexstar+ HC -> Focuser, cmd FOCUS_GET_FW_VER: AUX packet <0x3b, 0x03, 0x0d, 0x12, 0xfe, 0xe0> (checksum 0xe0)
[00808220ms,aux] Nexstar+ HC -> Focuser, cmd FOCUS_GET_FW_VER: AUX packet <0x3b, 0x03, 0x0d, 0x12, 0xfe, 0xe0> (checksum 0xe0)
[00808348ms,aux] Nexstar+ HC -> Focuser, cmd FOCUS_GET_FW_VER: AUX packet <0x3b, 0x03, 0x0d, 0x12, 0xfe, 0xe0> (checksum 0xe0)
[00808477ms,aux] Nexstar+ HC -> Focuser, cmd FOCUS_GET_FW_VER: AUX packet <0x3b, 0x03, 0x0d, 0x12, 0xfe, 0xe0> (checksum 0xe0)
[00808605ms,aux] Nexstar+ HC -> Focuser, cmd FOCUS_GET_FW_VER: AUX packet <0x3b, 0x03, 0x0d, 0x12, 0xfe, 0xe0> (checksum 0xe0)
[00808733ms,aux] Nexstar+ HC -> Dew Heater Controller, cmd DEWHEATER_GET_FW_VER: AUX packet <0x3b, 0x03, 0x0d, 0xbb, 0xfe, 0x37> (checksum 0x37)
[00809007ms,aux] Nexstar+ HC -> Dew Heater Controller, cmd DEWHEATER_GET_FW_VER: AUX packet <0x3b, 0x03, 0x0d, 0xbb, 0xfe, 0x37> (checksum 0x37)
[00809360ms,aux] Nexstar+ HC -> Dew Heater Controller, cmd DEWHEATER_GET_FW_VER: AUX packet <0x3b, 0x03, 0x0d, 0xbb, 0xfe, 0x37> (checksum 0x37)
[00809713ms,aux] Nexstar+ HC -> Dew Heater Controller, cmd DEWHEATER_GET_FW_VER: AUX packet <0x3b, 0x03, 0x0d, 0xbb, 0xfe, 0x37> (checksum 0x37)
[00810066ms,aux] Nexstar+ HC -> Dew Heater Controller, cmd DEWHEATER_GET_FW_VER: AUX packet <0x3b, 0x03, 0x0d, 0xbb, 0xfe, 0x37> (checksum 0x37)
[00810449ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_SET_POSITION: AUX packet <0x3b, 0x06, 0x0d, 0x10, 0x04, (0x40 0x00 0x00), 0x99> (checksum 0x99)
[00810587ms,aux] AZM MC -> Nexstar+ HC, cmd MTR_SET_POSITION: AUX packet <0x3b, 0x03, 0x10, 0x0d, 0x04, 0xdc> (checksum 0xdc)
[00810745ms,aux] Nexstar+ HC -> ALT MC, cmd MTR_SET_POSITION: AUX packet <0x3b, 0x06, 0x0d, 0x11, 0x04, (0x40 0x00 0x00), 0x98> (checksum 0x98)
[00810882ms,aux] ALT MC -> Nexstar+ HC, cmd MTR_SET_POSITION: AUX packet <0x3b, 0x03, 0x11, 0x0d, 0x04, 0xdb> (checksum 0xdb)
[00811040ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_SET_CORDWRAP_POSITION: AUX packet <0x3b, 0x06, 0x0d, 0x10, 0x3a, (0xc0 0x00 0x00), 0xe3> (checksum 0xe3)
[00811182ms,aux] AZM MC -> Nexstar+ HC, cmd MTR_SET_CORDWRAP_POSITION: AUX packet <0x3b, 0x03, 0x10, 0x0d, 0x3a, 0xa6> (checksum 0xa6)
[00811314ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_CORDWRAP_ON: AUX packet <0x3b, 0x03, 0x0d, 0x10, 0x38, 0xa8> (checksum 0xa8)
[00811441ms,aux] AZM MC -> Nexstar+ HC, cmd MTR_CORDWRAP_ON: AUX packet <0x3b, 0x03, 0x10, 0x0d, 0x38, 0xa8> (checksum 0xa8)
[00811598ms,aux] Nexstar+ HC -> ALT MC, cmd MTR_SET_CORDWRAP_POSITION: AUX packet <0x3b, 0x06, 0x0d, 0x11, 0x3a, (0xc0 0x00 0x00), 0xe2> (checksum 0xe2)
[00811740ms,aux] ALT MC -> Nexstar+ HC, cmd MTR_SET_CORDWRAP_POSITION: AUX packet <0x3b, 0x03, 0x11, 0x0d, 0x3a, 0xa5> (checksum 0xa5)
[00811873ms,aux] Nexstar+ HC -> ALT MC, cmd MTR_CORDWRAP_ON: AUX packet <0x3b, 0x03, 0x0d, 0x11, 0x38, 0xa7> (checksum 0xa7)
[00812000ms,aux] ALT MC -> Nexstar+ HC, cmd MTR_CORDWRAP_ON: AUX packet <0x3b, 0x03, 0x11, 0x0d, 0x38, 0xa7> (checksum 0xa7)
```
After these packets are sent from HC to MC, I confirmed on the HC: 
* Set to INDEX
* Date/Time
* Quick Align

After hitting ENTER (Quick Align) the following AUX packets are sent:

```
[00898865ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_SET_POSITION: AUX packet <0x3b, 0x06, 0x0d, 0x10, 0x04, (0x40 0x00 0x00), 0x99> (checksum 0x99)
[00899002ms,aux] AZM MC -> Nexstar+ HC, cmd MTR_SET_POSITION: AUX packet <0x3b, 0x03, 0x10, 0x0d, 0x04, 0xdc> (checksum 0xdc)
[00899160ms,aux] Nexstar+ HC -> ALT MC, cmd MTR_SET_POSITION: AUX packet <0x3b, 0x06, 0x0d, 0x11, 0x04, (0x40 0x00 0x00), 0x98> (checksum 0x98)
[00899297ms,aux] ALT MC -> Nexstar+ HC, cmd MTR_SET_POSITION: AUX packet <0x3b, 0x03, 0x11, 0x0d, 0x04, 0xdb> (checksum 0xdb)
[00899455ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_SET_CORDWRAP_POSITION: AUX packet <0x3b, 0x06, 0x0d, 0x10, 0x3a, (0xc0 0x00 0x00), 0xe3> (checksum 0xe3)
[00899597ms,aux] AZM MC -> Nexstar+ HC, cmd MTR_SET_CORDWRAP_POSITION: AUX packet <0x3b, 0x03, 0x10, 0x0d, 0x3a, 0xa6> (checksum 0xa6)
[00899729ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_CORDWRAP_ON: AUX packet <0x3b, 0x03, 0x0d, 0x10, 0x38, 0xa8> (checksum 0xa8)
[00899856ms,aux] AZM MC -> Nexstar+ HC, cmd MTR_CORDWRAP_ON: AUX packet <0x3b, 0x03, 0x10, 0x0d, 0x38, 0xa8> (checksum 0xa8)
[00900014ms,aux] Nexstar+ HC -> ALT MC, cmd MTR_SET_CORDWRAP_POSITION: AUX packet <0x3b, 0x06, 0x0d, 0x11, 0x3a, (0xc0 0x00 0x00), 0xe2> (checksum 0xe2)
[00900156ms,aux] ALT MC -> Nexstar+ HC, cmd MTR_SET_CORDWRAP_POSITION: AUX packet <0x3b, 0x03, 0x11, 0x0d, 0x3a, 0xa5> (checksum 0xa5)
[00900288ms,aux] Nexstar+ HC -> ALT MC, cmd MTR_CORDWRAP_ON: AUX packet <0x3b, 0x03, 0x0d, 0x11, 0x38, 0xa7> (checksum 0xa7)
[00900415ms,aux] ALT MC -> Nexstar+ HC, cmd MTR_CORDWRAP_ON: AUX packet <0x3b, 0x03, 0x11, 0x0d, 0x38, 0xa7> (checksum 0xa7)
[00900572ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_PTRACK: AUX packet <0x3b, 0x06, 0x0d, 0x10, 0x06, (0x00 0x00 0x00), 0xd7> (checksum 0xd7)
[00900706ms,aux] AZM MC -> Nexstar+ HC, cmd MTR_PTRACK: AUX packet <0x3b, 0x03, 0x10, 0x0d, 0x06, 0xda> (checksum 0xda)
[00900861ms,aux] Nexstar+ HC -> ALT MC, cmd MTR_PTRACK: AUX packet <0x3b, 0x06, 0x0d, 0x11, 0x06, (0x00 0x00 0x00), 0xd6> (checksum 0xd6)
[00900995ms,aux] ALT MC -> Nexstar+ HC, cmd MTR_PTRACK: AUX packet <0x3b, 0x03, 0x11, 0x0d, 0x06, 0xd9> (checksum 0xd9)
[00901140ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_PTRACK: AUX packet <0x3b, 0x05, 0x0d, 0x10, 0x06, (0xff 0xff), 0xda> (checksum 0xda)
[00901271ms,aux] AZM MC -> Nexstar+ HC, cmd MTR_PTRACK: AUX packet <0x3b, 0x03, 0x10, 0x0d, 0x06, 0xda> (checksum 0xda)
```
I used before the Celestron GPS driver and always did a quick align, and then sync'ed on 2-3 stars via astrometry and everything was fine. I guess the same should apply for the AUX driver as well. I tested the Celestron AUX driver just indoor.

The important packets after Quick-Align seems to be:

```
[00898865ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_SET_POSITION: AUX packet <0x3b, 0x06, 0x0d, 0x10, 0x04, (0x40 0x00 0x00), 0x99> (checksum 0x99)
[00899002ms,aux] AZM MC -> Nexstar+ HC, cmd MTR_SET_POSITION: AUX packet <0x3b, 0x03, 0x10, 0x0d, 0x04, 0xdc> (checksum 0xdc)
[00899160ms,aux] Nexstar+ HC -> ALT MC, cmd MTR_SET_POSITION: AUX packet <0x3b, 0x06, 0x0d, 0x11, 0x04, (0x40 0x00 0x00), 0x98> (checksum 0x98)
[00899297ms,aux] ALT MC -> Nexstar+ HC, cmd MTR_SET_POSITION: AUX packet <0x3b, 0x03, 0x11, 0x0d, 0x04, 0xdb> (checksum 0xdb)
[00899455ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_SET_CORDWRAP_POSITION: AUX packet <0x3b, 0x06, 0x0d, 0x10, 0x3a, (0xc0 0x00 0x00), 0xe3> (checksum 0xe3)
[00899597ms,aux] AZM MC -> Nexstar+ HC, cmd MTR_SET_CORDWRAP_POSITION: AUX packet <0x3b, 0x03, 0x10, 0x0d, 0x3a, 0xa6> (checksum 0xa6)
[00899729ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_CORDWRAP_ON: AUX packet <0x3b, 0x03, 0x0d, 0x10, 0x38, 0xa8> (checksum 0xa8)
[00899856ms,aux] AZM MC -> Nexstar+ HC, cmd MTR_CORDWRAP_ON: AUX packet <0x3b, 0x03, 0x10, 0x0d, 0x38, 0xa8> (checksum 0xa8)
[... cut ...]
[00900156ms,aux] ALT MC -> Nexstar+ HC, cmd MTR_SET_CORDWRAP_POSITION: AUX packet <0x3b, 0x03, 0x11, 0x0d, 0x3a, 0xa5> (checksum 0xa5)
[00900288ms,aux] Nexstar+ HC -> ALT MC, cmd MTR_CORDWRAP_ON: AUX packet <0x3b, 0x03, 0x0d, 0x11, 0x38, 0xa7> (checksum 0xa7)
[00900415ms,aux] ALT MC -> Nexstar+ HC, cmd MTR_CORDWRAP_ON: AUX packet <0x3b, 0x03, 0x11, 0x0d, 0x38, 0xa7> (checksum 0xa7)
```

_Originally posted by @tstibor in https://github.com/indilib/indi-3rdparty/issues/742#issuecomment-1445995393_

knro, 2023-02-28:
Following up from the comment by Thomas in #742. Seems like sniffing the traffic would allow us to replicate the HC procedures so that the mount may be used without HC, only a direct USB connection should suffice.

tstibor, 2023-02-28:
The HC sends the following 3 crucial AUX commands (_MTR_SET_POSITION,  MTR_SET_CORDWRAP_POSITION, MTR_CORDWRAP_ON_) at startup:

### Example 1)
```
[00898865ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_SET_POSITION: AUX packet <0x3b, 0x06, 0x0d, 0x10, 0x04, (0x40 0x00 0x00), 0x99> (checksum 0x99)
[00899002ms,aux] AZM MC -> Nexstar+ HC, cmd MTR_SET_POSITION: AUX packet <0x3b, 0x03, 0x10, 0x0d, 0x04, 0xdc> (checksum 0xdc)
[00899160ms,aux] Nexstar+ HC -> ALT MC, cmd MTR_SET_POSITION: AUX packet <0x3b, 0x06, 0x0d, 0x11, 0x04, (0x40 0x00 0x00), 0x98> (checksum 0x98)
[00899297ms,aux] ALT MC -> Nexstar+ HC, cmd MTR_SET_POSITION: AUX packet <0x3b, 0x03, 0x11, 0x0d, 0x04, 0xdb> (checksum 0xdb)

[00899455ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_SET_CORDWRAP_POSITION: AUX packet <0x3b, 0x06, 0x0d, 0x10, 0x3a, (0xc0 0x00 0x00), 0xe3> (checksum 0xe3)
[00899597ms,aux] AZM MC -> Nexstar+ HC, cmd MTR_SET_CORDWRAP_POSITION: AUX packet <0x3b, 0x03, 0x10, 0x0d, 0x3a, 0xa6> (checksum 0xa6)
[00899729ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_CORDWRAP_ON: AUX packet <0x3b, 0x03, 0x0d, 0x10, 0x38, 0xa8> (checksum 0xa8)
[00899856ms,aux] AZM MC -> Nexstar+ HC, cmd MTR_CORDWRAP_ON: AUX packet <0x3b, 0x03, 0x10, 0x0d, 0x38, 0xa8> (checksum 0xa8)

[00900014ms,aux] Nexstar+ HC -> ALT MC, cmd MTR_SET_CORDWRAP_POSITION: AUX packet <0x3b, 0x06, 0x0d, 0x11, 0x3a, (0xc0 0x00 0x00), 0xe2> (checksum 0xe2)
[00900156ms,aux] ALT MC -> Nexstar+ HC, cmd MTR_SET_CORDWRAP_POSITION: AUX packet <0x3b, 0x03, 0x11, 0x0d, 0x3a, 0xa5> (checksum 0xa5)
[00900288ms,aux] Nexstar+ HC -> ALT MC, cmd MTR_CORDWRAP_ON: AUX packet <0x3b, 0x03, 0x0d, 0x11, 0x38, 0xa7> (checksum 0xa7)
[00900415ms,aux] ALT MC -> Nexstar+ HC, cmd MTR_CORDWRAP_ON: AUX packet <0x3b, 0x03, 0x11, 0x0d, 0x38, 0xa7> (checksum 0xa7)
```

### Example 2)
```
[00426302ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_SET_POSITION: AUX packet <0x3b, 0x06, 0x0d, 0x10, 0x04, (0x40 0x00 0x00), 0x99> (checksum 0x99)
[00426440ms,aux] AZM MC -> Nexstar+ HC, cmd MTR_SET_POSITION: AUX packet <0x3b, 0x03, 0x10, 0x0d, 0x04, 0xdc> (checksum 0xdc)
[00426597ms,aux] Nexstar+ HC -> ALT MC, cmd MTR_SET_POSITION: AUX packet <0x3b, 0x06, 0x0d, 0x11, 0x04, (0x40 0x00 0x00), 0x98> (checksum 0x98)
[00426735ms,aux] ALT MC -> Nexstar+ HC, cmd MTR_SET_POSITION: AUX packet <0x3b, 0x03, 0x11, 0x0d, 0x04, 0xdb> (checksum 0xdb)

[00426892ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_SET_CORDWRAP_POSITION: AUX packet <0x3b, 0x06, 0x0d, 0x10, 0x3a, (0xc0 0x00 0x00), 0xe3> (checksum 0xe3)
[00427034ms,aux] AZM MC -> Nexstar+ HC, cmd MTR_SET_CORDWRAP_POSITION: AUX packet <0x3b, 0x03, 0x10, 0x0d, 0x3a, 0xa6> (checksum 0xa6)
[00427167ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_CORDWRAP_ON: AUX packet <0x3b, 0x03, 0x0d, 0x10, 0x38, 0xa8> (checksum 0xa8)
[00427294ms,aux] AZM MC -> Nexstar+ HC, cmd MTR_CORDWRAP_ON: AUX packet <0x3b, 0x03, 0x10, 0x0d, 0x38, 0xa8> (checksum 0xa8)

[00427451ms,aux] Nexstar+ HC -> ALT MC, cmd MTR_SET_CORDWRAP_POSITION: AUX packet <0x3b, 0x06, 0x0d, 0x11, 0x3a, (0xc0 0x00 0x00), 0xe2> (checksum 0xe2)
[00427593ms,aux] ALT MC -> Nexstar+ HC, cmd MTR_SET_CORDWRAP_POSITION: AUX packet <0x3b, 0x03, 0x11, 0x0d, 0x3a, 0xa5> (checksum 0xa5)
[00427726ms,aux] Nexstar+ HC -> ALT MC, cmd MTR_CORDWRAP_ON: AUX packet <0x3b, 0x03, 0x0d, 0x11, 0x38, 0xa7> (checksum 0xa7)
[00427853ms,aux] ALT MC -> Nexstar+ HC, cmd MTR_CORDWRAP_ON: AUX packet <0x3b, 0x03, 0x11, 0x0d, 0x38, 0xa7> (checksum 0xa7)
```

I guess at an initial startup when the mount is set at INDEX position, we could send some proper initial values in 
the AUX commands. When the mount is parked we need the information at the next INDI driver startup and set these parked positions. I will do more testing in the next days.

knro, 2023-02-28:
So for equatorial mounts, after these 3 commands, are both encoders set to 90/90 ?

tstibor, 2023-02-28:
Yes, this is how I would also interpret it. The payload in cmd `MTR_SET_POSITION` is

`(0x40 0x00 0x00)`

which interpreted as: `(uint32_t)(rad * 0x001000000 / (2 * M_PI));`

where rad = M_PI / 2, so 90°

knro, 2023-02-28:
(0x40 0x00 0x00) --> 4194304 steps. That's 1/4 of a revolution or 90 degrees, so it checks out.  I'll try to incorporate this in the startup procedure for equatorial/altaz mounts not connected via hand control.

tstibor, 2023-03-01:
Here is the usbauxrelay repo https://github.com/tstibor/usbauxrelay

tstibor, 2023-03-05:
Clicking on _Factory Reset_ on the HC sends the following AUX packets:

```
[00880742ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_EEPROM_WRITE: AUX packet <0x3b, 0x05, 0x0d, 0x10, 0x31, (0x00 0x00), 0xad> (checksum 0xad)
[00880877ms,aux] AZM MC -> Nexstar+ HC, cmd MTR_EEPROM_WRITE: AUX packet <0x3b, 0x03, 0x10, 0x0d, 0x31, 0xaf> (checksum 0xaf)
[00881024ms,aux] Nexstar+ HC -> ALT MC, cmd MTR_EEPROM_WRITE: AUX packet <0x3b, 0x05, 0x0d, 0x11, 0x31, (0x00 0x00), 0xac> (checksum 0xac)
[00881159ms,aux] ALT MC -> Nexstar+ HC, cmd MTR_EEPROM_WRITE: AUX packet <0x3b, 0x03, 0x11, 0x0d, 0x31, 0xae> (checksum 0xae)
```

After switching the mount off and on, the usual startup procedure starts:
```
[00975530ms,aux] Nexstar+ HC -> ALT MC, cmd MTR_GET_MODEL: AUX packet <0x3b, 0x03, 0x0d, 0x11, 0x05, 0xda> (checksum 0xda)
[00975676ms,aux] ALT MC -> Nexstar+ HC, cmd MTR_GET_MODEL: AUX packet <0x3b, 0x05, 0x11, 0x0d, 0x05, (0x14 0x85), 0x3f> (checksum 0x3f)
[00975809ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_GET_MODEL: AUX packet <0x3b, 0x03, 0x0d, 0x10, 0x05, 0xdb> (checksum 0xdb)
[00975955ms,aux] AZM MC -> Nexstar+ HC, cmd MTR_GET_MODEL: AUX packet <0x3b, 0x05, 0x10, 0x0d, 0x05, (0x14 0x85), 0x40> (checksum 0x40)
[00976088ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_GET_VERSION: AUX packet <0x3b, 0x03, 0x0d, 0x10, 0xfe, 0xe2> (checksum 0xe2)
[00976256ms,aux] AZM MC -> Nexstar+ HC, cmd NXS_GET_FW_VER: AUX packet <0x3b, 0x07, 0x10, 0x0d, 0xfe, (0x07 0x0f 0x20 0x4e), 0x5a> (checksum 0x5a)
[00976394ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_IS_APPROACH_DIR_NEG: AUX packet <0x3b, 0x03, 0x0d, 0x10, 0xfc, 0xe4> (checksum 0xe4)
[00976536ms,aux] AZM MC -> Nexstar+ HC, cmd MTR_IS_APPROACH_DIR_NEG: AUX packet <0x3b, 0x04, 0x10, 0x0d, 0xfc, (0x00), 0xe3> (checksum 0xe3)
[00976671ms,aux] Nexstar+ HC -> ALT MC, cmd MTR_IS_APPROACH_DIR_NEG: AUX packet <0x3b, 0x03, 0x0d, 0x11, 0xfc, 0xe3> (checksum 0xe3)
[00976813ms,aux] ALT MC -> Nexstar+ HC, cmd MTR_IS_APPROACH_DIR_NEG: AUX packet <0x3b, 0x04, 0x11, 0x0d, 0xfc, (0x00), 0xe2> (checksum 0xe2)
[00993295ms,aux] Nexstar+ HC -> Focuser, cmd FOCUS_GET_FW_VER: AUX packet <0x3b, 0x03, 0x0d, 0x12, 0xfe, 0xe0> (checksum 0xe0)
[00993647ms,aux] Nexstar+ HC -> Focuser, cmd FOCUS_GET_FW_VER: AUX packet <0x3b, 0x03, 0x0d, 0x12, 0xfe, 0xe0> (checksum 0xe0)
[00994000ms,aux] Nexstar+ HC -> Focuser, cmd FOCUS_GET_FW_VER: AUX packet <0x3b, 0x03, 0x0d, 0x12, 0xfe, 0xe0> (checksum 0xe0)
[00994352ms,aux] Nexstar+ HC -> Focuser, cmd FOCUS_GET_FW_VER: AUX packet <0x3b, 0x03, 0x0d, 0x12, 0xfe, 0xe0> (checksum 0xe0)
[00994705ms,aux] Nexstar+ HC -> Focuser, cmd FOCUS_GET_FW_VER: AUX packet <0x3b, 0x03, 0x0d, 0x12, 0xfe, 0xe0> (checksum 0xe0)
[00995057ms,aux] Nexstar+ HC -> Dew Heater Controller, cmd DEWHEATER_GET_FW_VER: AUX packet <0x3b, 0x03, 0x0d, 0xbb, 0xfe, 0x37> (checksum 0x37)
[00995410ms,aux] Nexstar+ HC -> Dew Heater Controller, cmd DEWHEATER_GET_FW_VER: AUX packet <0x3b, 0x03, 0x0d, 0xbb, 0xfe, 0x37> (checksum 0x37)
[00995762ms,aux] Nexstar+ HC -> Dew Heater Controller, cmd DEWHEATER_GET_FW_VER: AUX packet <0x3b, 0x03, 0x0d, 0xbb, 0xfe, 0x37> (checksum 0x37)
[00996115ms,aux] Nexstar+ HC -> Dew Heater Controller, cmd DEWHEATER_GET_FW_VER: AUX packet <0x3b, 0x03, 0x0d, 0xbb, 0xfe, 0x37> (checksum 0x37)
[00996467ms,aux] Nexstar+ HC -> Dew Heater Controller, cmd DEWHEATER_GET_FW_VER: AUX packet <0x3b, 0x03, 0x0d, 0xbb, 0xfe, 0x37> (checksum 0x37)
[00996850ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_SET_POSITION: AUX packet <0x3b, 0x06, 0x0d, 0x10, 0x04, (0x40 0x00 0x00), 0x99> (checksum 0x99)
[00996988ms,aux] AZM MC -> Nexstar+ HC, cmd MTR_SET_POSITION: AUX packet <0x3b, 0x03, 0x10, 0x0d, 0x04, 0xdc> (checksum 0xdc)
[00997145ms,aux] Nexstar+ HC -> ALT MC, cmd MTR_SET_POSITION: AUX packet <0x3b, 0x06, 0x0d, 0x11, 0x04, (0x40 0x00 0x00), 0x98> (checksum 0x98)
[00997283ms,aux] ALT MC -> Nexstar+ HC, cmd MTR_SET_POSITION: AUX packet <0x3b, 0x03, 0x11, 0x0d, 0x04, 0xdb> (checksum 0xdb)
[00997441ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_SET_CORDWRAP_POSITION: AUX packet <0x3b, 0x06, 0x0d, 0x10, 0x3a, (0xc0 0x00 0x00), 0xe3> (checksum 0xe3)
[00997583ms,aux] AZM MC -> Nexstar+ HC, cmd MTR_SET_CORDWRAP_POSITION: AUX packet <0x3b, 0x03, 0x10, 0x0d, 0x3a, 0xa6> (checksum 0xa6)
[00997715ms,aux] Nexstar+ HC -> AZM MC, cmd MTR_CORDWRAP_ON: AUX packet <0x3b, 0x03, 0x0d, 0x10, 0x38, 0xa8> (checksum 0xa8)
[00997842ms,aux] AZM MC -> Nexstar+ HC, cmd MTR_CORDWRAP_ON: AUX packet <0x3b, 0x03, 0x10, 0x0d, 0x38, 0xa8> (checksum 0xa8)
[00997999ms,aux] Nexstar+ HC -> ALT MC, cmd MTR_SET_CORDWRAP_POSITION: AUX packet <0x3b, 0x06, 0x0d, 0x11, 0x3a, (0xc0 0x00 0x00), 0xe2> (checksum 0xe2)
[00998141ms,aux] ALT MC -> Nexstar+ HC, cmd MTR_SET_CORDWRAP_POSITION: AUX packet <0x3b, 0x03, 0x11, 0x0d, 0x3a, 0xa5> (checksum 0xa5)
[00998274ms,aux] Nexstar+ HC -> ALT MC, cmd MTR_CORDWRAP_ON: AUX packet <0x3b, 0x03, 0x0d, 0x11, 0x38, 0xa7> (checksum 0xa7)
[00998401ms,aux] ALT MC -> Nexstar+ HC, cmd MTR_CORDWRAP_ON: AUX packet <0x3b, 0x03, 0x11, 0x0d, 0x38, 0xa7> (checksum 0xa7)
```

knro, 2023-03-06:
Excellent, so it's pretty much comes down to three commands:
+ MTR_SET_POSITION
+ MTR_SET_CORDWRAP_POSITION
+ MTR_CORDWRAP_ON

While testing CGX-L, looks like it needs to home to switch indexes on startup as well. Only after this is done that you can continue to setting location + quick align in the handset. For your model, I suspect the first step is not required?

tstibor, 2023-03-10:
For the AVX the "switch indexes" is not required. When powering up the HC and going through the alignment steps one
needs to confirm the "Set to index" step with hitting ENTER (see Screenshot).

[image]

This step I guess let the HC know that both axes are pointing 90°/90°. Usually I roughly align to the index marks printed on the mount and continue.

knro, 2023-03-10:
Thank you @tstibor so only 3 commands are good to go now to complete mount initialization without HC?

knro, 2023-03-10:
Lovely! It would be great if you can verify it works on your mount. Then I can use this as basis and see if it is enough for CGX-L or an additional step is required. This should probably cover all mounts. Now for Alt-Az mounts, I think both encoders need to be set to 0/0 respectively.

tstibor, 2023-03-10:
I switched tracking on, and the following changed, so all looks good:

```
[03049049ms,aux] ALT MC -> INDI Celestron AUX, cmd MTR_GET_POSITION: AUX packet <0x3b, 0x06, 0x11, 0x20, 0x01, (0x40 0x00 0x00), 0x88> (checksum 0x88)
[03050153ms,usb] INDI Celestron AUX -> AZM MC, cmd MTR_GET_POSITION: AUX packet <0x3b, 0x03, 0x20, 0x10, 0x01, 0xcc> (checksum 0xcc)
[03050315ms,aux] AZM MC -> INDI Celestron AUX, cmd MTR_GET_POSITION: AUX packet <0x3b, 0x06, 0x10, 0x20, 0x01, (0x40 0x3c 0x1a), 0x33> (checksum 0x33)
[03050417ms,usb] INDI Celestron AUX -> ALT MC, cmd MTR_GET_POSITION: AUX packet <0x3b, 0x03, 0x20, 0x11, 0x01, 0xcb> (checksum 0xcb)
[03050579ms,aux] ALT MC -> INDI Celestron AUX, cmd MTR_GET_POSITION: AUX packet <0x3b, 0x06, 0x11, 0x20, 0x01, (0x40 0x00 0x00), 0x88> (checksum 0x88)
[03051684ms,usb] INDI Celestron AUX -> AZM MC, cmd MTR_GET_POSITION: AUX packet <0x3b, 0x03, 0x20, 0x10, 0x01, 0xcc> (checksum 0xcc)
[03051846ms,aux] AZM MC -> INDI Celestron AUX, cmd MTR_GET_POSITION: AUX packet <0x3b, 0x06, 0x10, 0x20, 0x01, (0x40 0x3d 0x43), 0x09> (checksum 0x09)
[03051948ms,usb] INDI Celestron AUX -> ALT MC, cmd MTR_GET_POSITION: AUX packet <0x3b, 0x03, 0x20, 0x11, 0x01, 0xcb> (checksum 0xcb)
[03052110ms,aux] ALT MC -> INDI Celestron AUX, cmd MTR_GET_POSITION: AUX packet <0x3b, 0x06, 0x11, 0x20, 0x01, (0x40 0x00 0x00), 0x88> (checksum 0x88)
[03053214ms,usb] INDI Celestron AUX -> AZM MC, cmd MTR_GET_POSITION: AUX packet <0x3b, 0x03, 0x20, 0x10, 0x01, 0xcc> (checksum 0xcc)
[03053376ms,aux] AZM MC -> INDI Celestron AUX, cmd MTR_GET_POSITION: AUX packet <0x3b, 0x06, 0x10, 0x20, 0x01, (0x40 0x3e 0x6e), 0xdd> (checksum 0xdd)
[03053478ms,usb] INDI Celestron AUX -> ALT MC, cmd MTR_GET_POSITION: AUX packet <0x3b, 0x03, 0x20, 0x11, 0x01, 0xcb> (checksum 0xcb)
[03053640ms,aux] ALT MC -> INDI Celestron AUX, cmd MTR_GET_POSITION: AUX packet <0x3b, 0x06, 0x11, 0x20, 0x01, (0x40 0x00 0x00), 0x88> (checksum 0x88)
[03054744ms,usb] INDI Celestron AUX -> AZM MC, cmd MTR_GET_POSITION: AUX packet <0x3b, 0x03, 0x20, 0x10, 0x01, 0xcc> (checksum 0xcc)
[03054906ms,aux] AZM MC -> INDI Celestron AUX, cmd MTR_GET_POSITION: AUX packet <0x3b, 0x06, 0x10, 0x20, 0x01, (0x40 0x3f 0x97), 0xb3> (checksum 0xb3)
[03055009ms,usb] INDI Celestron AUX -> ALT MC, cmd MTR_GET_POSITION: AUX packet <0x3b, 0x03, 0x20, 0x11, 0x01, 0xcb> (checksum 0xcb)
```

[image]

knro, 2023-03-10:
Can you please prepare a PR?

tstibor, 2023-03-10:
I will test the quick hack and report back.

Another interesting HEX command which is taken from: [initializeconn](https://github.com/platini2/celestronauxbus/blob/5c395e66e80b76d6d0a6274b06e9e9e1831a7c1c/celestron.py#L858) enables to set the HC in passthrough.

I have extracted the python code:

```
import serial
import time

ser = serial.Serial()
ser.port = '/dev/ttyUSB1'
ser.baudrate = 9600
ser.bytesize = serial.EIGHTBITS
ser.parity = serial.PARITY_NONE
ser.stopbits = serial.STOPBITS_ONE
ser.timeout = 0
ser.xonxoff = False
ser.open()
if ser.isOpen():
    ser.flushInput()
    ser.flushOutput()
data = b'\x56'
ser.write(data)
time.sleep(.5)
#Should return 0x05 0x1f 0x23
ser.read(ser.inWaiting())
data = b'\x8a'
ser.write(data)
time.sleep(.5)
#Should return 0xf8
ser.read(ser.inWaiting())
# print(data)
ser.close
ser.baudrate = 115200
ser.open
time.sleep(1)
data = b'\x0a\x00\x02\x08\x00\x4b\x00\x00\xd0\xc0'
ser.write(data)
time.sleep(.5)
#Should return 0x05 0x00 0x06 0x38 0xc0
data = ser.read(ser.inWaiting())
```

Executing the code sets the HC in passthrough (see screenshot):
[image]

Will prepare a PR.

knro, 2023-06-04:
IIRC, this was already merged so we can consider this issue closed.

closed 2023-06-04 by knro

ckuethe, 2023-06-04:
I dunno about the pass-through part, but the "quick-align-and-get-outta-my-face-so-i-can-astrophotograpy" part is working pretty well. I've remotely power cycled my whole rig several times since march and have not had to touch the hand controller yet.

knro, 2023-06-04:
Thank you for the confirmation Chris, great to know.