indilib/indi-3rdparty #754: Could not build indi-gphoto on RaspberryPI 4 (astroberry)

DevTonyShepard, 2023-03-11:
**Describe the bug**
I am trying to build from source indi and i came across some errors while building, i was following a guide made on the astroberry repo (https://github.com/rkaczorek/astroberry-server/issues/59#issuecomment-828265566)
I wrote the stacktrace below, maybe i am doing something wrong, first time building this from source. also first time asking for help creating an issue, let me know if i made some mistakes while writing this thanks.
I also build from source libraw version 20

**To Reproduce**
mkdir -p ~/Projects
cd ~/Projects
git clone --depth=1 https://github.com/indilib/indi-3rdparty
mkdir -p ~/Projects/build/indi-gphoto
cd ~/Projects/build/indi-gphoto
cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi-3rdparty/indi-gphoto
make -j4

**Expected behavior**
should build i guess

**Desktop (please complete the following information):**
 - OS: Raspbian Buster 32 bit (astroberry)

**Log Files**
In file included from /home/astroberry/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.cpp:23:
/home/astroberry/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.h:101:76: error: 'FITSRecord' is not a member of 'INDI'
  void addFITSKeywords(INDI::CCDChip * targetChip, std::vector<INDI::FITSRecord> &fitsKeywords) override;
                                                                     ^~~~~~~~~~

/home/astroberry/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.h:101:76: error: 'FITSRecord' is not a member of 'INDI'
/home/astroberry/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.h:101:86: error: template argument 1 is invalid
 ITSKeywords(INDI::CCDChip * targetChip, std::vector<INDI::FITSRecord> &fitsKeywords) override;
                                                                     ^

/home/astroberry/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.h:101:86: error: template argument 2 is invalid
/home/astroberry/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.h:232:15: error: 'PropertyBlob' in namespace 'INDI' does not name a type
         INDI::PropertyBlob imageBP {INDI::Property()};
               ^~~~~~~~~~~~
/home/astroberry/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.h:232:9: note: suggested alternative: 'PropertyBasic'
         INDI::PropertyBlob imageBP {INDI::Property()};
         ^~~~
         PropertyBasic
/home/astroberry/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.h:101:14: error: 'void GPhotoCCD::addFITSKeywords(INDI::CCDChip*, int&)' marked 'override', but does not override
         void addFITSKeywords(INDI::CCDChip * targetChip, std::vector<INDI::FITSRecord> &fitsKeywords) override;
              ^~~~~~~~~~~~~~~
/home/astroberry/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.cpp: In member function 'virtual bool GPhotoCCD::updateProperties()':
/home/astroberry/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.cpp:389:9: error: 'imageBP' was not declared in this scope
         imageBP = getBLOB("CCD1");
         ^~~~~~~
/home/astroberry/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.cpp: In member function 'virtual bool GPhotoCCD::UpdateCCDFrame(int, int, int, int)':
/home/astroberry/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.cpp:1031:35: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second:
     if (EncodeFormatSP[FORMAT_FITS].getState() != ISS_ON && EncodeFormatSP[FORMAT_XISF].getState() != ISS_ON)
                                   ^
In file included from /usr/include/libindi/indipropertynumber.h:21,
                 from /usr/include/libindi/indiccd.h:32,
                 from /home/astroberry/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.h:27,
                 from /home/astroberry/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.cpp:23:
/usr/include/libindi/indipropertybasic.h:113:24: note: candidate 1: 'INDI::WidgetView<T>& INDI::PropertyBasic<T>::operator[](size_t) const [with T = _ISwitch; size_t = unsigned int]'
         WidgetView<T> &operator[](size_t index) const;
                        ^~~~~~~~
/home/astroberry/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.cpp:1031:35: note: candidate 2: 'operator[](const INDI::Property*, int)' <built-in>
     if (EncodeFormatSP[FORMAT_FITS].getState() != ISS_ON && EncodeFormatSP[FORMAT_XISF].getState() != ISS_ON)
                                   ^
/home/astroberry/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.cpp:1031:35: warning: ISO C++ says that these are ambiguous, even though the worst conversion for the first is better than the worst conversion for the second:
In file included from /usr/include/libindi/indipropertynumber.h:21,
                 from /usr/include/libindi/indiccd.h:32,
                 from /home/astroberry/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.h:27,
                 from /home/astroberry/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.cpp:23:
/usr/include/libindi/indipropertybasic.h:113:24: note: candidate 1: 'INDI::WidgetView<T>& INDI::PropertyBasic<T>::operator[](size_t) const [with T = _ISwitch; size_t = unsigned int]'
         WidgetView<T> &operator[](size_t index) const;
                        ^~~~~~~~
/home/astroberry/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.cpp:1031:35: note: candidate 2: 'operator[](INDI::Property*, int)' <built-in>
     if (EncodeFormatSP[FORMAT_FITS].getState() != ISS_ON && EncodeFormatSP[FORMAT_XISF].getState() != ISS_ON)
[... 106 lines cut ...]
 ITSKeywords(INDI::CCDChip * targetChip, std::vector<INDI::FITSRecord> &fitsKeywords)
                                                                     ^

/home/astroberry/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.cpp:2028:89: error: template argument 2 is invalid
/home/astroberry/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.cpp: In member function 'void GPhotoCCD::addFITSKeywords(INDI::CCDChip*, int&)':
/home/astroberry/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.cpp:2030:56: error: no matching function for call to 'GPhotoCCD::addFITSKeywords(INDI::CCDChip*&, int&)'
     INDI::CCD::addFITSKeywords(targetChip, fitsKeywords);
                                                        ^
In file included from /home/astroberry/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.h:27,
                 from /home/astroberry/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.cpp:23:
/usr/include/libindi/indiccd.h:508:22: note: candidate: 'virtual void INDI::CCD::addFITSKeywords(INDI::CCDChip*)'
         virtual void addFITSKeywords(CCDChip * targetChip);
                      ^~~~~~~~~~~~~~~
/usr/include/libindi/indiccd.h:508:22: note:   candidate expects 1 argument, 2 provided
/home/astroberry/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.cpp:2039:30: error: request for member 'push_back' in 'fitsKeywords', which is of non-class type 'int'
                 fitsKeywords.push_back({"ISOSPEED", isoSpeed, "ISO Speed"});
                              ^~~~~~~~~
/home/astroberry/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.cpp:2045:22: error: request for member 'push_back' in 'fitsKeywords', which is of non-class type 'int'
         fitsKeywords.push_back({"CCD-TEMP", TemperatureN[0].value, 3, "CCD Temperature (Celsius)"});
                      ^~~~~~~~~
make[2]: *** [CMakeFiles/indi_gphoto_ccd.dir/build.make:63: CMakeFiles/indi_gphoto_ccd.dir/gphoto_ccd.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/indi_gphoto_ccd.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

Thank you guys for your time :)

IanCassTwo, 2023-03-11:
I built this yesterday on my Rpi 4. You need to "apt-get remove libindi-dev" and build the latest one from git https://github.com/indilib/indi. After that's completed, you can successfully compile indi-3rdparty drivers.

DevTonyShepard, 2023-03-11:
Hi Ian! thanks a lot for your help, i managed to successfully built this :) 

for anyone getting the same error as me:
i "sudo apt-get remove libindi-dev" and then rebuilt the indi repo being sure to have pulled the latest code. then i ran all of these:
# Core indi library
sudo apt -y install libnova-dev libcfitsio-dev libusb-1.0-0-dev zlib1g-dev libgsl-dev build-essential cmake git libjpeg-dev libcurl4-gnutls-dev libtiff-dev libfftw3-dev libftdi-dev libgps-dev libdc1394-22-dev libgphoto2-dev libboost-dev libboost-regex-dev librtlsdr-dev liblimesuite-dev libftdi1-dev libavcodec-dev libavdevice-dev
# Removed libraw-dev from the above as we're providing a newer one from source
mkdir -p ~/Projects
cd ~/Projects
git clone --depth 1 https://github.com/indilib/indi.git
mkdir -p ~/Projects/build/indi-core
cd ~/Projects/build/indi-core
cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi
make -j4
sudo make install

# indi-gphoto
mkdir -p ~/Projects
cd ~/Projects
git clone --depth=1 https://github.com/indilib/indi-3rdparty
mkdir -p ~/Projects/build/indi-gphoto
cd ~/Projects/build/indi-gphoto
cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi-3rdparty/indi-gphoto
make -j4
sudo make install
sudo cp indi_gphoto.xml /usr/share/indi

closed 2023-03-11 by DevTonyShepard