indilib/indi-3rdparty #756: Error in indi-aok debian changelog file date format

thx8411, 2023-03-11:
Error in indi-aok debian changelog file date format

seconds are missing

```
indi-aok (2.0) bionic; urgency=low (overhaul for firmware 21 of TCS)
  * cleaned up structure of source code
  * improved documentation and parser for JSON response
  * added lunar tracking mode
  * adapted system speed to new firmware
  * augmented global JSON buffer to 256 chars
  * augmented JSON return buffer of helperfunctions to 32 chars

-- T. Schriber <escribana@bluewin.ch> Thu, 9 Dec 2021 16:44 +0200 
```

"16:44" should be something like "16:44:00".

It prevents the package build in my case.

Best regards.

thx8411, 2023-03-11:
I think a blank space is also missing before "--" on the same line

knro, 2023-03-11:
Can you please submit a PR to fix it?

thx8411, 2023-03-12:
Hello Jasem,

Done, PR #757

Best regards

thx8411, 2023-03-12:
Closed by PR #757

closed 2023-03-12 by thx8411