indilib/indi-3rdparty #760: Atik apx60 - image size is unexpected

toriqo, 2023-03-13:
**Describe the bug**
When using the Atik apx60 camera I get a warning `[WARNING] Image size is unexpected. Expecting 122189760 bytes but received 0 bytes.` and no image.

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run driver
2. Try capturing a frame
3. See error
4. Abort exposure otherwise it just hangs

**Expected behavior**
I'd expect to the the frame i just shot.

**Desktop (please complete the following information):**
- OS: Kubuntu 22.04
- KDE Plasma Version: 5.25.5
- KDE Frameworks Version: 5.98.0
- Qt Version: 5.15.3
- Kernel Version: 5.15.0-52-generic (64-bit)

**Log Files**
```
[2023-02-11T18:40:30.604 EET DEBG ][   org.kde.kstars.ekos.capture] - Preparing capture job "" for execution.
[2023-02-11T18:40:30.606 EET INFO ][   org.kde.kstars.ekos.capture] - "Capturing 1,000-second L image..."
[2023-02-11T18:40:30.647 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Start Exposure : 1.000s "
[2023-02-11T18:40:30.648 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Start Exposure: 1 1.000s (0) "
[2023-02-11T18:40:30.648 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] SetExposureState 1 "
[2023-02-11T18:40:30.648 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Start Exposure 1 "
[2023-02-11T18:40:30.649 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Start Exposure 2 "
[2023-02-11T18:40:30.651 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] ET_ThreadMain 4 "
[2023-02-11T18:40:30.651 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] ET_ThreadMain 5 "
[2023-02-11T18:40:30.651 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] ET_InitExposure - 1 "
[2023-02-11T18:40:30.651 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] SetExposureState 2 "
[2023-02-11T18:40:30.651 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] ET_InitExposure - 2 "
[2023-02-11T18:40:30.651 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] FX3FPGAPowerUp::PowerUp - 1 "
[2023-02-11T18:40:30.775 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] FX3FPGAPowerUp::PowerUp - 2 "
[2023-02-11T18:40:30.844 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] FX3FPGAPowerUp::PowerUp - 3 "
[2023-02-11T18:40:30.844 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] FX3FPGAPowerUp::PowerUp - 5 "
[2023-02-11T18:40:30.844 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] ET_InitExposure - 3 "
[2023-02-11T18:40:30.844 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] ET_ThreadMain 6, 0, 1000 "
[2023-02-11T18:40:31.122 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Cooling: flags (31) level (0), minlvl (0), maxlvl (100), setpoint (0) "
[2023-02-11T18:40:31.886 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] ET_ThreadMain 7 "
[2023-02-11T18:40:31.886 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] SetExposureState 3 "
[2023-02-11T18:40:31.886 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] NO OB Correction! "
[2023-02-11T18:40:31.887 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Download Loop 0ms 122189760 "
[2023-02-11T18:40:31.986 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Download Loop 101ms 122189760 "
[2023-02-11T18:40:32.072 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Cooling: flags (31) level (0), minlvl (0), maxlvl (100), setpoint (0) "
[2023-02-11T18:40:32.086 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Download Loop 201ms 122189760 "
[2023-02-11T18:40:32.187 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Download Loop 302ms 122189760 "
[2023-02-11T18:40:32.288 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Download Loop 402ms 122189760 "
[2023-02-11T18:40:32.388 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Download Loop 503ms 122189760 "
[2023-02-11T18:40:32.489 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Download Loop 603ms 122189760 "
[2023-02-11T18:40:32.589 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Download Loop 704ms 122189760 "
[2023-02-11T18:40:32.637 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Download: NO BYTES! 7 "
[2023-02-11T18:40:32.637 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Download: Remaining: 122124224 (1.30 MB/s) (0.08 MB/s) "
[2023-02-11T18:40:32.637 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Download Loop 752ms 122124224 "
[2023-02-11T18:40:32.638 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Download: NO BYTES! 7 "
[2023-02-11T18:40:32.638 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Download: Remaining: 122058688 (62.50 MB/s) (62.50 MB/s) "
[2023-02-11T18:40:32.638 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Download Loop 753ms 122058688 "
[2023-02-11T18:40:32.640 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Download: NO BYTES! 7 "
[2023-02-11T18:40:32.640 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Download: Remaining: 121993152 (62.50 MB/s) (62.50 MB/s) "
[2023-02-11T18:40:32.641 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Download Loop 754ms 121993152 "
[2023-02-11T18:40:32.641 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Download: NO BYTES! 7 "
[2023-02-11T18:40:32.641 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Download: Remaining: 121927616 (62.50 MB/s) (62.50 MB/s) "
[2023-02-11T18:40:32.641 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Download Loop 755ms 121927616 "
[2023-02-11T18:40:32.641 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Download: NO BYTES! 7 "
[2023-02-11T18:40:32.642 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Download: Remaining: 121862080 (-1.00 MB/s) (-1.00 MB/s) "
[2023-02-11T18:40:32.642 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Download Loop 755ms 121862080 "
[2023-02-11T18:40:32.642 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Download: NO BYTES! 7 "
[2023-02-11T18:40:32.642 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Download: Remaining: 121796544 (62.50 MB/s) (62.50 MB/s) "
[... 20 lines cut ...]
[2023-02-11T18:40:33.624 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] StartOfData: 0 "
[2023-02-11T18:40:33.673 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Failed to download image "
[2023-02-11T18:40:33.673 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] ET_ThreadMain 8 "
[2023-02-11T18:40:33.673 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] SetExposureState 0 "
[2023-02-11T18:40:33.673 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] ET_ThreadMain 9 "
[2023-02-11T18:40:33.673 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] ThreadTime: 0 187 1237 1237 3025 3025 3025 "
[2023-02-11T18:40:33.673 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] ET_ThreadMain 1 "
[2023-02-11T18:40:33.673 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] ET_ThreadMain 2 "
[2023-02-11T18:40:33.674 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] FX3FPGAPowerUp::PowerDown - 1 "
[2023-02-11T18:40:33.674 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] FX3FPGAPowerUp::PowerDown - 2 "
[2023-02-11T18:40:33.675 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] FX3FPGAPowerUp::PowerDown - 3 "
[2023-02-11T18:40:33.698 EET INFO ][           org.kde.kstars.indi] - Atik Apx60 :  "[WARNING] Image size is unexpected. Expecting 122189760 bytes but received 0 bytes. "
[2023-02-11T18:40:33.698 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] ArtemisImageBuffer Done:  "
[2023-02-11T18:40:33.877 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] FX3FPGAPowerUp::PowerDown - 5 "
[2023-02-11T18:40:33.877 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] ET_ThreadMain 3 "
[... cut ...]
[2023-02-11T18:40:36.563 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Cancel Exposure: "
[2023-02-11T18:40:37.077 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Cooling: flags (31) level (0), minlvl (0), maxlvl (100), setpoint (0) "
[2023-02-11T18:40:38.077 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Cooling: flags (31) level (0), minlvl (0), maxlvl (100), setpoint (0) "
[2023-02-11T18:40:39.078 EET DEBG ][           org.kde.kstars.indi] - Atik Apx60 : "[DEBUG] Cooling: flags (31) level (0), minlvl (0), maxlvl (100), setpoint (0) "
```

TallFurryMan, 2023-03-14:
Could you please provide the version of the Atik driver? It is visible in the first lines of logs when the driver starts up (a line mentioning a dll version).

toriqo, 2023-03-14:
there's no mention of a driver version or dll in the log. i've just installed everything yesterday, indilib, gsc, kstars-bleeding.

this is all the log i get, verbose:

```
[2023-03-14T10:34:26.317 EET INFO ][                org.kde.kstars] - Welcome to KStars 3.6.3 Stable
[2023-03-14T10:34:26.317 EET INFO ][                org.kde.kstars] - Build: 2023-03-02T22:19:06Z
[2023-03-14T10:34:26.317 EET INFO ][                org.kde.kstars] - OS: "ubuntu"
[2023-03-14T10:34:26.317 EET INFO ][                org.kde.kstars] - API: "x86_64-little_endian-lp64"
[2023-03-14T10:34:26.317 EET INFO ][                org.kde.kstars] - Arch: "x86_64"
[2023-03-14T10:34:26.317 EET INFO ][                org.kde.kstars] - Kernel Type: "linux"
[2023-03-14T10:34:26.317 EET INFO ][                org.kde.kstars] - Kernel Version: "5.19.0-35-generic"
[2023-03-14T10:34:26.317 EET INFO ][                org.kde.kstars] - Qt Version: 5.15.3
[2023-03-14T10:34:26.648 EET DEBG ][                org.kde.kstars] - Opened the User DB. Ready.
[2023-03-14T10:34:26.800 EET INFO ][                org.kde.kstars] - Processing  "unnamedstars.dat" , HTMesh Level 3
[2023-03-14T10:34:26.800 EET INFO ][                org.kde.kstars] -   Sky Mesh Size:  512
[2023-03-14T10:34:26.845 EET INFO ][                org.kde.kstars] - Loaded DSO catalog file:  "unnamedstars.dat"
[2023-03-14T10:34:27.142 EET WARN ][                org.kde.kstars] - "Star HD20.794 not found."
[2023-03-14T10:34:27.155 EET WARN ][                org.kde.kstars] - "Star HD98.230 not found."
[2023-03-14T10:34:27.168 EET INFO ][                org.kde.kstars] - Loaded DSO catalogs.
[2023-03-14T10:34:27.286 EET INFO ][                org.kde.kstars] - Loading comets
[2023-03-14T10:34:27.381 EET DEBG ][                org.kde.kstars] - Paths to color scheme :  ("/usr/share/kstars/themes")
[2023-03-14T10:34:27.432 EET DEBG ][                org.kde.kstars] - Reporting new timestep value:  60
[2023-03-14T10:34:27.656 EET WARN ][                       default] - QObject::connect(QAction, Ekos::Scheduler): invalid nullptr parameter
[2023-03-14T10:34:27.684 EET WARN ][                       default] - QObject::connect: No such signal QCPAxis::QCPAxis::rangeChanged(QCPRange)
[2023-03-14T10:34:27.684 EET WARN ][                       default] - QObject::connect:  (receiver name: 'Analyze')
[2023-03-14T10:34:27.823 EET WARN ][                       default] - libpng warning: iCCP: known incorrect sRGB profile
[2023-03-14T10:34:27.848 EET CRIT ][                     kf.xmlgui] - Shortcut for action  "get_data" "Download New Data..." set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
[2023-03-14T10:34:27.849 EET CRIT ][                     kf.xmlgui] - Shortcut for action  "timestep_increase_units" "Increase Time Scale" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
[2023-03-14T10:34:27.849 EET CRIT ][                     kf.xmlgui] - Shortcut for action  "timestep_decrease_units" "Decrease Time Scale" set with QAction::setShortcut()! Use KActionCollection::setDefaultShortcut(s) instead.
[2023-03-14T10:34:27.862 EET DEBG ][                org.kde.kstars] - Daylight Saving Time inactive
[2023-03-14T10:34:27.862 EET DEBG ][                org.kde.kstars] - Next Daylight Savings Time change (Local Time):  "dum. mar. 26 01:00:00 2023 GMT"
[2023-03-14T10:34:27.862 EET DEBG ][                org.kde.kstars] - Next Daylight Savings Time change (UTC):  "sÃ¢m. mar. 25 23:00:00 2023 GMT"
[2023-03-14T10:34:28.033 EET DEBG ][                org.kde.kstars] - Daylight Saving Time inactive
[2023-03-14T10:34:28.033 EET DEBG ][                org.kde.kstars] - Next Daylight Savings Time change (Local Time):  "dum. mar. 26 01:00:00 2023 GMT"
[2023-03-14T10:34:28.033 EET DEBG ][                org.kde.kstars] - Next Daylight Savings Time change (UTC):  "sÃ¢m. mar. 25 23:00:00 2023 GMT"
[2023-03-14T10:34:28.033 EET DEBG ][                org.kde.kstars] - Starting the timer
[2023-03-14T10:34:28.161 EET DEBG ][                org.kde.kstars] - Date/Time is: "mar. mar. 14 08:34:27 2023 GMT"
[2023-03-14T10:34:28.161 EET DEBG ][                org.kde.kstars] - Location: "Mount Location"
[2023-03-14T10:34:28.161 EET DEBG ][                org.kde.kstars] - TZ0: 2 TZ: 2
[2023-03-14T10:34:28.163 EET WARN ][                org.kde.kstars] - Current icon theme is "breeze"
[2023-03-14T10:34:28.237 EET WARN ][                       default] - QImage::scaled: Image is a null image
[2023-03-14T10:34:29.099 EET INFO ][                org.kde.kstars] - Loading comets
[2023-03-14T10:34:33.111 EET INFO ][                org.kde.kstars] - Loading asteroids
[2023-03-14T10:35:42.163 EET DEBG ][           org.kde.kstars.ekos] - Resetting Ekos Manager...
[2023-03-14T10:35:44.839 EET INFO ][           org.kde.kstars.ekos] - "Starting INDI services..."
[2023-03-14T10:35:44.876 EET INFO ][           org.kde.kstars.ekos] - "INDI services started on port 7.624."
[2023-03-14T10:35:44.936 EET INFO ][           org.kde.kstars.ekos] - Ekos received a new device:  "Telescope Simulator"
[2023-03-14T10:35:44.977 EET INFO ][           org.kde.kstars.indi] - Telescope Simulator :  "[INFO] update mount and pier side: Pier Side On, mount type 2 "
[2023-03-14T10:35:45.247 EET INFO ][           org.kde.kstars.ekos] - Connecting to "Telescope Simulator"
[... cut ...]
[2023-03-14T10:35:46.777 EET INFO ][           org.kde.kstars.indi] - Atik Apx60 :  "[INFO] No previous configuration found. To save driver configuration, click Save Configuration in Options tab. "
[2023-03-14T10:35:47.078 EET INFO ][     org.kde.kstars.ekos.focus] - "Idle."
[2023-03-14T10:35:47.083 EET INFO ][           org.kde.kstars.ekos] - "Atik Apx60 is online."
[2023-03-14T10:35:47.084 EET INFO ][           org.kde.kstars.ekos] - "Guider port from Atik Apx60 is ready."
[2023-03-14T10:36:01.660 EET INFO ][           org.kde.kstars.indi] - Atik Apx60 :  "[WARNING] Image size is unexpected. Expecting 122189760 bytes but received 0 byte
```

TallFurryMan, 2023-03-14:
This is adequate, thanks:
```
[2023-03-14T10:35:46.677 EET INFO ][           org.kde.kstars.indi] - Atik Apx60 :  "[INFO] Detected camera Atik Cameras Atik Apx60 with firmware 0.0 "
[2023-03-14T10:35:46.678 EET INFO ][           org.kde.kstars.indi] - Atik Apx60 :  "[INFO] Horizon current gain: data[0:1] 0x0000 data[2:3] 0xA401 data[4:5] 0x4B00 values min 0 max 420 cur 75 "
[2023-03-14T10:35:46.680 EET INFO ][           org.kde.kstars.indi] - Atik Apx60 :  "[INFO] Horizon FX3 v2022.15355.21759 "
[2023-03-14T10:35:46.680 EET INFO ][           org.kde.kstars.indi] - Atik Apx60 :  "[INFO] Horizon FPGA v2022.15355.21759 "
[2023-03-14T10:35:46.680 EET INFO ][           org.kde.kstars.ekos] - "Atik Apx60" Version: "3.1" Interface: 2 is connected.
```

TallFurryMan, 2023-04-12:
I'm sorry for the lack of feedback, my daytime job has been quite heavy. There was a beta SDK published in April at Atik, I'll integrate it shortly (say, under two weeks from now probably).

toriqo, 2023-07-21:
any news on this?

TallFurryMan, 2023-07-30:
Holidays now, at last. Beginning on this week I'll integrate SDK 2023.07.14 from Atik, and we'll take it from there.

knro, 2023-07-31:
@TallFurryMan INDI 2.0.3 is due tomorrow, you think this can be merged today?

TallFurryMan, 2023-07-31:
Branch is uploading (slowly) now as I write, so if the build completes on the PR, this is possible. I might be able to test it tonight on the Horizon2, but that's quite uncertain.

toriqo, 2023-08-01:
you might wanna hold of a bit on that. i moved on to windows just to find the same problem.
after some back and forth emails with atik, me sending them logs, debug data and my personal findings, they realized it's a problem on their side and they're working on it. so... this issue should be on hold for a bit until they have an answer for me next week (a dev on their team has been assigned to work on this).

TallFurryMan, 2023-08-01:
OK thanks for the information. In any case we had to upgrade the SDK before anything could be sent to Atik, so thanks a lot for getting in touch with them by yourself.

closed 2023-09-07 by toriqo