indilib/indi-3rdparty #763: Fxload dependency

returntoreality, 2023-03-18:
**Describe the bug**
There are some drivers which need fxload (e.g. many (all?) QHY cameras). I found that there are two forks of the original fxload (which was part of linux-hotplug), which added support for fx3: fx3load on sourceforge and fxload as part of the libusb examples. A older copy of the libusb version was part of this repo before 1.9.3. The udev rules use the "-D" parameter to supply the path to the usb device. It seems that this is only available in the fx3load version, the libusb version requires using either "-d" or "-p". I could just use the fx3load fork, but this fork was not updated for 10 years, while the libusb version is still getting fixes (last commit 2 months ago). Ideally the udev rules should support both versions. Or did I miss another more maintained fork that supports fx3?

**To Reproduce**
Try using libusb fxload and plug in camera. Firmware is not loaded.

**Expected behavior**
Ideally the udev rules work with both forks of fxload.

**Desktop (please complete the following information):**
 - OS: Nixos
 - Version 2.0.0

MattBlack85, 2023-04-02:
Hi @returntoreality I think this is a common issue for anyone which is not using Debian/Ubuntu

I use arch and had the same problem so I ended up compiling and mantaining fxload from the libusb repo (I believe I am running 1.0.26 at the moment), I had some troubles but I ended up being able to compile fxload only, you can have a look at those steps if it helps you in any way https://github.com/devDucks/astroarch-pkgs/blob/main/packages/fxload/PKGBUILD#L20

Unfortunately this means that I have to ship a fix for the udev rules for arch and derivatives https://github.com/devDucks/astroarch-pkgs/blob/main/packages/indi-3rdparty-libs/PKGBUILD#L32

I'd be more than happy tho if this "issue" could be fixed in the mainstream

knro, 2023-05-25:
Where is this libusb-fxload maintained exactly?

rsarwar87, 2023-06-02:
if you download the qhy drivers from the OEM, it comes with the fxload binaries which works fine on astroberry and the latest bit raspbian as well as debian 11 (on rock 5b). maybe include those deliverables in here?

knro, 2023-06-04:
fxload is a binary system package that we cannot override in INDI. It must be fixed at the distro level.

closed 2023-06-04 by knro