indilib/indi-3rdparty #774: indi_playerone_ccd fails on bin2X2

glundby, 2023-04-12:
When in client ekos you select bin2x2, with my camera Player One Saturn-C SQR, the resulting picture only fills a band on top of frame.
When you first select bin3x3 and then bin2x2 picture is normal.
In IND Controls for camera the HV bin has only one possible value 1 even if specification for the camera says HW bin is supported. 

**To Reproduce**
Start Ekos - INDI
1. In INDI Controls watch only 1 possible value for HW bin
2. In Ekos - Capture (or other tab with capture function) set bin to 2x2, Take a picture (raw, Fits).
3. See the picture showing as a band in upper quarter of frame.
4. Repeat 2 and 3 after first have selected other bin value and tested. Then and see a normal frame also for bin 2x2.

**Expected behavior**
Bin 2x2 should definitely work. I'm unsure of HW bin is part of this or not. Anyhow HW bin2 should be available in INDI Control.

**Screenshots**
Sorry! cannot recreate at this time. Happened on two different Raspberry Pi4b. Must happen only first time it is tried until higher than bin2x2 is selected.

**Desktop (please complete the following information):**
 - OS: Stellarmate OS 
 - Newest version Kstars Build: Build: 2023-04-06T17:51:47Z

knro, 2023-04-12:
@hiro3110i Can you please check out this issue?

hiro3110i, 2023-04-21:
@glundby The issue which you pointed out was reproduced in my Raspberry Pi4.
It happens at the first shot by bin2 in Capture tab only just after start-up KStars/Ekos.
After that I couldn't find the banding issue for any bin settings.

In addition, I found the issue with Uranus-C but didn't find with Neptune-C II.
This result also seems to depend on the presence or absence of hardware binning support.
Uranus-C (and Saturn-C also) has a capability of hardware binning, but Neptune-C II doesn't have.

Looking only at the phenomenon, it seems to be a buffer size mismatch but further analysis is required for the cause.
I'm starting a discussion with the manufacturer about this issue.

glundby, 2023-04-22:
Thank You! Sendt fra E-post for Windows Fra: Hiroshi SAITOSendt: fredag 21. april 2023 kl. 17:13Til: indilib/indi-3rdpartyKopi: Gustav Olav Lundby; MentionEmne: Re: [indilib/indi-3rdparty] indi_playerone_ccd fails on bin2X2 (Issue ***@***.*** The issue which you pointed out was reproduced in my Raspberry Pi4.It happens at the first shot by bin2 in Capture tab only just after start-up KStars/Ekos.After that I couldn't find the banding issue for any bin settings.In addition, I found the issue with Uranus-C but didn't find with Neptune-C II.This result also seems to depend on the presence or absence of hardware binning support.Uranus-C (and Saturn-C also) has a capability of hardware binning, but Neptune-C II doesn't have.Looking only at the phenomenon, it seems to be a buffer size mismatch but further analysis is required for the cause.I'm starting a discussion with the manufacturer about this issue.

hiro3110i, 2023-04-27:
@glundby I apologize for your inconvenience.
I was able to fix the issue with great help from manufacturer.
When calling the SDK, I had to set the bin first and then set the image size.

@knro I have sent a pull request, so please release a fixed version if possible.

glundby, 2023-05-26:
Hi! No darkness in Norway at the moment!Will check when possible!GustavSendt fra E-post for Windows Fra: Jasem MutlaqSendt: torsdag 25. mai 2023 kl. 18:50Til: indilib/indi-3rdpartyKopi: Gustav Olav Lundby; MentionEmne: Re: [indilib/indi-3rdparty] indi_playerone_ccd fails on bin2X2 (Issue #774) PR merged, @glundby can you verify the fix?

knro, 2023-05-27:
You can test this indoors. Please try to use plain-text in your responses.

glundby, 2023-05-28:
Solved! Tested after upgrading Stellarmate – Kstars. Cannot recreate problem. Good! Looking in INDI – Controls – HardwareBin: There is not possible to select 2. I had hoped that setting Bin 2x2 would use HardwareBin since Player One Saturn-C SQR supports it. I gather INDI do not have this possibility and that only SoftwareBin is supported? Thanks! Gustav Sendt fra E-post for Windows Fra: Jasem MutlaqSendt: lørdag 27. mai 2023 kl. 11:12Til: indilib/indi-3rdpartyKopi: Gustav Olav Lundby; MentionEmne: Re: [indilib/indi-3rdparty] indi_playerone_ccd fails on bin2X2 (Issue #774) You can test this indoors. Please try to use plain-text in your responses.

hiro3110i, 2023-06-27:
@glundby The subject of this topic has been resolved.
But I have overlooked one thing. That's your point about hardware_bin.
The setting of hardware_bin is an ON/OFF switch, even in the camera manufacturer's SDK.
ON is equivalent to 1 and OFF to 0. We can set hardware_bin as well as for software_bin.
I hope this helps as a reference.

hiro3110i, 2023-09-15:
@knro Could you please close this topic?

closed 2023-09-15 by knro