indilib/indi-3rdparty #775: Driver indi_pentax: select: Bad file descriptor

JohnnySSH, 2023-04-16:
**Describe the bug**
As the title says: bad file descriptor; KStars/EKOS unable to connect to my Pentax K1 II camera

**To Reproduce**
Follow instructions on the README.md file to install

**Expected behavior**
Connection to camera and EKOS control

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: Arch Linux (up to date as of 04/16/2023)
 

**Log Files**

JohnnySSH, 2023-04-16:
Quick update... I updated Kstars to -git and indilib to -git version from the Arch Linux AUR.

Initially I had trouble with both my Nikon and Pentax cameras as EKOS logs showed that the drivers couldn't not start. It turns out that one problem seemed to stem from claiming the USB port in the machine. I solved this by unmounting the SD cards which it seems got auto mounted. <- this is an OS thing that I should probably disable in the future

In the log file for the Pentax Native Driver test I get the same thing:

`org.kde.kstars.indi] - INDI Server:  "2023-04-16T16:28:06: Driver indi_pentax: select: Bad file descriptor"`

Using the Legacy Driver, these appear in the logs:

`[2023-04-16T17:26:46.632 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-16T16:26:46: startup: /usr/bin/indiserver -v -p 7624 -m 1024 -r 0 -f /tmp/indififod2d8e7b9"
[2023-04-16T17:26:46.632 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-16T16:26:46: listening to port 7624 on fd 5"
[2023-04-16T17:26:46.632 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-16T16:26:46: Local server: listening on local domain at: @/tmp/indiserver"
[2023-04-16T17:26:46.633 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-16T16:26:46: FIFO: start indi_pentax_ccd"
[2023-04-16T17:26:46.633 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-16T16:26:46: FIFO: Starting driver indi_pentax_ccd"
[2023-04-16T17:26:46.633 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-16T16:26:46: Driver indi_pentax_ccd: pid=37533 rfd=9 wfd=9 efd=10"
[2023-04-16T17:26:46.633 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-16T16:26:46: Client 8: new arrival from 127.0.0.1:50550 - welcome!"
[2023-04-16T17:26:46.633 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2023-04-16T17:26:46.668 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-16T16:26:46: Driver indi_pentax_ccd: Number of cameras detected: 1."
[2023-04-16T17:26:46.668 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-16T16:26:46: Driver indi_pentax_ccd: Detected camera model USB PTP Class Camera on port usb:001,012"
[2023-04-16T17:26:46.668 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-16T16:26:46: Driver indi_pentax_ccd: Failed to find model USB PTP Class Camera in supported cameras."
[2023-04-16T17:26:46.668 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-16T16:26:46: Driver indi_pentax_ccd: Falling back to generic name."
[2023-04-16T17:26:46.668 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2023-04-16T17:26:56.183 BST INFO ][           org.kde.kstars.ekos] - "Unable to establish:\n+ Pentax DSLR\nPlease ensure the device is connected and powered on."`

I don't recall which driver I tested or mode I was in with the camera but I have also got logs containing this:

`[2023-04-16T17:21:40.685 BST INFO ][           org.kde.kstars.ekos] - "INDI services started on port 7,624."
[2023-04-16T17:21:40.686 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-16T16:21:40: startup: /usr/bin/indiserver -v -p 7624 -m 1024 -r 0 -f /tmp/indififo4e8ee5e4"
[2023-04-16T17:21:40.686 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-16T16:21:40: listening to port 7624 on fd 5"
[2023-04-16T17:21:40.686 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-16T16:21:40: Local server: listening on local domain at: @/tmp/indiserver"
[2023-04-16T17:21:40.686 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-16T16:21:40: FIFO: start indi_pentax"
[2023-04-16T17:21:40.686 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-16T16:21:40: FIFO: Starting driver indi_pentax"
[2023-04-16T17:21:40.686 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-16T16:21:40: Driver indi_pentax: pid=37020 rfd=9 wfd=9 efd=10"
[2023-04-16T17:21:40.686 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-16T16:21:40: Client 8: new arrival from 127.0.0.1:34316 - welcome!"
[2023-04-16T17:21:40.686 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2023-04-16T17:21:40.695 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-16T16:21:40: Driver indi_pentax: Looking for Pentax camera in  PTP mode."
[2023-04-16T17:21:40.695 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-16T16:21:40: Driver indi_pentax: Looking for Pentax camera in MSC mode."
[2023-04-16T17:21:40.695 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-16T16:21:40: Driver indi_pentax: ioctl: Operation not permitted"
[2023-04-16T17:21:40.695 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-16T16:21:40: Driver indi_pentax: /vr/home/kaya/Projects/indi-3rdparty/libpktriggercord/src/pslr.c:1712:scsi_write(fd, cmd, sizeof (cmd), 0, 0) failed: 1"
[2023-04-16T17:21:40.695 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-16T16:21:40: Driver indi_pentax: /vr/home/kaya/Projects/indi-3rdparty/libpktriggercord/src/pslr.c:1302:command(p->fd, 0, 1, 0) failed: 1"
[2023-04-16T17:21:40.695 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-16T16:21:40: Driver indi_pentax: /vr/home/kaya/Projects/indi-3rdparty/libpktriggercord/src/pslr.c:420:ipslr_status(p, statusbuf) failed: 1"
[... cut ...] pported Pentax cameras were found.  Perhaps the camera is not supported, not powered up, or needs to be in MSC mode?"
[2023-04-16T17:21:40.695 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  ""
[2023-04-16T17:21:40.697 BST DEBG ][           org.kde.kstars.indi] - INDI Server:  "2023-04-16T16:21:40: Driver indi_pentax: indi_pentax XML error: Line 1: early XML EOF"`

The camera is a Pentax K1 mkII which used to work both in MSC and PTP modes. Now, each mode has seems to be having issues

Angoosh, 2023-10-09:
I get `Driver indi_pentax: select: Bad file descriptor` on RPi 4 8gb when I manually builded indilib and pentax driver. Os is RPi OS bullseye, but the same happens on bookworm. Any help would be appreciated.

knro, 2024-01-23:
@karlrees Can you please check this issue? Is it possible to have this driver running under arm64 given the external dependencies? We need amd64 and arm64 builds at least.

karlrees, 2024-05-21:
It's been a long time since I looked at the code.  I had thought it was working with arm64, but maybe something changed.  What probably needs to happen for arm64 is that the indi-pentax driver needs to be told not to look for the libricohcamerasdk library (which was used for PTP mode), and to expect only to be able to use the pktriggercord-based library (for MSC mode only).

knro, 2024-05-22:
Given that PTP mode is no longer maintained (if I understood correctly), maybe the driver can be updated to only operate with pktriggercoord ?

karlrees, 2024-05-23:
I installed on arm64 tonight, and the bad file descriptor message is actually unrelated to PTP mode.  The driver will try to connect to PTP mode first, and then if it cannot connect (or if the library is unavailable), it tries MSC mode.  If it still cannot connect, it is supposed to then give an error message suggesting that the user switch the camera to MSC mode, and be done.

Unfortunately, what happens instead is that after giving the error message, it starts spurting out so many bad file descriptor errors that the original error message is never seen.  I will dig in more closely to see what happens.  In the interim, I can confirm that the messages do not appear if you are already connected via MSC mode when you start the driver.

knro, 2024-05-23:
Thank you for the update @karlrees Hopefully a fix can be released soon!

karlrees, 2024-05-23:
I think I solved it - looks like a memory deallocation problem with the libpktriggercord driver.  On the plus side, there are a few minor enhancements that I'll pull into the libpktriggercord driver as well.  Hopefully I'll have time to do it all this weekend.

knro, 2024-06-08:
@karlrees Any update on this issue?

Graf2242, 2024-07-02:
Probably I found a solution for K1 II too in https://github.com/indilib/indi-3rdparty/pull/936

Should be a common problem with ioctl and if so it may be fixed with 
`setcap cap_sys_rawio+ep /usr/bin/indi_pentax` for now

knro, 2024-07-03:
Ok that's a workaround but users would have to apply this directly. There is no way to do this programatically within the driver without escalated privileges?

Graf2242, 2024-07-03:
Investigating this now. Looks like it's possible to run `setcap` from CMakeLists ([example](https://gitweb.gentoo.org/repo/gentoo.git/diff/games-emulation/pcsx2/files/pcsx2-1.6.0-disable-setcap.patch?id=5d316c08962efd9c4f8dd213206aaea9de686f8f)). I need some time to test it

closed 2024-07-24 by knro