indilib/indi-3rdparty #779: indi_duino driver not serial connect on Ubuntu Mate22.04ARM64.

T-STUDIO, 2023-04-28:
After updating to 2.0.1 version, the indi_dhuino driver does not connect serially.

symptoms
・The serial port is displayed. (but can't connect)

・Turn off services that adversely affect such as modem manager and brltty

・Check serial communication with cutecom

environment
RaspberryPi4 4GB
[log_08-09-54.txt](https://github.com/indilib/indi-3rdparty/files/11348778/log_08-09-54.txt)

Ubuntu22.04

gnthibault, 2023-07-30:
Any update if this has been confirmed/fixed ?
I am experiencing issues while trying to connect to my Arduino nano (no issue through Arduino IDE V1), I suspect the issue comes from serial port management but I am not sure.

gnthibault, 2023-08-21:
I moved to a mele mini computer on x86. I had to downgrade bth indi-3rd party and indi to version 1.9.8 in order to be able to connect with indi_duino.
For me it means there is somehow a bug either in indi-duino or indi regarding the serial connection

T-STUDIO, 2025-07-17:
It still hasn't healed...

knro, 2025-07-30:
Not sure, need to test on an Adruino board to see if I can replicate this.

T-STUDIO, 2025-07-30:
Have you checked the log we submitted?

The device is recognized serially, but the driver cannot connect to the device.

This driver is no longer able to connect after an update that uses FIFO for connections.

Replacing it with an older driver will cause it to crash.

All we can do is submit the log and let you know there's a problem.

closed 2025-08-02 by knro

gnthibault, 2025-08-02:
Will test within the next few hours and let you know, thank you @knro

gnthibault, 2025-08-03:
Sorry I had to delay my test, I'll come back at it very soon