indilib/indi-3rdparty #789: indi-playerone unable to connect to Ekos

NeilHushush, 2023-05-26:
I'm trying to connect my Artemis-M 492 Mono and my Phoenix Wheel 7x36mm to kstars/ekos but so far I haven't been able to get them to connect using the inid-playerone driver. Both instruments work fine in NINA with proprietary drivers (not the ASCOM ones though).

I've been trying both on Arch/Garuda and Kubuntu, and in both distros there are fairly updated pkgs in the AUR and official repos. Other drivers can connect easily (indi-eqmod for my mount) so I don't think I'm missing anything fundamental (lib or dep). Both distro run on systemd, where I linked the 99-player_one rule from /usr/lib/udev to /etc/udev/rules.d. I tried changing the port from 7264 (default) to some random number in between 7264 and 8263. That didn't change anything.

On Arch I also tried to ['custom build'](https://gist.github.com/italic-r/9e951be4574388590dcb5e07f616b77b) the pkg directly from the 3dparty github but again, no luck there either. The logs for the errors are quite lacky in details, basically saying

Unable to establish:
+ PlayerOne CCD
Please ensure the device is connected and powered on.

More details about my attempts [here](https://www.cloudynights.com/topic/877042-playerone-stuff-and-indi-drivers-where-we-at/) and [here](https://indilib.org/forum/ekos/13400-playerone-drivers-issues-camera-and-filter-wheel.html#93057)

Kstars version is 3.6.2 stable
indi-playerone 0.9+ and libplayeronecamera2 3.1+ from official kubuntu repos
built from github or [AUR](https://aur.archlinux.org/packages/libindi_3rdparty) on Arch

hiro3110i, 2023-05-26:
@NeilHushush Thank you for your report.
Actually, Artemis is very new camera product, so KStars v3.6.4 (the latest official release today) doesn't support yet.
The latest indi-playerone driver code which was merged on GitHub is already support Artemis by using libPlayerOneCamera v3.3.0.
I believe KStars will officially update to v3.6.5 soon. It should be containing the latest indi-playerone driver.
Could you please wait a moment until v3.6.5 release?

If you can build indi core, indi-3rdparty drivers and KStars from source code, you will be able to use Artemis now.
I'm testing self built KStars/Ekos v3.6.5 beta with the latest indi-playerone driver on my Raspberry Pi4 (Ubuntu Mate 22.04).
It is working very well with my Poseidon-C/Uranus-C/Neptune-C II/Sedna-M and Phoenix Wheel 7x36mm.
(Sorry I don't have Artemis...)

NeilHushush, 2023-05-26:
Sure I can wait no problem. Or I could try to compile them as well. Seems like a good task to keep the hamsters busy during the weekend.

Btw, do kstars and indi require systemd? Cause I run gentoo on openrc as my daily driver and I'd really like to make it work over there too.

hiro3110i, 2023-05-27:
@NeilHushush Building a system with Gentoo is interesting.

I'm not that knowledgeable, so it's hard to give an exact answer.
At least indi's usb devices rely on udev.
Personally, I have no problem with 32bit debian (arm only) and 64bit ubuntu (arm and x86).

NeilHushush, 2023-05-27:
Afaik in all non-systemd systems (opernrc, runit...) all basic udev functionalities are handled by elogind, so I hope it won't be an issue. But I'm 99.99% confident that if there will be an issue it'll be related to systemd (or lack thereof).

So far so good, indi, indi-eqmod, libplayerone and indi-playerone compiled successfully from sources using the info in the github. One piece I'm missing is Kstars 3.6.5 source code. All I can find is 3.6.4. I think I'll go with that and report back, but if you know where to find it please let me know.

knro, 2023-05-27:
There are no systemd requirements. INDI 2.0.2 is due May 31st.

closed 2023-05-27 by knro

hiro3110i, 2023-05-28:
@knro Thank you for closing the topic.

@NeilHushush Just information about KStars repository.
You can clone the latest beta of v3.6.5 from here.
https://invent.kde.org/education/kstars

knro, 2023-05-28:
I'd like to point out that this isn't related to KStars at all, it's matter of INDI drivers update. So any INDI client using latest drivers would be work OK.

NeilHushush, 2023-06-02:
I can confirm the camera works perfectly with indi 2.0.2 and kstars 3.6.5 on Arch. Some parts taken from AUR (indilib), some compiled manually (playerone libs and drivers), and kstars from official repo.