indilib/indi-3rdparty #793: eqmod: code duplication

rsarwar87, 2023-06-02:
Hi I run a custom board on my mount, which requires me to load up rebuild the eqmod such that it uses the a different protocol to talk to my board, which is why i frequently keep track of all the change in indi-eqmod and merge changes from here. Yesterday, i was making the jump from tag 2.0.0 to 2.0.2 and found oddities:

InquireBoardVersion
InquireRAEncoderInfo
InquireDEEncoderInfo
The above function has been overloaded to reflect change in the indi api, i.e. to move away from deprecated functions. However there is a lot of duplication of code involved in the way it was set up. will the code be cleaned up, either by removing the deprecated function all together, or by encapsulating the duplicate codes into a separate function? If not, i will clean it up for my benefit, but i donot wish to make the change myself if it will be done by as that will lead to merge conflicts and i hate dealing them.

knro, 2023-06-04:
Please go ahead and then we can review the changes.

rsarwar87, 2023-06-04:
Ill give it a go, but the problem is i dont have a mount that can test the current v2.0.2 implementation. none the less, i did end up implementing it:

```
 12 void Skywatcher::InquireRAEncoderInfo(INDI::PropertyNumber encoderNP)
  11 {
  10     double steppersvalues[3];
   9     const char *steppersnames[] = { "RASteps360", "RAStepsWorm", "RAHighspeedRatio" };
   8     InquireEncoderInfo(Axis1, steppersvalues);
   7     LOGF_INFO("%s(): %f %f %f", __func__, steppersvalues[0], steppersvalues[1], steppersvalues[2]);
   6     // should test this is ok
   5     encoderNP.update(steppersvalues, (char **)steppersnames, 3);
   4     encoderNP.apply();
   3 } 

```
the LOGF_INFO produces the right output according to my device config:
```
2023-06-04T18:35:30: [INFO] InquireDEEncoderInfo(): 9216000.000000 50000000.000000 1.000000 
2023-06-04T18:35:30: [INFO] InquireRAEncoderInfo(): 9216000.000000 50000000.000000 1.000000 
```

however, the Fireware page is incorrect. Seems like the lines 5 and 4 in the above snip did not actuate. can someone please confirm that the firmware page is indeed correct. Note that this is a display issue, and eqmod actually functions correctly as the values were correctly updated.

[image]

rsarwar87, 2023-06-04:
sorted, apparently 50MHz freq is beyond the range of the encoderNP. i just devided it by 5, but that is specific to my mount implementation only of course. No warning was posted -- the deprecated function did post warnings if i recall correctly.

#796

knro, 2023-06-05:
Thanks, I'll take a look at and test locally.