indilib/indi-3rdparty #794: indi-3rdparty drivers fail to compile on Debian 12 Bookworm

nordcomputer, 2023-06-03:
**Describe the bug**
3rdparty drivers fail compiling on Debian Bookworm.

**To Reproduce**
Exact steps to reproduce the behavior:
Install dependencies and compile and install indi-3rdparty-libs as in the description.
run:
```
mkdir -p ~/Projects/build/indi-3rdparty
cd ~/Projects/build/indi-3rdparty
cmake -DCMAKE_INSTALL_PREFIX=/usr -DCMAKE_BUILD_TYPE=Debug ~/Projects/indi-3rdparty
make -j4
```
 The process fails at 38%. 

**Expected behavior**
When you have installed all dependencies, indi-server and indi-3rdparty-libs the drivers should compile just fine.

**Desktop (please complete the following information):**
 - OS: Debian 12 (Bookworm)
 - Version: Commit fa2b41cf317d020885d6f169df721a39d1869db4

**Log Files**
Output of the failed compile process: https://pastebin.com/15vFDED1

knro, 2023-06-04:
Did you compile and install INDI first from GIT?

tstibor, 2023-06-04:
I compiled approx. 2 weeks ago INDI + INDI-3rd-party on Debian 12 (bookworm). It works without any problems.
As Jasem suggests, make sure first to build and install INDI and then INDI-3rd-party. Installing INDI from official Debian 12 repo. and
mixing it with INDI-3rd-party from github, will not work.

nordcomputer, 2023-06-04:
Hey there, thank you for your answer. 
I also compiled it 2 weeks ago on a second machine without problems. And yes, I compiled the indi server from the official git repo first - BUT: I may have messed up compiling the 3rdparty drivers and libs in a first attempt, as I can see multiple drivers that are not part of the core indi-server like the DSLR drivers. I am using indistarter for that. 
I will try to start from the beginning and recompile the core indi-server and then the libs and then the drivers, if you have no more suggestions for now.

tstibor, 2023-06-04:
Actually the cleanest approach is to build DEB package from github. I developed the following simple bash
script.

```
#!/bin/bash

SCRIPT_NAME=$(basename $0)

BUILD_CORE=0
BUILD_EXTRA=0
LIST_INDI_DRIVERS=0
INDI_DRIVERS_DEFAULT=(libatik indi-atik libmicam indi-mi indi-gpsd indi-sx indi-celestronaux)
INDI_DRIVERS_OPTION=()

DEB_BUILD_OPTIONS="parallel=$(nproc --all)"
MAIN_PATH=$(pwd)
INDI_CORE_PATH=${MAIN_PATH}/indi
INDI_EXTRA_PATH=${MAIN_PATH}/indi-3rdparty
DEB_CORE_PATH=${MAIN_PATH}/deb_core
DEB_EXTRA_PATH=${MAIN_PATH}/deb_extra

__usage() {
   cat << DOC

   Usage: $SCRIPT_NAME [--core] [--extra <NAMES>] [--list]

   optional arguments:
     -h, --help           show this help message and exit
     -c, --core           build INDI core package
     -e, --extra <NAMES>  build INDI extra packages listed in <NAMES>
                          if <NAMES> is empty default packages are used
     -l, --list           show list of default extra packages

   version: 0.1, thomas@stibor.net, Mon 23 Jan 2023 09:40:09 AM CET
DOC
}  

OPTS=$(getopt -o "celh" --long "help,core,extra,list" -n "$SCRIPT_NAME" -- "$@")
if [ $? != 0 ] ; then
    echo "Error in command line arguments." >&2 ;
    __usage ;
    exit 1 ;
fi
eval set -- "$OPTS"

while true; do
    case "$1" in
        -h | --help ) __usage; exit; ;;
        -c | --core ) BUILD_CORE=1; shift ;;
        -e | --extra ) BUILD_EXTRA=1; shift ;;
        -l | --list ) LIST_INDI_DRIVERS=1; shift ;;    
        -- ) shift; break ;;
        * ) break ;;
    esac
    shift
    N=$#
    for ((n=0; n < N; ++n)); do
        INDI_DRIVERS_OPTION+=("$1")
        shift
    done
done

if [[ $LIST_INDI_DRIVERS -eq 0 && $BUILD_CORE -eq 0 && $BUILD_EXTRA -eq 0 ]] ; then
    __usage
fi

if [[ $LIST_INDI_DRIVERS -eq 1 ]] ; then
    echo "${INDI_DRIVERS_DEFAULT[@]}" && exit 0
fi

if [[ $BUILD_CORE -eq 1 ]] ; then
    # Create empty core directory for DEB packages.
    rm -rf ${DEB_CORE_PATH}
    mkdir -p ${DEB_CORE_PATH}

    # Build INDI core.
    cp -r ${INDI_CORE_PATH} ${DEB_CORE_PATH} && cd ${DEB_CORE_PATH}/indi
    fakeroot debian/rules binary
fi

if [[ $BUILD_EXTRA -eq 1 ]] ; then
    # Create empty extra directory for DEB packages.
    rm -rf ${DEB_EXTRA_PATH}
    mkdir -p ${DEB_EXTRA_PATH}

    DRIVERS=${INDI_DRIVERS_DEFAULT[@]}
    
    if [[ -n ${INDI_DRIVERS_OPTION} ]]; then
        DRIVERS=${INDI_DRIVERS_OPTION[@]}
    fi
    
    for DRVS in ${DRIVERS[@]}
    do
        PATH_DRVS=${DEB_EXTRA_PATH}/${DRVS}
        mkdir -p ${PATH_DRVS}/{${DRVS},debian,cmake_modules} && cd ${PATH_DRVS}
        cp -r ${INDI_EXTRA_PATH}/${DRVS}/* ${PATH_DRVS}/${DRVS}
        cp -r ${INDI_EXTRA_PATH}/debian/${DRVS}/* ${PATH_DRVS}/debian/
        cp -r ${INDI_EXTRA_PATH}/cmake_modules/* ${PATH_DRVS}/cmake_modules/
        fakeroot debian/rules binary
    done
fi
```

Checkout indi and indi-3rdparty into directory and add there the script:
```
[tstibor@x230 ~/dev/astronomy/indi-build] ll
total 12
-rwxr-xr-x  1 tstibor tstibor 2604 Jun  4 11:29 build-deb.sh
drwxr-xr-x 18 tstibor tstibor 4096 Jun  4 11:28 indi
drwxr-xr-x 90 tstibor tstibor 4096 Jun  4 11:28 indi-3rdparty
```

The following command will build the INDI core DEB package:

`[tstibor@x230 ~/dev/astronomy/indi-build] ./build-deb.sh -c`

Then I installed the DEB package and build the listed 3rd-party packages (libatik indi-atik libmicam indi-mi indi-gpsd indi-sx indi-celestronaux)

`[tstibor@x230 ~/dev/astronomy/indi-build] ./build-deb.sh -e`

nordcomputer, 2023-06-04:
Ok, short update: after recompiling INDI core, it worked. The script for building the deb package looks promising though - Does it create all packages (indi, indi-3rdparty-libs and indi-3rdparty)? That would be very helpful, so you dont have to recompile every time. 

Another thing and a small suggestion for the readme file - it is kind of irritating, that the part with the drivers comes first, then there is the part for the libs and then there is the part for installing all drivers. I suggest to put the libs part first, as you dont have to jump up and down. But that is just a suggestion, as I stumbled over it. I will close the issue for now.

Thank you for your help @tstibor @knro