indilib/indi-3rdparty #800: Camera failed to start and get Temperature Permission Denied

jcmartinez, 2023-06-13:
**Describe the bug**

When connecting an OGMA camera, we get an error indicating that the camera failed to start and that INDI can't control the temperature.

This affects `indi_ogmacam_ccd` and probably it also affects `indi_toupcam_ccd`

**To Reproduce**
Exact steps to reproduce the behavior.
1. Run KStars
2. Select a profile that uses Camera = OGMACam (or Toupcam)
3. Click Connect
4. In the INDI Control Panel, under the Tab OGMAVision (or Toupcam), click Connect
5. You will see in the logs: 
2023-06-12T21:46:56: [ERROR] get Temperature error. Permission denied 
2023-06-12T21:46:55: [ERROR] get Temperature error. Permission denied 
2023-06-12T21:46:54: [ERROR] Failed to start camera. Permission denied 
2023-06-12T21:46:54: [INFO] OGMAVision AP26MC connect 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Desktop

This has been detected in the following system:

- INDI installed on Ubuntu desktop on a PC

**Log Files**
[log_18-05-27.zip](https://github.com/indilib/indi-3rdparty/files/11737190/log_18-05-27.zip)
 following https://github.com/avarakin/AstroPiMaker4

jcmartinez, 2023-06-14:
Linux Audit produced logs indicating that the current user that's executing `/usr/bin/indi_ogmacam_ccd` doesn't have file permission to write to the USB device. This explains why the camera is detected, but no configurations or commands can be sent to the camera.

These are the logs from Linux Audit:

`type=SYSCALL msg=audit(1686765633.743:357): arch=c000003e syscall=257 success=no exit=-13 a0=ffffff9c a1=7ffea6d64f20 a2=80002 a3=0 items=1 ppid=5825 pid=5827 auid=1000 uid=1000 gid=1000 euid=1000 suid=1000 fsuid=1000 egid=1000 sgid=1000 fsgid=1000 tty=(none) ses=2 comm="indi_ogmacam_cc" exe="/usr/bin/indi_ogmacam_ccd" subj=unconfined key="access"ARCH=x86_64 SYSCALL=openat AUID="jc" UID="jc" GID="jc" EUID="jc" SUID="jc" FSUID="jc" EGID="jc" SGID="jc" FSGID="jc"
type=CWD msg=audit(1686765633.743:357): cwd="/home/jc"
type=PATH msg=audit(1686765633.743:357): item=0 name="/dev/bus/usb/002/002" inode=733 dev=00:05 mode=020664 ouid=0 ogid=0 rdev=bd:81 nametype=NORMAL cap_fp=0 cap_fi=0 cap_fe=0 cap_fver=0 cap_frootid=0OUID="root" OGID="root"
type=PROCTITLE msg=audit(1686765633.743:357): proctitle="indi_ogmacam_ccd"`

Listing my USB devices I can see the location of the camera:

`$ sudo lsusb
Bus 002 Device 002: ID 9745:9428 TP AP26MC`

Listing permissions I can see that other users don't have write permission:

`$ sudo ls -la /dev/bus/usb/002/002
crw-rw-r-- 1 root root 189, 129 Jun 14 14:25 /dev/bus/usb/002/002`

After adding write permission to others, everything works fine:

`$ sudo chmod o+w /dev/bus/usb/002/002
$ sudo ls -la /dev/bus/usb/002/002
crw-rw-rw- 1 root root 189, 129 Jun 14 13:37 /dev/bus/usb/002/002`

ckuethe, 2023-06-14:
Seems like https://github.com/indilib/indi-3rdparty/blob/master/libogmacam/99-ogmacam.rules is missing an entry for vendor `0x9745`.

Try add it to the an appropriate rules file in `/etc/udev/rules.d` so that udev will set permissions properly.

knro, 2023-08-11:
The missing VID was added already.

closed 2023-08-11 by knro

cemno, 2025-12-19:
I just had the same problem with an flatpak install on my machine with an omegon velox 178c camera. Just found this issue and was able to get it to work with @jcmartinez solution. But this is only a temporary solution. Is there a permanent fix i am not aware of?