indilib/indi-3rdparty #813: error: ‘virtual void ASIBase::addFITSKeywords(INDI::CCDChip*)’ marked ‘override’, but does not override

mzeising, 2023-07-11:
**Describe the bug**

`indi-asi` fails to build with the error `error: ‘virtual void ASIBase::addFITSKeywords(INDI::CCDChip*)’ marked ‘override’, but does not override`

**To Reproduce**

`$ cmake -DCMAKE_INSTALL_PREFIX=/usr ~/sources/indi-3rdparty/indi-asi`

```
-- The CXX compiler identification is GNU 10.2.1
-- The C compiler identification is GNU 10.2.1
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found ASI: /usr/lib/arm-linux-gnueabihf/libASICamera2.so;/usr/lib/arm-linux-gnueabihf/libEFWFilter.so;/usr/lib/arm-linux-gnueabihf/libUSB2ST4Conv.so;/usr/lib/arm-linux-gnueabihf/libEAFFocuser.so
-- Found CFITSIO 3.49: /usr/lib/arm-linux-gnueabihf/libcfitsio.so
-- Found INDI: /usr/lib/arm-linux-gnueabihf/libindidriver.so;/usr/lib/arm-linux-gnueabihf/libindiAlignmentDriver.so (found version "2.0.2")
-- Found ZLIB: /usr/lib/arm-linux-gnueabihf/libz.so (found version "1.2.11")
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.2")
-- Found USB1: /usr/lib/arm-linux-gnueabihf/libusb-1.0.so (found version "1.0.24")
-- Performing Test USB1_HAS_LIBUSB_ERROR_NAME
-- Performing Test USB1_HAS_LIBUSB_ERROR_NAME - Success
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Performing Test COMPATIBLE_FORTIFY_SOURCE
-- Performing Test COMPATIBLE_FORTIFY_SOURCE - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/binaries/indi-3rdparty
```

`$ make -j4`

```
Scanning dependencies of target asi_camera_test
Scanning dependencies of target indi_asi_focuser
Scanning dependencies of target indi_asi_wheel
Scanning dependencies of target indi_asi_st4
[  7%] Building CXX object CMakeFiles/asi_camera_test.dir/asi_camera_test.cpp.o
[ 21%] Building CXX object CMakeFiles/indi_asi_st4.dir/asi_st4.cpp.o
[ 21%] Building CXX object CMakeFiles/indi_asi_wheel.dir/asi_wheel.cpp.o
[ 28%] Building CXX object CMakeFiles/indi_asi_focuser.dir/asi_focuser.cpp.o
[ 35%] Linking CXX executable asi_camera_test
[ 35%] Built target asi_camera_test
Scanning dependencies of target indi_asi_single_ccd
[ 42%] Building CXX object CMakeFiles/indi_asi_single_ccd.dir/asi_base.cpp.o
[ 50%] Linking CXX executable indi_asi_st4
[ 57%] Linking CXX executable indi_asi_wheel
In file included from /home/pi/sources/indi-3rdparty/indi-asi/asi_base.cpp:23:
/home/pi/sources/indi-3rdparty/indi-asi/asi_base.h:76:22: error: ‘virtual void ASIBase::addFITSKeywords(INDI::CCDChip*)’ marked ‘override’, but does not override
   76 |         virtual void addFITSKeywords(INDI::CCDChip *targetChip) override;
      |                      ^~~~~~~~~~~~~~~
[ 57%] Built target indi_asi_st4
Scanning dependencies of target indi_asi_ccd
[ 64%] Building CXX object CMakeFiles/indi_asi_ccd.dir/asi_base.cpp.o
/home/pi/sources/indi-3rdparty/indi-asi/asi_base.cpp: In member function ‘virtual void ASIBase::addFITSKeywords(INDI::CCDChip*)’:
[... 47 lines cut ...]
make: *** [Makefile:149: all] Error 2
```

**Expected behavior**
Build finishes

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: 
 - Version [e.g. 1.7.6, GIT..etc]

`gcc -v`

```
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/10/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Raspbian 10.2.1-6+rpi1' --with-bugurl=file:///usr/share/doc/gcc-10/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-10 --program-prefix=arm-linux-gnueabihf- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --disable-libquadmath-support --enable-plugin --with-system-zlib --enable-libphobos-checking=release --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --disable-werror --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 10.2.1 20210110 (Raspbian 10.2.1-6+rpi1)
```

knro, 2023-07-12:
is latest INDI from GIT installed as well?

mzeising, 2023-07-12:
Hi Jasem, yes it is, I did a `git pull`, `cmake`, `make install` right before

knro, 2023-08-11:
Cannot reproduce, you must have an old INDI installation somewhere, perhaps in /usr/local

closed 2024-01-17 by knro