indilib/indi-3rdparty #823: Driver for Altair 183M Pro keeps failing because of temperature reading

gnthibault, 2023-08-10:
**Describe the bug**
affected program: indi_altair_ccd
The following error keeps appearing very often, making the camera very hard to use:
[ERROR] get Temperature error. Generic failure 

**To Reproduce**

1. Run driver:
indiserver indi_altair_ccd
3. Perform action
Nothing specific, it seems more or less random
5. See error
In Device manager you will see: [ERROR] get Temperature error. Generic failure  

**Expected behavior**
I expect the driver to work properly

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: 
Description:	Manjaro ARM Linux
Release:	23.02
 - Version: v2.0.2

knro, 2023-08-10:
Can you please try with the latest SDK released in INDI v2.0.3 ?

gnthibault, 2023-08-10:
Will test tonight, thank you for your feedback. In the meantime, I simply changed the code to force the temperature to 20 degrees instead of doing the proper getTemperature call that fails

gnthibault, 2023-08-10:
I confirm the issue has not been fixed with the new driver

gnthibault, 2023-08-22:
I'll test this soon with 2.0.3 under a x86 platform