indilib/indi-3rdparty #828: 3rd-party-drivers not compiling (again)

nordcomputer, 2023-09-02:
I tried to compile the drivers today - the libs compiled fine, but the drivers dont. I am using Debian 12 (Stable - bookworm).
Here is my log:
```
make -j4 
[  1%] Building C object indi-eqmod/CMakeFiles/indi_eqmod_telescope.dir/align/htm.c.o
[  1%] Building C object indi-eqmod/CMakeFiles/indi_azgti_telescope.dir/align/htm.c.o
[  1%] Building C object indi-eqmod/CMakeFiles/indi_ahpgt_telescope.dir/align/htm.c.o
[  1%] Building C object indi-eqmod/CMakeFiles/indi_skyadventurergti_telescope.dir/align/htm.c.o
[  1%] Building C object indi-eqmod/CMakeFiles/indi_eqmod_telescope.dir/align/chull/chull.c.o
[  1%] Building C object indi-eqmod/CMakeFiles/indi_azgti_telescope.dir/align/chull/chull.c.o
[  2%] Building C object indi-eqmod/CMakeFiles/indi_skyadventurergti_telescope.dir/align/chull/chull.c.o
[  2%] Building C object indi-eqmod/CMakeFiles/indi_ahpgt_telescope.dir/align/chull/chull.c.o
[  3%] Building CXX object indi-eqmod/CMakeFiles/indi_eqmod_telescope.dir/eqmod.cpp.o
[  3%] Building CXX object indi-eqmod/CMakeFiles/indi_ahpgt_telescope.dir/ahp-gt/ahpgt.cpp.o
[  4%] Building CXX object indi-eqmod/CMakeFiles/indi_azgti_telescope.dir/azgti.cpp.o
[  4%] Building CXX object indi-eqmod/CMakeFiles/indi_skyadventurergti_telescope.dir/skyadventurergti.cpp.o
[  4%] Building CXX object indi-eqmod/CMakeFiles/indi_skyadventurergti_telescope.dir/skyadventurergtibase.cpp.o
[  4%] Building CXX object indi-eqmod/CMakeFiles/indi_ahpgt_telescope.dir/ahp-gt/ahpgtbase.cpp.o
[  4%] Building CXX object indi-eqmod/CMakeFiles/indi_azgti_telescope.dir/azgtibase.cpp.o
[  5%] Building CXX object indi-eqmod/CMakeFiles/indi_eqmod_telescope.dir/eqmodbase.cpp.o
[  5%] Building CXX object indi-eqmod/CMakeFiles/indi_azgti_telescope.dir/eqmodbase.cpp.o
[  5%] Building CXX object indi-eqmod/CMakeFiles/indi_skyadventurergti_telescope.dir/eqmodbase.cpp.o
[  5%] Building CXX object indi-eqmod/CMakeFiles/indi_ahpgt_telescope.dir/eqmodbase.cpp.o
[  5%] Building CXX object indi-eqmod/CMakeFiles/indi_eqmod_telescope.dir/eqmoderror.cpp.o
[  6%] Building CXX object indi-eqmod/CMakeFiles/indi_skyadventurergti_telescope.dir/eqmoderror.cpp.o
[  6%] Building CXX object indi-eqmod/CMakeFiles/indi_eqmod_telescope.dir/skywatcher.cpp.o
[  6%] Building CXX object indi-eqmod/CMakeFiles/indi_azgti_telescope.dir/eqmoderror.cpp.o
[  7%] Building CXX object indi-eqmod/CMakeFiles/indi_ahpgt_telescope.dir/eqmoderror.cpp.o
[  7%] Building CXX object indi-eqmod/CMakeFiles/indi_skyadventurergti_telescope.dir/skywatcher.cpp.o
[  8%] Building CXX object indi-eqmod/CMakeFiles/indi_azgti_telescope.dir/skywatcher.cpp.o
[  8%] Building CXX object indi-eqmod/CMakeFiles/indi_ahpgt_telescope.dir/skywatcher.cpp.o
[  8%] Building CXX object indi-eqmod/CMakeFiles/indi_eqmod_telescope.dir/simulator/simulator.cpp.o
[  9%] Building CXX object indi-eqmod/CMakeFiles/indi_eqmod_telescope.dir/simulator/skywatcher-simulator.cpp.o
[  9%] Building CXX object indi-eqmod/CMakeFiles/indi_azgti_telescope.dir/simulator/simulator.cpp.o
[  9%] Building CXX object indi-eqmod/CMakeFiles/indi_skyadventurergti_telescope.dir/simulator/simulator.cpp.o
[  9%] Building CXX object indi-eqmod/CMakeFiles/indi_eqmod_telescope.dir/align/align.cpp.o
[  9%] Building CXX object indi-eqmod/CMakeFiles/indi_ahpgt_telescope.dir/simulator/simulator.cpp.o
[  9%] Building CXX object indi-eqmod/CMakeFiles/indi_azgti_telescope.dir/simulator/skywatcher-simulator.cpp.o
[  9%] Building CXX object indi-eqmod/CMakeFiles/indi_skyadventurergti_telescope.dir/simulator/skywatcher-simulator.cpp.o
[ 10%] Building CXX object indi-eqmod/CMakeFiles/indi_ahpgt_telescope.dir/simulator/skywatcher-simulator.cpp.o
[ 12%] Building CXX object indi-eqmod/CMakeFiles/indi_skyadventurergti_telescope.dir/align/align.cpp.o
[ 12%] Building CXX object indi-eqmod/CMakeFiles/indi_azgti_telescope.dir/align/align.cpp.o
[ 12%] Building CXX object indi-eqmod/CMakeFiles/indi_ahpgt_telescope.dir/align/align.cpp.o
[ 12%] Building CXX object indi-eqmod/CMakeFiles/indi_eqmod_telescope.dir/align/pointset.cpp.o
[ 12%] Building CXX object indi-eqmod/CMakeFiles/indi_skyadventurergti_telescope.dir/align/pointset.cpp.o
[ 12%] Building CXX object indi-eqmod/CMakeFiles/indi_azgti_telescope.dir/align/pointset.cpp.o
[ 12%] Building CXX object indi-eqmod/CMakeFiles/indi_ahpgt_telescope.dir/align/pointset.cpp.o
[ 13%] Building CXX object indi-eqmod/CMakeFiles/indi_eqmod_telescope.dir/align/triangulate.cpp.o
[ 13%] Building CXX object indi-eqmod/CMakeFiles/indi_skyadventurergti_telescope.dir/align/triangulate.cpp.o
[ 13%] Building CXX object indi-eqmod/CMakeFiles/indi_azgti_telescope.dir/align/triangulate.cpp.o
[ 13%] Building CXX object indi-eqmod/CMakeFiles/indi_eqmod_telescope.dir/align/triangulate_chull.cpp.o
[ 13%] Building CXX object indi-eqmod/CMakeFiles/indi_ahpgt_telescope.dir/align/triangulate.cpp.o
[ 13%] Building CXX object indi-eqmod/CMakeFiles/indi_azgti_telescope.dir/align/triangulate_chull.cpp.o
[ 14%] Building CXX object indi-eqmod/CMakeFiles/indi_skyadventurergti_telescope.dir/align/triangulate_chull.cpp.o
[ 15%] Building CXX object indi-eqmod/CMakeFiles/indi_ahpgt_telescope.dir/align/triangulate_chull.cpp.o
[ 15%] Building CXX object indi-eqmod/CMakeFiles/indi_eqmod_telescope.dir/scope-limits/scope-limits.cpp.o
[ 16%] Building CXX object indi-eqmod/CMakeFiles/indi_azgti_telescope.dir/scope-limits/scope-limits.cpp.o
[ 16%] Building CXX object indi-eqmod/CMakeFiles/indi_skyadventurergti_telescope.dir/scope-limits/scope-limits.cpp.o
[ 16%] Building CXX object indi-eqmod/CMakeFiles/indi_ahpgt_telescope.dir/scope-limits/scope-limits.cpp.o
[ 17%] Linking CXX executable indi_eqmod_telescope
[ 17%] Built target indi_eqmod_telescope
[ 17%] Building C object indi-eqmod/CMakeFiles/indi_staradventurer2i_telescope.dir/align/htm.c.o
[ 17%] Building C object indi-eqmod/CMakeFiles/indi_staradventurer2i_telescope.dir/align/chull/chull.c.o
[ 18%] Building CXX object indi-eqmod/CMakeFiles/indi_staradventurer2i_telescope.dir/staradventurer2i.cpp.o
[ 18%] Linking CXX executable indi_azgti_telescope
[ 18%] Linking CXX executable indi_ahpgt_telescope
[ 18%] Linking CXX executable indi_skyadventurergti_telescope
[ 18%] Built target indi_ahpgt_telescope
[ 18%] Built target indi_azgti_telescope
[ 18%] Built target indi_skyadventurergti_telescope
[... 249 lines cut ...]
 2056 | void GPhotoCCD::addFITSKeywords(INDI::CCDChip * targetChip, std::vector<INDI::FITSRecord> &fitsKeywords)
      |                                                                                         ^
/home/mario/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.cpp:2056:89: error: template argument 2 is invalid
/home/mario/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.cpp: In member function ‘void GPhotoCCD::addFITSKeywords(INDI::CCDChip*, int&)’:
/home/mario/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.cpp:2058:31: error: no matching function for call to ‘GPhotoCCD::addFITSKeywords(INDI::CCDChip*&, int&)’
 2058 |     INDI::CCD::addFITSKeywords(targetChip, fitsKeywords);
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/mario/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.h:27:
/usr/include/libindi/indiccd.h:508:22: note: candidate: ‘virtual void INDI::CCD::addFITSKeywords(INDI::CCDChip*)’
  508 |         virtual void addFITSKeywords(CCDChip * targetChip);
      |                      ^~~~~~~~~~~~~~~
/usr/include/libindi/indiccd.h:508:22: note:   candidate expects 1 argument, 2 provided
/home/mario/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.cpp:2067:30: error: request for member ‘push_back’ in ‘fitsKeywords’, which is of non-class type ‘int’
 2067 |                 fitsKeywords.push_back({"ISOSPEED", isoSpeed, "ISO Speed"});
      |                              ^~~~~~~~~
/home/mario/Projects/indi-3rdparty/indi-gphoto/gphoto_ccd.cpp:2073:22: error: request for member ‘push_back’ in ‘fitsKeywords’, which is of non-class type ‘int’
 2073 |         fitsKeywords.push_back({"CCD-TEMP", TemperatureN[0].value, 3, "CCD Temperature (Celsius)"});
      |                      ^~~~~~~~~
make[2]: *** [indi-celestronaux/CMakeFiles/indi_celestron_aux.dir/build.make:90: indi-celestronaux/CMakeFiles/indi_celestron_aux.dir/celestronaux.cpp.o] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:1659: indi-celestronaux/CMakeFiles/indi_celestron_aux.dir/all] Fehler 2
make[2]: *** [indi-gphoto/CMakeFiles/indi_gphoto_ccd.dir/build.make:76: indi-gphoto/CMakeFiles/indi_gphoto_ccd.dir/gphoto_ccd.cpp.o] Fehler 1
make[2]: *** Es wird auf noch nicht beendete Prozesse gewartet....
make[1]: *** [CMakeFiles/Makefile2:1685: indi-gphoto/CMakeFiles/indi_gphoto_ccd.dir/all] Fehler 2
make: *** [Makefile:156: all] Fehler 2
```

knro, 2023-09-02:
You need to compile libindi first.

nordcomputer, 2023-09-02:
@knro 
I know and I did - but I will try again

nordcomputer, 2023-09-02:
nevermind - it worked after recompiling indi-core and the libs. Have a nice weekend!

closed 2023-09-02 by nordcomputer