indilib/indi-3rdparty #830: indi_svbony_ccd driver crash on connect (svbony sv705c camera)

pap3rinik, 2023-09-08:
**Describe the bug**
Diver crashes as soon as someone attempts to connect to a SVBony sv705c camera on Ubuntu Linux 23.04

**To Reproduce**

- start indiserver: eg. indiserver -v indi_svbony_ccd
- open kstars 3.6.6 (local or remote, any platform, doesn't matter)
- connect to indiserver
- start 705c camera
- indiserver crash or connection close

**Log Files**
```
2023-09-08T23:17:24: startup: indiserver -v indi_svbony_ccd
2023-09-08T23:17:24: Driver indi_svbony_ccd: pid=6314 rfd=6 wfd=6 efd=7
2023-09-08T23:17:24: listening to port 7624 on fd 5
2023-09-08T23:17:24: Local server: listening on local domain at: @/tmp/indiserver
2023-09-08T23:17:24: Driver indi_svbony_ccd: Camera(s) found
2023-09-08T23:17:24: Driver indi_svbony_ccd: snooping on Telescope Simulator.EQUATORIAL_EOD_COORD
2023-09-08T23:17:24: Driver indi_svbony_ccd: snooping on Telescope Simulator.EQUATORIAL_COORD
2023-09-08T23:17:24: Driver indi_svbony_ccd: snooping on Telescope Simulator.TELESCOPE_INFO
2023-09-08T23:17:24: Driver indi_svbony_ccd: snooping on Telescope Simulator.GEOGRAPHIC_COORD
2023-09-08T23:17:24: Driver indi_svbony_ccd: snooping on Telescope Simulator.TELESCOPE_PIER_SIDE
2023-09-08T23:17:24: Driver indi_svbony_ccd: snooping on Rotator Simulator.ABS_ROTATOR_ANGLE
2023-09-08T23:17:24: Driver indi_svbony_ccd: snooping on Focuser Simulator.ABS_FOCUS_POSITION
2023-09-08T23:17:24: Driver indi_svbony_ccd: snooping on Focuser Simulator.FOCUS_TEMPERATURE
2023-09-08T23:17:24: Driver indi_svbony_ccd: snooping on CCD Simulator.FILTER_SLOT
2023-09-08T23:17:24: Driver indi_svbony_ccd: snooping on CCD Simulator.FILTER_NAME
2023-09-08T23:17:24: Driver indi_svbony_ccd: snooping on SQM.SKY_QUALITY
2023-09-08T23:21:30: Client 9: new arrival from 192.168.1.135:60295 - welcome!
2023-09-08T23:21:45: Driver indi_svbony_ccd: Impossible IPState 14666
2023-09-08T23:21:45: Driver indi_svbony_ccd: Impossible IPerm 856293408
2023-09-08T23:21:45: Client 10: new arrival from 192.168.1.135:60303 - welcome!
2023-09-08T23:21:46: Driver indi_svbony_ccd: read: Connection reset by peer
<delProperty device="SVBONY SV705C 0"/>
2023-09-08T23:21:46: Driver indi_svbony_ccd: restart #0
2023-09-08T23:21:46: Driver indi_svbony_ccd: pid=6361 rfd=7 wfd=7 efd=11
2023-09-08T23:21:46: Driver indi_svbony_ccd: Camera(s) found
2023-09-08T23:21:46: Driver indi_svbony_ccd: snooping on Telescope Simulator.EQUATORIAL_EOD_COORD
2023-09-08T23:21:46: Driver indi_svbony_ccd: snooping on Telescope Simulator.EQUATORIAL_COORD
2023-09-08T23:21:46: Driver indi_svbony_ccd: snooping on Telescope Simulator.TELESCOPE_INFO
2023-09-08T23:21:46: Driver indi_svbony_ccd: snooping on Telescope Simulator.GEOGRAPHIC_COORD
2023-09-08T23:21:46: Driver indi_svbony_ccd: snooping on Telescope Simulator.TELESCOPE_PIER_SIDE
2023-09-08T23:21:46: Driver indi_svbony_ccd: snooping on Rotator Simulator.ABS_ROTATOR_ANGLE
2023-09-08T23:21:46: Driver indi_svbony_ccd: snooping on Focuser Simulator.ABS_FOCUS_POSITION
2023-09-08T23:21:46: Driver indi_svbony_ccd: snooping on Focuser Simulator.FOCUS_TEMPERATURE
2023-09-08T23:21:46: Driver indi_svbony_ccd: snooping on CCD Simulator.FILTER_SLOT
2023-09-08T23:21:46: Driver indi_svbony_ccd: snooping on CCD Simulator.FILTER_NAME
2023-09-08T23:21:46: Driver indi_svbony_ccd: snooping on SQM.SKY_QUALITY
```

knro, 2023-09-09:
Can you run it via [Ekos Debugger](https://stellarmate.com/support.html?view=kb&kbartid=16) and send us the backtrace?

pap3rinik, 2023-09-09:
Sure, here we go:

complete ekos-debugger output

```
This GDB supports auto-downloading debuginfo from the following URLs:
  <https://debuginfod.ubuntu.com>
Enable debuginfod for this session? (y or [n]) [answered N; input not from terminal]
Debuginfod has been disabled.
To make this setting permanent, add 'set debuginfod enabled off' to .gdbinit.
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[Attaching after Thread 0x7ffff7f89740 (LWP 6952) fork to child process 6955]
[New inferior 2 (process 6955)]
[Detaching after fork from parent process 6952]
[Inferior 1 (process 6952) detached]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
process 6955 is executing new program: /usr/bin/indi_svbony_ccd
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff52866c0 (LWP 6957)]
[New Thread 0x7ffff4a856c0 (LWP 6958)]
This GDB supports auto-downloading debuginfo from the following URLs:
  <https://debuginfod.ubuntu.com>
Enable debuginfod for this session? (y or [n]) [answered N; input not from terminal]
Debuginfod has been disabled.
To make this setting permanent, add 'set debuginfod enabled off' to .gdbinit.
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[Attaching after Thread 0x7ffff7f89740 (LWP 6992) fork to child process 6995]
[New inferior 2 (process 6995)]
[Detaching after fork from parent process 6992]
[Inferior 1 (process 6992) detached]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
process 6995 is executing new program: /usr/bin/indi_svbony_ccd
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff52866c0 (LWP 6997)]
[New Thread 0x7ffff4a856c0 (LWP 6998)]
[New Thread 0x7fffeffff6c0 (LWP 6999)]
[New Thread 0x7fffef7fe6c0 (LWP 7000)]
[New Thread 0x7fffecef36c0 (LWP 7044)]
Thread 2.1 "indi_svbony_ccd" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7ffff5289b40 (LWP 6995)]
0x00007ffff76f06b9 in userio_xml_escape () from /lib/x86_64-linux-gnu/libindidriver.so.2
#0  0x00007ffff76f06b9 in userio_xml_escape () from /lib/x86_64-linux-gnu/libindidriver.so.2
#1  0x00007ffff76f7ce4 in IUUserIODefNumberVA () from /lib/x86_64-linux-gnu/libindidriver.so.2
#2  0x00007ffff767fb07 in IDDefNumberVA () from /lib/x86_64-linux-gnu/libindidriver.so.2
#3  0x00007ffff7713051 in INDI::PropertyView<_INumber>::vdefine(char const*, __va_list_tag*) const () from /lib/x86_64-linux-gnu/libindidriver.so.2
#4  0x00007ffff7713f5a in INDI::PropertyView<_INumber>::define(char const*, ...) const () from /lib/x86_64-linux-gnu/libindidriver.so.2
#5  0x00007ffff768b27c in INDI::DefaultDevice::defineProperty(_INumberVectorProperty*) () from /lib/x86_64-linux-gnu/libindidriver.so.2
#6  0x0000555555560486 in SVBONYCCD::updateProperties (this=0x5555555abca0) at /home/alfio/Projects/indi-3rdparty/indi-svbony/svbony_ccd.cpp:144
#7  0x00007ffff768595f in ?? () from /lib/x86_64-linux-gnu/libindidriver.so.2
#8  0x00007ffff77188db in INDI::PropertySwitch::update(ISState const*, char const* const*, int) () from /lib/x86_64-linux-gnu/libindidriver.so.2
#9  0x00007ffff7687202 in INDI::DefaultDevice::ISNewSwitch(char const*, char const*, ISState*, char**, int) () from /lib/x86_64-linux-gnu/libindidriver.so.2
#10 0x00007ffff7694df0 in INDI::CCD::ISNewSwitch(char const*, char const*, ISState*, char**, int) () from /lib/x86_64-linux-gnu/libindidriver.so.2
#11 0x0000555555569a43 in SVBONYCCD::ISNewSwitch (this=0x5555555abca0, dev=0x5555555c1f60 "SVBONY SV705C 0", name=0x5555555c7cc0 "CONNECTION", states=0x5555555e4970, names=0x5555555e40a0, n=1) at /home/alfio/Projects/indi-3rdparty/indi-svbony/svbony_ccd.cpp:1634
#12 0x00007ffff76811f1 in ISNewSwitch () from /lib/x86_64-linux-gnu/libindidriver.so.2
#13 0x00007ffff7681f7d in dispatch () from /lib/x86_64-linux-gnu/libindidriver.so.2
#14 0x00007ffff7682870 in ?? () from /lib/x86_64-linux-gnu/libindidriver.so.2
#15 0x00007ffff770e364 in ?? () from /lib/x86_64-linux-gnu/libindidriver.so.2
#16 0x00007ffff77167a2 in eventLoop () from /lib/x86_64-linux-gnu/libindidriver.so.2
#17 0x00007ffff7681122 in main () from /lib/x86_64-linux-gnu/libindidriver.so.2
#18 0x00007ffff6e23a90 in __libc_start_call_main (main=main@entry=0x7ffff7680fd0 <main>, argc=argc@entry=1, argv=argv@entry=0x7fffffffdb38) at ../sysdeps/nptl/libc_start_call_main.h:58
#19 0x00007ffff6e23b49 in __libc_start_main_impl (main=0x7ffff7680fd0 <main>, argc=1, argv=0x7fffffffdb38, init=<optimised out>, fini=<optimised out>, rtld_fini=<optimised out>, stack_end=0x7fffffffdb28) at ../csu/libc-start.c:360
#20 0x000055555555ff25 in _start ()put

This GDB supports auto-downloading debuginfo from the following URLs:
[... 89 lines cut ...]
2023-09-09T10:52:08: Driver indi_simulator_telescope: HaAxis: TrackRate 1, trackingRateDegSec 15.041067 arcsec
2023-09-09T10:52:08: Driver indi_simulator_telescope: snooping on GPS Simulator.GEOGRAPHIC_COORD
2023-09-09T10:52:08: Driver indi_simulator_telescope: snooping on GPS Simulator.TIME_UTC
2023-09-09T10:52:08: Driver indi_simulator_telescope: snooping on Dome Simulator.DOME_PARK
[... cut ...]
2023-09-09T10:52:08: Driver indi_svbony_ccd: snooping on SQM.SKY_QUALITY
2023-09-09T10:52:53: Client 11: new arrival from 127.0.0.1:54694 - welcome!
2023-09-09T10:53:04: Driver indi_svbony_ccd: Impossible IPState 14666
2023-09-09T10:53:04: Driver indi_svbony_ccd: Impossible IPerm 856293408
2023-09-09T10:53:04: Client 12: new arrival from 127.0.0.1:47208 - welcome!
2023-09-09T10:53:04: Driver indi_svbony_ccd: read: Connection reset by peer
<delProperty device="SVBONY SV705C 0"/>
2023-09-09T10:53:04: Driver indi_svbony_ccd: Terminated after #0 restarts.
```

knro, 2023-09-09:
Need remote access to debug further. You're running StellarMate or something else?

pap3rinik, 2023-09-09:
running a Ubuntu machine on a old macbook pro (mid 2012)
which kind of acces do you need, maybe I could arrange something and leave the machine running.

knro, 2023-09-10:
you can install Rustdesk and send the credentials to support@stellarmate.com

pap3rinik, 2023-09-10:
that's fine, I'll send you credential ASAP
(in the meantime I tried on another machine and found a whole new crash, LOL, opening another issue)

knro, 2024-01-24:
is it still crashing with latest SVBony SDK?

alemassimo, 2025-11-10:
Hi, has this bug been fixed or is there a workaround? Tnx