indilib/indi-3rdparty #832: Connection lost on camera Altair AA183CPRO

tclaus, 2023-09-10:
**Describe the bug**
Randomly the connection to the driver for the camera Altair AA183CPRO gets lost. After taking some images (somtimes it does even not need to take a picture to get this error). The driver responds: 

Altair AA183CPRO :  "[ERROR] Camera disconnected "
followed by a 
Altair AA183CPRO :  "[ERROR] get Temperature error. Generic failure "

Only stopping and restarting the Indi service will stop this continuous message.

**To Reproduce**
Run driver:
indiserver indi_altair_ccd
Perform action
Nothing specific, it seems more or less random
See error
In Device manager you will see: [ERROR] get Temperature error. Generic failure

**Expected behavior**
The driver should not loose connection to the camera.

**Desktop (please complete the following information):**
 - OS: Stellarmate with WebManager 1.7.8
 - Linux stellarmate 6.1.21-v8+ on Raspberry Pi

**Log Files**

[log_14-59-48.txt](https://github.com/indilib/indi-3rdparty/files/12567848/log_14-59-48.Kopie.txt)

knro, 2023-09-10:
@touptek Can you please check this issue?

tclaus, 2024-01-23:
Thank you for responding. 
By reading the user groups I found out that this is a frequently occurring problem especially with this model. 
I tested this on Astroberry, Stellarmate and  directly attached to my (Mac) laptop. No stable connection with this camera model.

However - I sold this camera and bought another one. I also switched from INDI to INDIGO and know have a stable environment. 
Sorry that I can not test this again.

closed 2024-01-23 by knro