indilib/indi-3rdparty #843: indi_nightscape with the KAI-10100 sensor

felixgonsug, 2023-09-26:
**Describe the bug**
Hi, im trying to use the indi-nightscape driver with a camera with the KAI-10100 sensor. The driver is written for de 8300 sensor. 

Reading the code and the logs, i realized that in the file indi-nightscape/nsmsg.cpp line 10, the driver compares some exadecimal string to detect that the camera is an 8300 one. So i change that string to match the one a KAI-10100 returns and the driver connects correctly.
When viewing the camera from ekos, the interface works as expected, i can define the exposure time and get the temperature of the sensor. The problem is with the image downloaded, it is all black. After failing, i also changed some values to match the KAI-10100 sensor, those changes are shown on the diff file included 
[nightscape_diff.txt](https://github.com/indilib/indi-3rdparty/files/12731658/nightscape_diff.txt).

**To Reproduce**
Exact steps to reproduce the behavior.
1. Change the driver code in file indi-nightscape/nsmsg.cpp line 10.
```
-               const unsigned char Nsmsg::inqr_8600 [CMD_SIZE] = {0xA5,1,1,0x80,0,0x4F,0xD8,1,0x26,0,0,0,0,0,0,0};
+               const unsigned char Nsmsg::inqr_8600 [CMD_SIZE] = {0xA5,1,0x3,0,0,0x29,0x3C,0,0x2B,0,0x2B,0,0,0,0,0};
```
2. Connect with ekos and take a picture
3. Error in image downloaded

**Expected behavior**
Get a correct image

**Screenshots**
This is an image obtained when taking a photo.
[image]

**Desktop (please complete the following information):**
 - OS: Linux oam 5.10.0-23-amd64 SMP Debian 5.10.179-1 (2023-05-12) x86_64 GNU/Linux
 - INDI Library: 2.0.2 Code 2.0.2-tgz. Protocol 1.7.
 - Driver version: 1.7
 - Client OS: Ubuntu 18.04
 - Client: Kstars/Ekos from ubuntu repo

**Logs**
The log is truncated in the image part in order to be less dan 25MB
[nightscape_log.txt](https://github.com/indilib/indi-3rdparty/files/12731742/nightscape_log.txt)

A thing to note is that the image that was downloaded when logging had a grey band in the upper part, the other images downloaded, like the one in the screenshots, are all black.

@dirkenstein we exchange some emails in order to make this work. I'll thank you very much if you can point me in the right direction.

felixgonsug, 2023-10-03:
@knro Since you are one of the active commiters, maybe you can help with this. Thanks in advance.

knro, 2024-01-23:
Ok it looks it is more complicated than just changing a few parameters? Did you make any advances with this sensor? I'm afraid I have no experience with this driver, but if there is a solution that works, then we can work on implementing it so that it works for both sensors within the driver.