indilib/indi-3rdparty #844: The library: libPkTriggerCord is outdated. Can it be updated from PkTriggerCord?

ian-morgan99, 2023-09-30:
**Describe the bug**
The server 'Astroberry' for Raspberry Pi uses Ekos, that uses IndiLib. IndiLib uses libPkTriggerCord. This is a library that is copied from PkTriggerCode. However, PkTriggerCord was updated to include the Pentax K3iii and Pentax K1ii cameras some time ago. Those changes don't seem to have found their way through to AstroBerry , IndiLib, possibly as they are not in libPkTriggerCord.

**To Reproduce**
1. Build an astroberry server on Raspberry Pi 4 B 8GB.
2. Open 'Ekos'.
3. Configure the Pentax Native Driver. 
4. Connect a Pentax camera. (K3iii or K1ii)
5. The camera does not connect. 
6. Further manual configuration of the IndiLib server also yields similar results. I have been unable to connect to these cameras.

**Expected behavior**
It should be possible to connect to the camera, using the IndiLib driver, in Ekos within Astroberry.

knro, 2023-09-30:
Can you please submit a PR to update this library? I also read that libgphoto2 merged PkTriggerCoord ? can you check?

ian-morgan99, 2023-09-30:
I'm sorry. I'm new to this. What's a 'PR' ?

ian-morgan99, 2023-09-30:
(However libgphoto2 was updated yesterday, I think, after a similar request , but the next release may be a while)

ian-morgan99, 2023-09-30:
Commit from glibphoto2 : https://github.com/gphoto/libgphoto2/commit/1c4f11313827f4a5bacabc0ca3b9179c963d9115

knro, 2023-09-30:
If you're using INDI's PPA, then we just built libgphoto2 today with latest master.

ian-morgan99, 2023-09-30:
I'm still learning this process , but if I do an 'update' in astroberry , which should update any dependencies , do you think this will pick up the new indilib, and therefore the new libgphoto ?

closed 2024-01-23 by knro